USBGuard/usbguard #165: feature request: add rule condition "screen is locked"

noam1023, 2017-02-21:
I want this functionality:
* when working on the computer, allow all devices
* when the screen lock is activated, reject any USB connection events.

dkopecek, 2017-02-21:
Hi, this is achievable with the latest version in git (HEAD) and will be with the 0.6.3 release.

Although there's no "screen locked" condition in the current rule language, there is an `InsertedDevicePolicy` run-time parameter now. The value of this parameter can be set to:
 - apply-policy = apply the policy (rules.conf) for every inserted device. 
 - block = block every inserted device
 - reject = reject every inserted device (logically disconnect from the system -- kernel won't see it anymore)

So, to implement the behavior you want with the current version, you'll have to:
 1. call `usbguard set-parameter InsertedDevicePolicy block` when the screen gets locked
 2. call `usbguard set-parameter InsertedDevicePolicy apply-policy` when the screen gets unlocked

The missing bits preventing a pure "usbguard-based solution" are the hooks to the screen lock/unlock events. If you know how to implement listening to screen lock/unlock events in a generic way (i.e. support for all or most of the different screen lockers in the wild), then please tell me and I'm going to try to implement such a condition in the rule language.

noam1023, 2017-02-22:
Thanks.
There is of course the danger to lock yourself out if you lock the screen,
unplug the keyboard, then plug it again only to find that you can't type
the password. I don't have an idea yet how to protect against this.

dkopecek, 2017-04-26:
Related to #139

BenjaminSchubert, 2017-07-06:
Actually there is a DBus interface for this : https://people.freedesktop.org/~hadess/idle-inhibition-spec/re01.html

But this would require DBus which is not in the base usbguard daemon.

One way I see this is a hook in the usbguard-dbus system / the GUI (when supporting DBus) that could dynamically change the default rule to block (so that the already created rules also apply, not to get locked out :) )

I think this option makes no sense on a server but lots of sense on a desktop, therefore implementing it in the GUI would not be such a problem (I know how to do this in case we decide to go this way).

grawity, 2017-07-15:
> Actually there is a DBus interface for this : https://people.freedesktop.org/~hadess/idle-inhibition-spec/re01.html

That's a completely unrelated interface (inhibits only prevent things like suspend) &ndash; you'll instead need to query systemd-logind or ConsoleKit2 about the session status, e.g. logind has `LockedHint`.

Remember that screensavers/screenlockers _are not required_ to actually report the 'locked' status at all. (Most screensavers merely grab all input via X11 and that's as far as the "locking" goes...) So this check will always have false negatives.

(Also, a daemon-wide "screen locked" rule needs you to decide whether the condition means "active (foreground) session on seat0 is locked" or "active sessions on all seats are locked" or "at least one active session is locked" or something else. Even if you ignore multihead systems, "seat0" might still have multiple sessions on different TTYs. The login screen is also a session, but it needs to be exempt.)

Might be better to implement something in the _applet_ rather than the daemon (e.g. block without prompting if a screensaver is detected), though it'd probably have X11-specific hacks.

BenjaminSchubert, 2017-07-15:
> Might be better to implement something in the applet rather than the daemon (e.g. block without prompting if a screensaver is detected), though it'd probably have X11-specific hacks.

I'd rather not have this, as more and more distros are shipping with wayland now :/

muelli, 2019-01-29:
cf. https://gitlab.gnome.org/GNOME/gnome-settings-daemon/merge_requests/75

What we would like to see is the ability for USBGuard to allow devices when the session is not running (correctly) i.e. to make sure we don't lock users out. The scenario is that we lock the screen and what USBGuard to *temporarily* block all USB devices. If we crash or reboot before unlocking the screen, we want the USB ports to work normally.  Currently, it seems impossible to tell USBGuard to only take a decision when there is an agent making the decision, i.e. GNOME or to make the setting transient rather than persistent.