USBGuard/usbguard #166: usbguard generate-policy writes /sys

cgzones, 2017-02-21:
While writing a SELinux policy for usbguard, i noticed `usbguard generate-policy` tries to write to `usb1/authorized_default`. Why does it so? The generated policy looks ok, when prohibiting the access.

usbguard output:
```
root@raspberrypi:/root/selinux/policy# usbguard generate-policy -p
[1487696646.646] (E) sysioWriteFileAt: Cannot open usb1/authorized_default: errno=13
[1487696646.646] (W) Cannot set default authorization state for devpath=usb1: Write operation failed.
allow id 1d6b:0002 serial "3f980000.usb" name "DWC OTG Controller" hash "EYx+GlFlkA3XIp4+eFRwuSFQM3DN0fehOlLVfI65aWw=" parent-hash "z+EE0r7/c8O/LqMIXf34DcLJgMjgneWBmFWXqrBWzsc=" via-port "usb1" with-interface 09:00:00
allow id 0424:9514 serial "" name "" hash "E0MWcMEgHt8SbS4HZnYsm7CORelcUwUjAtJKvvAP8yg=" parent-hash "EYx+GlFlkA3XIp4+eFRwuSFQM3DN0fehOlLVfI65aWw=" via-port "1-1" with-interface { 09:00:01 09:00:02 }
allow id 0424:ec00 serial "" name "" hash "t3Q3NWJOsmxMC5TAcN4lYL6vjeaHcShDXODSx2q2U08=" parent-hash "E0MWcMEgHt8SbS4HZnYsm7CORelcUwUjAtJKvvAP8yg=" via-port "1-1.1" with-interface ff:00:ff
[1487696646.666] (E) sysioWriteFileAt: Cannot open usb1/authorized_default: errno=13
[1487696646.667] (W) Cannot set default authorization state for devpath=usb1: Write operation failed.
```

SELinux output:
```
type=PROCTITLE msg=audit(02/21/17 18:02:10.916:1067) : proctitle=usbguard generate-policy
type=PATH msg=audit(02/21/17 18:02:10.916:1067) : item=0 name=usb1/authorized_default inode=9705 dev=00:10 mode=file,644 ouid=root ogid=root rdev=00:00 obj=system_u:object_r:sysfs_t:s0 nametype=NORMAL
type=CWD msg=audit(02/21/17 18:02:10.916:1067) : cwd=/root/selinux/policy
type=SYSCALL msg=audit(02/21/17 18:02:10.916:1067) : arch=armeb syscall=openat per=PER_LINUX_32BIT success=no exit=EACCES(Permission denied) a0=0x5 a1=0x15919c8 a2=O_WRONLY a3=0x0 items=1 ppid=24509 pid=25351 auid=christian uid=root gid=root euid=root suid=root fsuid=root egid=root sgid=root fsgid=root tty=pts5 ses=1 comm=usbguard exe=/usr/bin/usbguard subj=staff_u:sysadm_r:usbguard_app_t:s0 key=(null)
type=AVC msg=audit(02/21/17 18:02:10.916:1067) : avc:  denied  { write } for  pid=25351 comm=usbguard name=authorized_default dev="sysfs" ino=9705 scontext=staff_u:sysadm_r:usbguard_app_t:s0 tcontext=system_u:object_r:sysfs_t:s0 tclass=file permissive=0
```

inode resolution:
```
root@raspberrypi:/root/selinux/policy# find /sys/ -inum 9705
/sys/devices/platform/soc/3f980000.usb/usb1/authorized_default
```

dkopecek, 2017-02-21:
Hi, thanks for the report. I'll change the behavior so that it doesn't touch that file during device enumeration.

dkopecek, 2017-03-05:
@cgzones which version did you test? I realized that this is already fixed in the git version (unreleased):
 
 https://github.com/dkopecek/usbguard/blob/master/src/Library/UEventDeviceManager.cpp#L818

It was introduced in f42fd93cd6a97db911ae32e32a376a27ebe5f76f

dkopecek, 2017-05-10:
Closing as this should be fixed in the latest release.

closed 2017-05-10 by dkopecek