USBGuard/usbguard #168: USBGuard should ship required config directories

genodeftest, 2017-04-07:
Steps to reproduce:
1. update to usbguard-0.6.3-0.2.1491223929.fc25.x86_64
2. reboot

What happens:
usbguard-daemon.service fails to start with this message: ```usbguard-daemon[900]: [1491549121.593] (E) ERROR: loadFiles: /etc/usbguard/IPCAccessControl.d/: No such file or directory```

Additional info:
systemd tries to restart the service, but this fails too.

The usbguard RPM package does not ship the required file /etc/usbguard/IPCAAccessControl.d, but it should, so it can own the directory and set permissions (e.g. readable only by root). If the package does not make sure the directory exists, the daemon should not fail if it is missing.

dkopecek, 2017-04-07:
Thanks for the report! Should be fixed in git and new copr build underway.

closed 2017-04-07 by dkopecek

genodeftest, 2017-04-09:
Thanks, fix confirmed!

commit 226f1350 (2017-04-07): Add IPCAccessControl.d dir into the RPM .spec file

Resolves: #168