USBGuard/usbguard #172: make check fails with undefined reference

murinicanor, 2017-05-19:
Hi, i'm trying to build 0.7, but when doing `make check` g++ doesn't find some libs:
```
   dh_auto_test -O--parallel
	make -j4 check VERBOSE=1
make[1]: Entering directory '/build/usbguard-0.7.0'
make  check-recursive
make[2]: Entering directory '/build/usbguard-0.7.0'
Making check in src/Tests/
make[3]: Entering directory '/build/usbguard-0.7.0/src/Tests'
make  test-unit test-regression
make[4]: Entering directory '/build/usbguard-0.7.0/src/Tests'
g++ -DHAVE_CONFIG_H -I. -I../../src  -I../../src -I../../src/Library -I/usr/include/catch  -Wdate-time -D_FORTIFY_SOURCE=2  -std=c++11 -g -O2 -fdebug-prefix-map=/build/usbguard-0.7.0=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC  -fvisibility=hidden  -pedantic -Wno-unknown-pragmas -Wall -Wextra -Wformat=2 -Wredundant-decls -Wcast-align -Wmissing-declarations -Wmissing-include-dirs -Wmissing-format-attribute -Wswitch-enum -Wswitch-default -Winvalid-pch -Wformat-nonliteral -Wno-missing-field-initializers -Wno-implicit-fallthrough  -c -o main.o main.cpp
g++ -DHAVE_CONFIG_H -I. -I../../src  -I../../src -I../../src/Library -I/usr/include/catch  -Wdate-time -D_FORTIFY_SOURCE=2  -std=c++11 -g -O2 -fdebug-prefix-map=/build/usbguard-0.7.0=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC  -fvisibility=hidden  -pedantic -Wno-unknown-pragmas -Wall -Wextra -Wformat=2 -Wredundant-decls -Wcast-align -Wmissing-declarations -Wmissing-include-dirs -Wmissing-format-attribute -Wswitch-enum -Wswitch-default -Winvalid-pch -Wformat-nonliteral -Wno-missing-field-initializers -Wno-implicit-fallthrough  -c -o Unit/test_Rule.o Unit/test_Rule.cpp
g++ -DHAVE_CONFIG_H -I. -I../../src  -I../../src -I../../src/Library -I/usr/include/catch  -Wdate-time -D_FORTIFY_SOURCE=2  -std=c++11 -g -O2 -fdebug-prefix-map=/build/usbguard-0.7.0=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC  -fvisibility=hidden  -pedantic -Wno-unknown-pragmas -Wall -Wextra -Wformat=2 -Wredundant-decls -Wcast-align -Wmissing-declarations -Wmissing-include-dirs -Wmissing-format-attribute -Wswitch-enum -Wswitch-default -Winvalid-pch -Wformat-nonliteral -Wno-missing-field-initializers -Wno-implicit-fallthrough  -c -o Unit/test_RuleParser.o Unit/test_RuleParser.cpp
cc1plus: warning: /usr/include/catch: No such file or directory [-Wmissing-include-dirs]
cc1plus: warning: /usr/include/catch: No such file or directory [-Wmissing-include-dirs]
g++ -DHAVE_CONFIG_H -I. -I../../src  -I../../src -I../../src/Library -I/usr/include/catch  -Wdate-time -D_FORTIFY_SOURCE=2  -std=c++11 -g -O2 -fdebug-prefix-map=/build/usbguard-0.7.0=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC  -fvisibility=hidden  -pedantic -Wno-unknown-pragmas -Wall -Wextra -Wformat=2 -Wredundant-decls -Wcast-align -Wmissing-declarations -Wmissing-include-dirs -Wmissing-format-attribute -Wswitch-enum -Wswitch-default -Winvalid-pch -Wformat-nonliteral -Wno-missing-field-initializers -Wno-implicit-fallthrough  -c -o Unit/test_Base64.o Unit/test_Base64.cpp
cc1plus: warning: /usr/include/catch: No such file or directory [-Wmissing-include-dirs]
cc1plus: warning: /usr/include/catch: No such file or directory [-Wmissing-include-dirs]
cc1plus: warning: unrecognized command line option '-Wno-implicit-fallthrough'
g++ -DHAVE_CONFIG_H -I. -I../../src  -I../../src -I../../src/Library -I/usr/include/catch  -Wdate-time -D_FORTIFY_SOURCE=2  -std=c++11 -g -O2 -fdebug-prefix-map=/build/usbguard-0.7.0=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC  -fvisibility=hidden  -pedantic -Wno-unknown-pragmas -Wall -Wextra -Wformat=2 -Wredundant-decls -Wcast-align -Wmissing-declarations -Wmissing-include-dirs -Wmissing-format-attribute -Wswitch-enum -Wswitch-default -Winvalid-pch -Wformat-nonliteral -Wno-missing-field-initializers -Wno-implicit-fallthrough  -c -o Unit/test_UEvent.o Unit/test_UEvent.cpp
cc1plus: warning: /usr/include/catch: No such file or directory [-Wmissing-include-dirs]
cc1plus: warning: unrecognized command line option '-Wno-implicit-fallthrough'
g++ -DHAVE_CONFIG_H -I. -I../../src  -I../../src -I../../src/Library -I/usr/include/catch  -Wdate-time -D_FORTIFY_SOURCE=2  -std=c++11 -g -O2 -fdebug-prefix-map=/build/usbguard-0.7.0=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC  -fvisibility=hidden  -pedantic -Wno-unknown-pragmas -Wall -Wextra -Wformat=2 -Wredundant-decls -Wcast-align -Wmissing-declarations -Wmissing-include-dirs -Wmissing-format-attribute -Wswitch-enum -Wswitch-default -Winvalid-pch -Wformat-nonliteral -Wno-missing-field-initializers -Wno-implicit-fallthrough  -c -o Unit/test_UEventParser.o Unit/test_UEventParser.cpp
cc1plus: warning: /usr/include/catch: No such file or directory [-Wmissing-include-dirs]
cc1plus: warning: unrecognized command line option '-Wno-implicit-fallthrough'
g++ -DHAVE_CONFIG_H -I. -I../../src  -I../../src -I../../src/Library -I/usr/include/catch  -Wdate-time -D_FORTIFY_SOURCE=2  -std=c++11 -g -O2 -fdebug-prefix-map=/build/usbguard-0.7.0=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC  -fvisibility=hidden  -pedantic -Wno-unknown-pragmas -Wall -Wextra -Wformat=2 -Wredundant-decls -Wcast-align -Wmissing-declarations -Wmissing-include-dirs -Wmissing-format-attribute -Wswitch-enum -Wswitch-default -Winvalid-pch -Wformat-nonliteral -Wno-missing-field-initializers -Wno-implicit-fallthrough  -c -o Unit/test_Utility.o Unit/test_Utility.cpp
cc1plus: warning: /usr/include/catch: No such file or directory [-Wmissing-include-dirs]
cc1plus: warning: unrecognized command line option '-Wno-implicit-fallthrough'
g++ -DHAVE_CONFIG_H -I. -I../../src  -I../../src -I../../src/Library -I/usr/include/catch  -Wdate-time -D_FORTIFY_SOURCE=2  -std=c++11 -g -O2 -fdebug-prefix-map=/build/usbguard-0.7.0=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC  -fvisibility=hidden  -pedantic -Wno-unknown-pragmas -Wall -Wextra -Wformat=2 -Wredundant-decls -Wcast-align -Wmissing-declarations -Wmissing-include-dirs -Wmissing-format-attribute -Wswitch-enum -Wswitch-default -Winvalid-pch -Wformat-nonliteral -Wno-missing-field-initializers -Wno-implicit-fallthrough  -c -o Unit/test_RuleAttribute_id.o Unit/test_RuleAttribute_id.cpp
cc1plus: warning: /usr/include/catch: No such file or directory [-Wmissing-include-dirs]
cc1plus: warning: unrecognized command line option '-Wno-implicit-fallthrough'
g++ -DHAVE_CONFIG_H -I. -I../../src  -I../../src -I../../src/Library -I/usr/include/catch  -Wdate-time -D_FORTIFY_SOURCE=2  -std=c++11 -g -O2 -fdebug-prefix-map=/build/usbguard-0.7.0=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC  -fvisibility=hidden  -pedantic -Wno-unknown-pragmas -Wall -Wextra -Wformat=2 -Wredundant-decls -Wcast-align -Wmissing-declarations -Wmissing-include-dirs -Wmissing-format-attribute -Wswitch-enum -Wswitch-default -Winvalid-pch -Wformat-nonliteral -Wno-missing-field-initializers -Wno-implicit-fallthrough  -c -o Unit/test_IPCServer_AccessControl.o Unit/test_IPCServer_AccessControl.cpp
cc1plus: warning: /usr/include/catch: No such file or directory [-Wmissing-include-dirs]
cc1plus: warning: unrecognized command line option '-Wno-implicit-fallthrough'
g++ -DHAVE_CONFIG_H -I. -I../../src  -I../../src -I../../src/Library -I/usr/include/catch  -Wdate-time -D_FORTIFY_SOURCE=2  -std=c++11 -g -O2 -fdebug-prefix-map=/build/usbguard-0.7.0=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC  -fvisibility=hidden  -pedantic -Wno-unknown-pragmas -Wall -Wextra -Wformat=2 -Wredundant-decls -Wcast-align -Wmissing-declarations -Wmissing-include-dirs -Wmissing-format-attribute -Wswitch-enum -Wswitch-default -Winvalid-pch -Wformat-nonliteral -Wno-missing-field-initializers -Wno-implicit-fallthrough  -c -o Regression/test_Rule_ghi37.o Regression/test_Rule_ghi37.cpp
cc1plus: warning: /usr/include/catch: No such file or directory [-Wmissing-include-dirs]
cc1plus: warning: unrecognized command line option '-Wno-implicit-fallthrough'
g++ -DHAVE_CONFIG_H -I. -I../../src  -I../../src -I../../src/Library -I/usr/include/catch  -Wdate-time -D_FORTIFY_SOURCE=2  -std=c++11 -g -O2 -fdebug-prefix-map=/build/usbguard-0.7.0=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC  -fvisibility=hidden  -pedantic -Wno-unknown-pragmas -Wall -Wextra -Wformat=2 -Wredundant-decls -Wcast-align -Wmissing-declarations -Wmissing-include-dirs -Wmissing-format-attribute -Wswitch-enum -Wswitch-default -Winvalid-pch -Wformat-nonliteral -Wno-missing-field-initializer [... cut ...]
make[3]: Leaving directory '/build/usbguard-0.7.0/src/Tests'
Makefile:2634: recipe for target 'check-recursive' failed
make[2]: *** [check-recursive] Error 1
make[2]: Leaving directory '/build/usbguard-0.7.0'
Makefile:2924: recipe for target 'check' failed
make[1]: *** [check] Error 2
make[1]: Leaving directory '/build/usbguard-0.7.0'
dh_auto_test: make -j4 check VERBOSE=1 returned exit code 2
debian/rules:16: recipe for target 'build' failed
make: *** [build] Error 2
```

do you have an idea what i'm missing?

dkopecek, 2017-05-19:
Hi, try configuring with --enable-static and recompile. The unit tests need a static version of libusbguard.

murinicanor, 2017-05-20:
ah, yes, that fixes it. thanks!

closed 2017-05-20 by murinicanor