USBGuard/usbguard #175: Questions regarding Tests

murinicanor, 2017-06-06:
Hi Daniel,

i've some questions regarding the tests:
first, just to let you know, 002_cli_devices.sh won't work in Debian, as dummy_hcd and g_mass_storage are not included as modules. is there a way of disabling the test or do i have to patch `src/Tests/Makefile.in`?

and i have a problem with 001_cli_policy.sh and i'm not able to find out why it fails, maybe you can have a look?
```
FAIL: UseCase/001_cli_policy.sh
UseCase/001_cli_policy.sh.log:  ++ COMMAND_QUEUE=()
UseCase/001_cli_policy.sh.log:  ++ COMMAND_JOBID=()
UseCase/001_cli_policy.sh.log:  ++ COMMAND_STATE=()
UseCase/001_cli_policy.sh.log:  ++ COMMAND_SETUP=()
UseCase/001_cli_policy.sh.log:  ++ COMMAND_OFILE=()
UseCase/001_cli_policy.sh.log:  + '[' -d /sys/bus/usb/devices ']'
UseCase/001_cli_policy.sh.log:  ++ mktemp -d --tmpdir usbguard-test.XXXXXX
UseCase/001_cli_policy.sh.log:  + export USBGUARD_TESTLIB_TMPDIR=/tmp/usbguard-test.YcUCyd
UseCase/001_cli_policy.sh.log:  + USBGUARD_TESTLIB_TMPDIR=/tmp/usbguard-test.YcUCyd
UseCase/001_cli_policy.sh.log:  + export config_path=/tmp/usbguard-test.YcUCyd/daemon.conf
UseCase/001_cli_policy.sh.log:  + config_path=/tmp/usbguard-test.YcUCyd/daemon.conf
UseCase/001_cli_policy.sh.log:  + export policy_path=/tmp/usbguard-test.YcUCyd/policy.conf
UseCase/001_cli_policy.sh.log:  + policy_path=/tmp/usbguard-test.YcUCyd/policy.conf
UseCase/001_cli_policy.sh.log:  + cat
UseCase/001_cli_policy.sh.log:  ++ id -un
UseCase/001_cli_policy.sh.log:  ++ id -gn
UseCase/001_cli_policy.sh.log:  + cat
UseCase/001_cli_policy.sh.log:  + schedule '../../usbguard-daemon -d -k -c /tmp/usbguard-test.YcUCyd/daemon.conf' :service
UseCase/001_cli_policy.sh.log:  + local 'command=../../usbguard-daemon -d -k -c /tmp/usbguard-test.YcUCyd/daemon.conf'
UseCase/001_cli_policy.sh.log:  + local setup=:service
UseCase/001_cli_policy.sh.log:  + COMMAND_QUEUE[${#COMMAND_QUEUE[@]}]='../../usbguard-daemon -d -k -c /tmp/usbguard-test.YcUCyd/daemon.conf'
UseCase/001_cli_policy.sh.log:  + COMMAND_JOBID[${#COMMAND_JOBID[@]}]=0
UseCase/001_cli_policy.sh.log:  + COMMAND_STATE[${#COMMAND_STATE[@]}]=0
UseCase/001_cli_policy.sh.log:  + COMMAND_SETUP[${#COMMAND_SETUP[@]}]=:service
UseCase/001_cli_policy.sh.log:  + COMMAND_OFILE[${#COMMAND_OFILE[@]}]=/dev/null
UseCase/001_cli_policy.sh.log:  + COMMAND_VFILE[${#COMMAND_VFILE[@]}]=/dev/null
UseCase/001_cli_policy.sh.log:  + schedule test_cli_policy
UseCase/001_cli_policy.sh.log:  + local command=test_cli_policy
UseCase/001_cli_policy.sh.log:  + local setup=
UseCase/001_cli_policy.sh.log:  + COMMAND_QUEUE[${#COMMAND_QUEUE[@]}]=test_cli_policy
UseCase/001_cli_policy.sh.log:  + COMMAND_JOBID[${#COMMAND_JOBID[@]}]=0
UseCase/001_cli_policy.sh.log:  + COMMAND_STATE[${#COMMAND_STATE[@]}]=0
UseCase/001_cli_policy.sh.log:  + COMMAND_SETUP[${#COMMAND_SETUP[@]}]=
UseCase/001_cli_policy.sh.log:  + COMMAND_OFILE[${#COMMAND_OFILE[@]}]=/dev/null
UseCase/001_cli_policy.sh.log:  + COMMAND_VFILE[${#COMMAND_VFILE[@]}]=/dev/null
UseCase/001_cli_policy.sh.log:  + execute 60
UseCase/001_cli_policy.sh.log:  + local timeout=60
UseCase/001_cli_policy.sh.log:  + local job_count=2
UseCase/001_cli_policy.sh.log:  ++ date +%s
UseCase/001_cli_policy.sh.log:  + local job_start=1496764987
UseCase/001_cli_policy.sh.log:  + local job_rc=0
UseCase/001_cli_policy.sh.log:  + local valgrind=
UseCase/001_cli_policy.sh.log:  + '[' -n '' ']'
UseCase/001_cli_policy.sh.log:  + '[' '' = off ']'
UseCase/001_cli_policy.sh.log:  + valgrind=
UseCase/001_cli_policy.sh.log:  + set +m
UseCase/001_cli_policy.sh.log:  + trap handle_SIGCHLD SIGCHLD
UseCase/001_cli_policy.sh.log:  + (( i = 0 ))
UseCase/001_cli_policy.sh.log:  + (( i < 2 ))
UseCase/001_cli_policy.sh.log:  + local setup=:service
UseCase/001_cli_policy.sh.log:  + local sudocmd=
UseCase/001_cli_policy.sh.log:  + local valgrindcmd=
UseCase/001_cli_policy.sh.log:  + local valgrindlog=
UseCase/001_cli_policy.sh.log:  + local valgrindenv=
UseCase/001_cli_policy.sh.log:  + echo :service
UseCase/001_cli_policy.sh.log:  + grep -q :sudo
UseCase/001_cli_policy.sh.log:  + test -n ''
UseCase/001_cli_policy.sh.log:  + local 'command=  ../../usbguard-daemon -d -k -c /tmp/usbguard-test.YcUCyd/daemon.conf'
UseCase/001_cli_policy.sh.log:  ++ mktemp --tmpdir usbguard-test-1496764987_job_0.XXXXXX.log
UseCase/001_cli_policy.sh.log:  + local logfile=/tmp/usbguard-test-1496764987_job_0.ZcoNaK.log
UseCase/001_cli_policy.sh.log:  + COMMAND_JOBID[$i]=6855
UseCase/001_cli_policy.sh.log:  + COMMAND_STATE[$i]=1
[... 216 lines cut ...]
UseCase/001_cli_policy.sh.log:  + test -n ''
UseCase/001_cli_policy.sh.log:  + (( ++i ))
UseCase/001_cli_policy.sh.log:  + (( i < 2 ))
UseCase/001_cli_policy.sh.log:  + local command=test_cli_policy
UseCase/001_cli_policy.sh.log:  + local jobid=6867
UseCase/001_cli_policy.sh.log:  + local setup=
UseCase/001_cli_policy.sh.log:  + local logfile=/tmp/usbguard-test-1496764987_job_1.bSYbEd.log
UseCase/001_cli_policy.sh.log:  + local valgrindlog=
UseCase/001_cli_policy.sh.log:  + local sudocmd=
UseCase/001_cli_policy.sh.log:  + echo ''
UseCase/001_cli_policy.sh.log:  + grep -q :sudo
UseCase/001_cli_policy.sh.log:  + echo ''
UseCase/001_cli_policy.sh.log:  + grep -q :service
UseCase/001_cli_policy.sh.log:  + wait 6867
UseCase/001_cli_policy.sh.log:  + '[' 1 -ne 0 ']'
UseCase/001_cli_policy.sh.log:  + job_rc=1
UseCase/001_cli_policy.sh.log:  + test -n ''
UseCase/001_cli_policy.sh.log:  + (( ++i ))
UseCase/001_cli_policy.sh.log:  + (( i < 2 ))
UseCase/001_cli_policy.sh.log:  + return 1
UseCase/001_cli_policy.sh.log:  FAIL UseCase/001_cli_policy.sh (exit status: 1)
```

thanks, cheers
muri

dkopecek, 2017-06-12:
> is there a way of disabling the test or do i have to patch src/Tests/Makefile.in?

For now you have to patch it. I'll implement an option to disable the tests or add a check for the kernel modules into the test script.

> UseCase/001_cli_policy.sh.log:  [1496764987.954] (E) ERROR: IPC server initialization: service: Address already in use

I think it fails because there's already a usbguard-daemon instance running on your system. The test assumes there's no other usbguard-daemon active.

Thanks for testing!

murinicanor, 2017-06-19:
> I think it fails because there's already a usbguard-daemon instance running on your system. The test assumes there's no other usbguard-daemon active.

ah, yes, you're right- in that case the test ran on my production system, thats why there was an usbguard instance running ;)

another thing: the file `src/Tests/USB/data/0001.bin`, is it possible to generate it somehow from source?
same goes for `src/Tests/UseCase/DummyDevices/root.tar.xz`, but thats probably `src/Tests/UseCase/DummyDevices/*`? (i've plan to remove the latter from the debian package for now, as the test is disabled)

whats the license for `src/Tests/custom.supp`?

dkopecek, 2017-06-24:
> another thing: the file src/Tests/USB/data/0001.bin, is it possible to generate it somehow from source?

No. This is a copy of a binary USB descriptor of one of my devices. I don't think I know anymore which one :-) Is that a problematic file? Maybe I could create a script to generate a bunch of testing descriptors in a deterministic way.

> same goes for src/Tests/UseCase/DummyDevices/root.tar.xz, but thats probably src/Tests/UseCase/DummyDevices/*? (i plan to remove the latter from the debian package for now, as the test is disabled)

Correct, that archive is just the contents of DummyDevices/.

> whats the license for src/Tests/custom.supp?

I'm going to add a public domain header in there. I don't think it makes sense to license that under GPL or something...

closed 2017-06-28 by dkopecek

murinicanor, 2017-07-07:
> Is that a problematic file? Maybe I could create a script to generate a bunch of testing descriptors in a deterministic way.

that would be great. i've created #188 for this

commit c7bb02dc (2017-06-28): Run full test suite only if configured with --enable-full-test-suite

Resolves: #175
commit faa84a64 (2017-06-28): Release custom.supp file into the public domain

Resolves: #175