USBGuard/usbguard #176: USBHuard won't build

LizziAS, 2017-06-15:
i use openrc init system not systemd and while i am able to run many programs using manjaro's way of bypassing the need for systemd on my openrc system, you package specifically requires and checks for systemd and the build fails.  Please consider versions for various init systems other than systemd.  Thank you.

robin0800, 2017-06-15:
I too have this problem, but even checking for the init system in use so the install does not fail probably will not be enough. I think we will also need an openrc-usbguard pkg to tell usbguard how to work with openrc

dkopecek, 2017-06-15:
Hello, thanks for the feedback! I'll be happy to integrate the project with other init systems as well. systemd is used because that's what I use daily on Fedora :-)

dkopecek, 2017-06-15:
@LizziAS Could you please send me some error output from the failed build? What exactly fails?

robin0800, 2017-06-15:
Here you go

 ``
Package systemd was not found in the pkg-config search path.
Perhaps you should add the directory containing `systemd.pc'
to the PKG_CONFIG_PATH environment variable
No package 'systemd' found
configure: error: in `/tmp/pamac-build-robin0800/usbguard/src/usbguard-0.7.0':
configure: error: Cannot detect the systemd system unit dir
See `config.log' for more details
==> ERROR: A failure occurred in build().
    Aborting...
``

dkopecek, 2017-06-15:
@robin0800 thanks. Does that happen even with "./configure --disable-systemd" ?

robin0800, 2017-06-15:
Got this error now
`checking catch.hpp usability... no
checking catch.hpp presence... no
checking for catch.hpp... no
configure: error: in `/tmp/yaourt-tmp-robin0800/aur-usbguard/src/usbguard-0.7.0':
configure: error: catch.hpp not found or not usable. Re-run with --with-bundled-catch to use the bundled library.
See `config.log' for more details
==> ERROR: A failure occurred in build().
    Aborting...
==> ERROR: Makepkg was unable to build usbguard.
`

dkopecek, 2017-06-15:
Either install https://github.com/philsquared/Catch to your system or try to use the bundled version by using the --with-bundled-catch configure option.

robin0800, 2017-06-15:
I have fixed that by installing "libcatch-cpp-headers"
the next problem is  "pegtl.hh"  I see it is an rpm package and using --with-bundled-pegtl in ./configure does not work as I guess it is not available in manjaro.

robin0800, 2017-06-15:
Yes it builds all I had to do was in the ./configure
``
  ./configure --prefix=/usr --sysconfdir=/etc -sbindir=/usr/bin --libdir=/usr/lib \
        --enable-systemd \
        --with-gui-qt=qt5 \
        --with-bundled-catch \
        --with-bundled-pegtl
``
was to change the enable systemd to disable systemd so it is now installed, what next?

dkopecek, 2017-06-16:
Configure the daemon (modify /etc/usbguard/usbguard-daemon.conf to your needs), then generate an initial policy using something like:

    # usbguard generate-policy > /etc/usbguard/rules.conf

and then start the daemon. Since you don't use systemd, just start it manually:

    # usbguard-daemon -c /etc/usbguard/usbguard-daemon.conf -k -d

Please note that it won't fork into background. The `-d` option is for debugging level messages. In case you find a bug, just file a new ticket and attache the debugging output from the daemon.

Also, configure the usbguard IPC permissions for your desktop session user so that you can run the usbguard-applet-qt, see the IPC ACCESS CONTROL section in usbguard-daemon.conf(5) man page for details.

robin0800, 2017-06-16:
Modified first command and got
``
usbguard generate-policy >sudo /etc/usbguard/rules.conf
[1497609941.582] (W) device enumeration exception: /sys/bus/usb/devices/usb1: SysFSDevice: uevent: Permission denied
[1497609941.582] (W) device enumeration exception: /sys/bus/usb/devices/usb2: SysFSDevice: uevent: Permission denied
[1497609941.582] (W) device enumeration exception: /sys/bus/usb/devices/usb3: SysFSDevice: uevent: Permission denied
[1497609941.582] (W) device enumeration exception: /sys/bus/usb/devices/usb4: SysFSDevice: uevent: Permission denied
[1497609941.582] (W) device enumeration exception: /sys/bus/usb/devices/1-1: SysFSDevice: uevent: Permission denied
[1497609941.582] (W) device enumeration exception: /sys/bus/usb/devices/3-2: SysFSDevice: uevent: Permission denied
[1497609941.583] (W) device enumeration exception: /sys/bus/usb/devices/3-3: SysFSDevice: uevent: Permission denied
[1497609941.583] (W) device enumeration exception: /sys/bus/usb/devices/4-1: SysFSDevice: uevent: Permission denied
[1497609941.583] (W) device enumeration exception: /sys/bus/usb/devices/4-1.5: SysFSDevice: uevent: Permission denied
[1497609941.583] (W) device enumeration exception: /sys/bus/usb/devices/4-1.6: SysFSDevice: uevent: Permission denied
``

dkopecek, 2017-06-16:
usbguard-daemon needs to be run under the root user...

dkopecek, 2017-06-16:
and the "usbguard generate-policy" too

robin0800, 2017-06-16:
With sudo in front (note no prompt for password)
``
 sudo usbguard generate-policy > /etc/usbguard/rules.conf
bash: /etc/usbguard/rules.conf: Permission denied
``
``
[robin0800@robins-desktop ~]$ sudo usbguard generate-policy
[sudo] password for robin0800: 
allow id 1d6b:0002 serial "0000:00:1a.0" name "EHCI Host Controller" hash "ej1WVedyLyUMLiQxzEcrwbY45zCodwV85Kzy7hm2Gv4=" parent-hash "uvJm0y/N2iYeJgfKJsJqWKTJts/duhYZ7W2zzAYk7Y8=" with-interface 09:00:00
allow id 1d6b:0002 serial "0000:00:14.0" name "xHCI Host Controller" hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" parent-hash "G1ehGQdrl3dJ9HvW9w2HdC//pk87pKzFE1WY25bq8k4=" with-interface 09:00:00
allow id 1d6b:0003 serial "0000:00:14.0" name "xHCI Host Controller" hash "3Wo3XWDgen1hD5xM3PSNl3P98kLp1RUTgGQ5HSxtf8k=" parent-hash "G1ehGQdrl3dJ9HvW9w2HdC//pk87pKzFE1WY25bq8k4=" with-interface 09:00:00
allow id 1d6b:0002 serial "0000:00:1d.0" name "EHCI Host Controller" hash "WHBTxNaEoMGNSNc31KpFNSAeFF4HbLMQgSBqORlC6S8=" parent-hash "FjkaT8Rp/Bh++KC4RQhk++hWack2wTDa1a1G5yXqYys=" with-interface 09:00:00
allow id 8087:0024 serial "" name "" hash "kv3v2+rnq9QvYI3/HbJ1EV9vdujZ0aVCQ/CGBYIkEB0=" parent-hash "ej1WVedyLyUMLiQxzEcrwbY45zCodwV85Kzy7hm2Gv4=" via-port "1-1" with-interface 09:00:00
allow id 0781:5581 serial "A200556A2D62A260" name "Sandisk Ultra" hash "dadGLgJni1BC5k5YZgYmpw+qyzBWnUc33SNGnJiYdU4=" parent-hash "3Wo3XWDgen1hD5xM3PSNl3P98kLp1RUTgGQ5HSxtf8k=" with-interface 08:06:50
allow id 0bc2:ab31 serial "NA7A1LG7" name "Backup+  Desk" hash "WL3F9lswN7Nht9KrSuqQMUgYdKnZ+EExIIe/GPE2Tic=" parent-hash "3Wo3XWDgen1hD5xM3PSNl3P98kLp1RUTgGQ5HSxtf8k=" with-interface { 08:06:50 08:06:62 }
allow id 8087:0024 serial "" name "" hash "kv3v2+rnq9QvYI3/HbJ1EV9vdujZ0aVCQ/CGBYIkEB0=" parent-hash "WHBTxNaEoMGNSNc31KpFNSAeFF4HbLMQgSBqORlC6S8=" via-port "4-1" with-interface 09:00:00
allow id 045e:0745 serial "" name "Microsoft\xc2\xae 2.4GHz Transceiver v7.0" hash "4ftrhDpz2Rl8IHH4y2k+MXzZa5lETJ3IpdGqvRzIlvU=" parent-hash "kv3v2+rnq9QvYI3/HbJ1EV9vdujZ0aVCQ/CGBYIkEB0=" via-port "4-1.5" with-interface { 03:01:01 03:01:02 03:00:00 }
allow id 04f9:01fe serial "BROL8F111953" name "MFC-250C" hash "+yAB2bQPa1dACQoiw7rTu5iiU8/j4/UzArNS+xLPfyo=" parent-hash "kv3v2+rnq9QvYI3/HbJ1EV9vdujZ0aVCQ/CGBYIkEB0=" with-interface { 07:01:02 ff:ff:ff 08:06:50 }
``
With first half only

dkopecek, 2017-06-16:
Use `sudo -i` to get a root shell and execute the command there. You are executing the command as root but then the redirect `>` is executed under your non-root user again.

robin0800, 2017-06-16:
Thanks yes fully working now  ran with no phone connection Phone not seen when connected.
Ran generate again and restarted the daemon phone now seen.

dkopecek, 2017-06-16:
Great! Closing the issue then.

closed 2017-06-16 by dkopecek