USBGuard/usbguard #178: Polkit integration

BenjaminSchubert, 2017-06-18:
Polkit is a standard linux way of managing dbus-access rights.

It is for instance used by Firewalld, systemd, and many other programs to allow users to temporary elevate their privileges if they have the right to do so.

I think it would be great to support this with usbguard and the usbguard-applet, as this would allow sysadmins to allow users to run the applet and view information, but require a sudo password very easily to do modifications as this is supported natively by polkit.

I would be willing to help if this is a feature you'd like to incorporate. I may need a bit of help starting though.

Thank you for this awesome tool !

dkopecek, 2017-06-21:
Hi! The applet work with polkit because it doesn't use the D-Bus interface. The applet was a PoC when it started, so it needs some refactoring or a rewrite from scratch with better UI etc. :-) Third party apps using the usbguard D-Bus interface (provided by the usbguard-dbus service) should already be able to use polkit, we ship a policy as part of the D-Bus component.

 https://github.com/dkopecek/usbguard/blob/master/src/DBus/org.usbguard.policy

There are some other GUI/applet/desktop related issues with the applet such as #157, #152, #116, #60.

If you are willing to help with any of that (incl. the D-Bus support in applet), I would certainly appreciate the contribution.

BenjaminSchubert, 2017-06-22:
Hi,

Yes I would be willing to help for this !

For the applet, what would be the best : a complete rewrite or an update ? What would you prefer ?

If you would prefer from scratch, is there somewhere a list of features that should be in the applet ?

dkopecek, 2017-06-22:
I'm fine with a complete rewrite. I don't have any feature list that the applet or whatever GUI component should stick to. The main feature should be asking users about policy decisions when a USB device get's connected and doesn't match any pre-defined rule in the policy (indicated in the D-Bus/IPC event from the daemon).

Other tham that, you can look at the [open & closed issues related to GUI](https://github.com/dkopecek/usbguard/issues?utf8=%E2%9C%93&q=is%3Aissue%20label%3AGUI%20label%3Aenhancement%20)

BenjaminSchubert, 2017-06-24:
@dkopecek I think splitting the program in two would be better. That would give : 

1) the background notifier, with or without icon
2) the frontend GUI allowing to make changes. This is what would be open by the background notifier when something needs handling.

What do you think of this design ?

dkopecek, 2017-06-24:
> @dkopecek I think splitting the program in two would be better. That would give :
> 
>     the background notifier, with or without icon
>     the frontend GUI allowing to make changes. This is what would be open by the background notifier when something needs handling.
> 
> What do you think of this design ?

Makes sense to me. Would there also be an option to run the GUI manually, view the current device tree + authorization states and make changes to that?

@intrigeri any feedback/proposals here?

BenjaminSchubert, 2017-06-25:
I have a design question, should we require DBUS when using the GUI or should we still allow the IPC method that is now in place ?

Also I've got a problem. I'm not familiar with libtool and autogen, how can I add the qt dbus dependencies (conditionnally or not, depending on the answer above) ?

dkopecek, 2017-06-25:
> I have a design question, should we require DBUS when using the GUI or should we still allow the IPC method that is now in place ?

Let's keep things simple and go with D-Bus only/required now. I think most desktop users run D-Bus and for the server use case, there's the CLI which doesn't require D-Bus. Does that sound reasonable?

> Also I've got a problem. I'm not familiar with libtool and autogen, how can I add the qt dbus dependencies (conditionnally or not, depending on the answer above) ?

It should be enough to add "Qt5DBus >= 5.0" here https://github.com/dkopecek/usbguard/blob/master/configure.ac#L343

Then the required include parameters and libraries will be propagated to the qt_CFLAGS and qt_LIBS makefile variables used here https://github.com/dkopecek/usbguard/blob/master/Makefile.am#L583 and here https://github.com/dkopecek/usbguard/blob/master/Makefile.am#L590

BenjaminSchubert, 2017-06-26:
Also what would you think of using https://developer.gnome.org/notification-spec/ for notifications that blend in the desktop instead of that yellow ? 

We could also make this optional, but I think that graphically it's a big plus. From what I know, KDE, Gnome and XFCE should be good with this, no idea about the others though. Should we make this configurable ? If so, we could add a flag at configuration to add this. What would you prefer ?

And finally just to be sure, are you ok by discussing all of this here, or should we take it somewhere else ?

dkopecek, 2017-06-26:
> Also what would you think of using https://developer.gnome.org/notification-spec/ for notifications that blend in the desktop instead of that yellow ?
>
> We could also make this optional, but I think [...]

Yeah, I guess following the notification spec is a better approach, so +1 here. I thought that Qt is able to transparently use both systems (the spec and their own).

> And finally just to be sure, are you ok by discussing all of this here, or should we take it somewhere else ?

It's ok. I hope others will join the discussion as well, as I'm no UI designer :D

BenjaminSchubert, 2017-06-26:
> Yeah, I guess following the notification spec is a better approach, so +1 here. I thought that Qt is able to transparently use both systems (the spec and their own).

There is https://github.com/KDE/snorenotify that allows for cross-platform notification. I never used it though, I'm quite new to QT and C++. What do you prefer ?

dkopecek, 2017-06-26:
> There is https://github.com/KDE/snorenotify that allows for cross-platform notification. I never used it though, I'm quite new to QT and C++. What do you prefer ?

That doesn't seem to be packaged for Fedora/RHEL/CentOS and looks quite heavy duty just to support simple notifications.

The desktop notification specification is provided as a D-Bus interface, isn't it? So it should suffice just to have a Qt DBus client code for that.

BenjaminSchubert, 2017-06-26:
Yeah, but not every one might not have an implementation of this protocol. This should not be the case on mainstream distos though

genodeftest, 2017-08-19:
As a partially related side-note: GNOME [is planning to remove system tray support from gnome-shell](https://bugzilla.gnome.org/show_bug.cgi?id=721596#c72). If possible, please keep that in mind.

genodeftest, 2017-08-19:
> As a partially related side-note: GNOME [is planning to remove system tray support from gnome-shell](https://bugzilla.gnome.org/show_bug.cgi?id=721596#c72). If possible, please keep that in mind.

Oh, I just found that this is in #157 already. Sorry for the noise.

closed 2019-03-11 by dkopecek