USBGuard/usbguard #184: Unable to build the QT applet

BenjaminSchubert, 2017-06-24:
I've been trying to build manually the QT applet for development.

However I stumbled upon multiple errors. I would need help on this : 

Steps to reproduce : 

- git clone https://github.com/dkopecek/usbguard
- cd usbguard/srv/GUI.QT
- qmake-qt5
- make

I get the following errors (clearing them makes me go to the next step) :

- icon.qrc is not found => remove the line in QtApplet.pro
- Cannot find <Rule.hpp> => adding `INCLUDEPATH += $$PWD/../Library` in QtApplet.pro fixes it
- cannot find "MainWindow.ui.h". This one bugs me, as I have ui_MainWindow.h available in the path ... maybe change ? If so I get : 
- cannot find "DeviceDialog.ui.h" not found, as above
- `build-config.h` not found which I cannot even find in the whole repo.

Could someone help me with this ? I did not submit a PR for the first problems as I'm not sure they are the best fixes.

Thank you !

dkopecek, 2017-06-24:
> I get the following errors (clearing them makes me go to the next step) :

Many of these errors stem from the fact that the missing files are generated at build time by the configure and Makefile scripts, see my inline comments. Also, the QtApplet.pro file should not be there as I don't maintain it anymore...

>     icon.qrc is not found => remove the line in QtApplet.pro

Generated here: https://github.com/dkopecek/usbguard/blob/master/Makefile.am#L614

>     cannot find "MainWindow.ui.h". This one bugs me, as I have ui_MainWindow.h available in the path ... maybe change ? If so I get :
>     cannot find "DeviceDialog.ui.h" not found, as above

Unfortunately, the Qt tools (AFAIK) doesn't allow to specify how the generated files should be named. And since the automake based build system doesn't use qmake to generate these files, the file names are different. If the files had ui_ prefix, I would not be able to write the Makefile patterns and correctly handle the various qt-related file generation. The files .ui.h are generate here:

https://github.com/dkopecek/usbguard/blob/master/Makefile.am#L599

>     build-config.h not found which I cannot even find in the whole repo.

This one is generated by the configure script:

https://github.com/dkopecek/usbguard/blob/master/configure.ac#L6

I think you can safely ignore it during the development time unless you depend on results of some of the configure script checks.

BenjaminSchubert, 2017-06-24:
Oh ok, so the .pro should be removed and the configure used right ? This makes sense, thanks !

dkopecek, 2017-06-24:
> Oh ok, so the .pro should be removed and the configure used right ? This makes sense, thanks !

Yes, I'll take care of removing it as I'm currently doing some code cleanups.

dkopecek, 2017-06-28:
The file was removed, closing.

closed 2017-06-28 by dkopecek