USBGuard/usbguard #185: Style guide specification

BenjaminSchubert, 2017-06-29:
To help contribution, a style guide specification, either through a linter configuration (and a Make target) or through a clang-format file would help have a consistent coding style.

Would it be possible to have such a thing ? I can help with this, if you tell me what style you'd like to follow.

Thank you very much !

dkopecek, 2017-06-29:
Yes, this is a good idea. I tried clang-format, but it doesn't support the if-else brace formatting I currently use among other things. However, if we find a set of clang-format parameters that result in as little changes as possible, I would go for it. One of the benefits would be that a clang-format based check could be implemented in the CI to check all future PRs.

BenjaminSchubert, 2017-06-30:
@dkopecek I started trying to make a .clang-format file.

I however have a few questions as the style is not consistent on all the project : 

- Concerning pointers and references (* and &), should it be : `int* test` or `int *test` ?
- Concerning `else` statements, should it be 
```
} else {

or 

}
else {
```
- same question about `catch` statements
- Do you want `case` statements to be indented compared to the `switch` statement ?
- Do you want to enforce a maximum line length ?

I could not infere these options from the source as both  were used quite equally.

dkopecek, 2017-06-30:
> I however have a few questions as the style is not consistent on all the project :
> 
>     Concerning pointers and references (* and &), should it be : int* test or int *test ?

```
int* test
```
>     Concerning else statements, should it be
> 
> } else {
> 
> or 
> 
> }
> else {

This:
```
if () {
}
else if () {
}
else {
}
```
> 
>     same question about catch statements
```
try {
  ...
}
catch(foo& x) {
}
catch(...) {
}
```
>     Do you want case statements to be indented compared to the switch statement ?

Here I don't have a strong preference, but I'd go probably with:
```
switch(...) {
case 1:
   break;
case 2:
   break;
default:
   foo();
}
```
>     Do you want to enforce a maximum line length ?

Hmm, I don't really care about the line length unless something is too long... but if we have to choose an arbitrary number, then let's use something over 100. 128 is a nice number :-)

BenjaminSchubert, 2017-06-30:
Ok, I incorporated the changes in the format.

I tried reformatting the files to see what that gives, and I have  `137 files changed, 4488 insertions(+), 4622 deletions(-)` as a result. I don't know what you think about this.

If you want to try, here is the content of the `.clang-format` file : 

```
---
Language:        Cpp
# BasedOnStyle:  LLVM
AccessModifierOffset: -2
AlignAfterOpenBracket: Align
AlignConsecutiveAssignments: false
AlignConsecutiveDeclarations: false
AlignEscapedNewlinesLeft: false
AlignOperands:   true
AlignTrailingComments: true
AllowAllParametersOfDeclarationOnNextLine: false
AllowShortBlocksOnASingleLine: false
AllowShortCaseLabelsOnASingleLine: false
AllowShortFunctionsOnASingleLine: false
AllowShortIfStatementsOnASingleLine: false
AllowShortLoopsOnASingleLine: false
AlwaysBreakAfterReturnType: None
AlwaysBreakBeforeMultilineStrings: false
AlwaysBreakTemplateDeclarations: false
BinPackArguments: false
BinPackParameters: false
BraceWrapping:   
  AfterClass:      true
  AfterControlStatement: false
  AfterEnum:       true
  AfterFunction:   true
  AfterNamespace:  true
  AfterObjCDeclaration: false
  AfterStruct:     false
  AfterUnion:      false
  BeforeCatch:     true
  BeforeElse:      true
  IndentBraces:    false
BreakBeforeBinaryOperators: None
BreakBeforeBraces: Custom
BreakBeforeTernaryOperators: true
BreakConstructorInitializersBeforeComma: true
BreakAfterJavaFieldAnnotations: false
BreakStringLiterals: true
ColumnLimit:     120
CommentPragmas:  '^ IWYU pragma:'
ConstructorInitializerAllOnOneLineOrOnePerLine: false
ConstructorInitializerIndentWidth: 4
ContinuationIndentWidth: 4
Cpp11BracedListStyle: true
DerivePointerAlignment: false
DisableFormat:   false
ExperimentalAutoDetectBinPacking: false
ForEachMacros:   [ foreach, Q_FOREACH, BOOST_FOREACH ]
IncludeCategories: 
  - Regex:           '^"(llvm|llvm-c|clang|clang-c)/'
    Priority:        2
  - Regex:           '^(<|"(gtest|isl|json)/)'
    Priority:        3
  - Regex:           '.*'
    Priority:        1
IncludeIsMainRegex: '$'
IndentCaseLabels: false
IndentWidth:     2
IndentWrappedFunctionNames: false
KeepEmptyLinesAtTheStartOfBlocks: true
MacroBlockBegin: ''
MacroBlockEnd:   ''
MaxEmptyLinesToKeep: 1
NamespaceIndentation: All
ObjCBlockIndentWidth: 2
ObjCSpaceAfterProperty: false
ObjCSpaceBeforeProtocolList: true
PenaltyBreakBeforeFirstCallParameter: 19
PenaltyBreakComment: 300
PenaltyBreakFirstLessLess: 120
PenaltyBreakString: 1000
PenaltyExcessCharacter: 1000000
PenaltyReturnTypeOnItsOwnLine: 60
PointerAlignment: Left
ReflowComments:  true
SortIncludes:    true
SpaceAfterCStyleCast: false
SpaceAfterTemplateKeyword: true
SpaceBeforeAssignmentOperators: true
SpaceBeforeParens: ControlStatements
SpaceInEmptyParentheses: false
SpacesBeforeTrailingComments: 1
SpacesInAngles:  false
SpacesInContainerLiterals: true
SpacesInCStyleCastParentheses: false
SpacesInParentheses: false
SpacesInSquareBrackets: false
Standard:        Cpp11
TabWidth:        8
UseTab:          Never
...
```

Reformatting all files can be done by `find src -iname "*.cpp" -o -iname "*.hpp" | xargs clang-format -i --style=file` if you want to see what difference your changes make.

I don't know what is best for the follow up, as this is quite a change in the projet. Might be best to enforce only on new files/code. That's your decision :) I'm happy to help if I can help you more with this !

dkopecek, 2017-06-30:
I guess I need a newer version of clang for this, the following is not supported by clang 3.9.1 (Fedora 25):
```
YAML:79:28: error: unknown key 'SpaceAfterTemplateKeyword'
SpaceAfterTemplateKeyword: true
                           ^~~~
```
I've commented that out and run the reformat without that setting.

I think there are several things to adjust to make the diff smaller:

1. Don't change indent for the member initializers
```
   PolicyGenerator::PolicyGenerator()
-   : _ruleset(nullptr)
+      : _ruleset(nullptr)
```

2. Don't reorder #include statements
```
-#include "usbguard/Rule.hpp"
-#include "usbguard/RuleSet.hpp"
 #include "usbguard/DeviceManager.hpp"
 #include "usbguard/DeviceManagerHooks.hpp"
+#include "usbguard/Rule.hpp"
+#include "usbguard/RuleSet.hpp"
```

3. Don't do this:
```
-  static const struct ::option options_long[] = {
-    { "help", no_argument, nullptr, 'h' },
-    { "user", no_argument, nullptr, 'u' },
-    { "group", no_argument, nullptr, 'g' },
-    { "policy", required_argument, nullptr, 'p' },
-    { "devices", required_argument, nullptr, 'd' },
-    { "exceptions", required_argument, nullptr, 'e' },
-    { "parameters", required_argument, nullptr, 'P' },
-    { "no-root-check", no_argument, nullptr, 'N' },
-    { nullptr, 0, nullptr, 0 }
-  };
+  static const struct ::option options_long[] = {{"help", no_argument, nullptr, 'h'},
+                                                 {"user", no_argument, nullptr, 'u'},
+                                                 {"group", no_argument, nullptr, 'g'},
+                                                 {"policy", required_argument, nullptr, 'p'},
+                                                 {"devices", required_argument, nullptr, 'd'},
+                                                 {"exceptions", required_argument, nullptr, 'e'},
+                                                 {"parameters", required_argument, nullptr, 'P'},
+                                                 {"no-root-check", no_argument, nullptr, 'N'},
+                                                 {nullptr, 0, nullptr, 0}};

```

4. And don't do this:
```
...
-    stream << " Usage: " << usbguard_arg0
-           << " [OPTIONS] <command> [COMMAND OPTIONS] ..." << std::endl;
+    stream << " Usage: " << usbguard_arg0 << " [OPTIONS] <command> [COMMAND OPTIONS] ..." << std::endl;
     stream << std::endl;
     stream << " Options:" << std::endl;
     stream << "" << std::endl;
     stream << " Commands:" << std::endl;
-    stream << "  get-parameter <name>           Get the value of a runtime parameter." << std::endl;
-    stream << "  set-parameter <name> <value>   Set the value of a runtime parameter." << std::endl;
-    stream << "  list-devices                   List all USB devices recognized by the USBGuard daemon." << std::endl;
-    stream << "  allow-device <id>              Authorize a device to interact with the system." << std::endl;
+    stream << "  get-parameter <name>           Get the value of a runtime "
+              "parameter."
+           << std::endl;
+    stream << "  set-parameter <name> <value>   Set the value of a runtime "
+              "parameter."
+           << std::endl;
...
```

dkopecek, 2017-06-30:
5. This is also unwanted. With the long function names, there's then too much empty space after the alignment:
```
-  catch(...) {
-    g_dbus_method_invocation_return_error(invocation, G_DBUS_ERROR,
-       G_DBUS_ERROR_FAILED, "BUG: Unknown exception; method call failed for unknown reasons.");
+  catch (...) {
+    g_dbus_method_invocation_return_error(invocation,
+                                          G_DBUS_ERROR,
+                                          G_DBUS_ERROR_FAILED,
+                                          "BUG: Unknown exception; method call failed for unknown reasons.");

```

BenjaminSchubert, 2017-07-04:
After changing 1 and 2, I get 200 lines changed less. 

However, it seems like clang-format is unable to handle the three last correctly.

Should we remove some of the formatting and use only a minimal styleguide ?

What do you think ?

dkopecek, 2017-07-04:
> After changing 1 and 2, I get 200 lines changed less.
> 
> However, it seems like clang-format is unable to handle the three last correctly.

Are the any options left? For example, if the column limit would be higher, would the string literals in 4. get still reformatted?

Or for the case 5., are there any other formatting options available? I don't mind that there's a change in that case, but the change is making the code uglier (IMHO). If there is a setting that leads to a different kind of change, then that might be acceptable.

> Should we remove some of the formatting and use only a minimal styleguide ?
> What do you think ?

Well, if there aren't any other options then this one is the only way forward, I guess. The drawback is that it won't be possible to check the formatting automatically.

dkopecek, 2017-07-04:
We could also try whether there would reasonable results with [AStyle](http://astyle.sourceforge.net/).

dkopecek, 2017-07-04:
Looks like astyle in it's newest version (svn trunk) should be usable. I'll try later whether the olders versions give similar results.

Here's the current iteration of the options / astylerc that I'm working with:
```
--style=kr
--indent=spaces=2
--indent-namespaces
--indent-after-parens
--indent-labels
--indent-preproc-define
--indent-col1-comments
--break-blocks
--pad-comma
--pad-header
--delete-empty-lines
--align-pointer=type
--align-reference=type
--break-closing-braces
--break-one-line-headers
--break-blocks
--add-braces
--convert-tabs
--close-templates
--max-code-length=128
```

dkopecek, 2017-07-04:
astyle 3.0.1, the latest stable release, is usable.
astyle 2.05 is too old and therefore doesn't recognize some of the options.

dkopecek, 2017-07-04:
FIY working on the change in a private branch now:
https://github.com/dkopecek/usbguard/tree/private-astyle-reformat

Let me know if you have any suggestions.

BenjaminSchubert, 2017-07-05:
@dkopecek Astyle seems nice indeed, but the only problem I see with this is that it wouldn't be recognized by text editors right ?

I know vim, vscode and CLion have great support for clang-format, which makes a big difference in my opinion. What do you think ?

But if Astyle works, it's far better than nothing :)

dkopecek, 2017-07-10:
> @dkopecek Astyle seems nice indeed, but the only problem I see with this is that it wouldn't be recognized by text editors right ?
>
> I know vim, vscode and CLion have great support for clang-format, which makes a big difference in my opinion. What do you think ?

I would hope that AStyle is also supported because it seems to be a much older project, there's some information on the astyle project website:

 http://astyle.sourceforge.net/links.html

and for vim, there's https://github.com/Chiel92/vim-autoformat

commit 41616a22 (2017-10-27): AStyle source code reformat

 - defined coding style via AStyle configuration in src/astylerc
   (requires astyle version 3+)
 - added convenience scripts for reformatting the sources
   (scripts/reformat-sources.sh)
 - reformated sources
 - added code-style.sh check to the test suite

Resolves: #185
commit 6b7863ae (2017-10-27): AStyle source code reformat

 - defined coding style via AStyle configuration in src/astylerc
   (requires astyle version 3+)
 - added convenience scripts for reformatting the sources
   (scripts/reformat-sources.sh)
 - reformated sources
 - added code-style.sh check to the test suite

Resolves: #185
closed 2017-10-27 by dkopecek