USBGuard/usbguard #195: Certain Hub not working as expected

AHeinlein, 2017-07-18:
I have written a few rules which should allow all USB mass storage, Hub and HID automatically. This works, except for one single hub which triggers an allow/block dialog. I can't find any meaningful difference between this hub and any other. Am I overlooking something or is this a bug?
I am (still) using 0.6.1 since that is what is in Debian 9.

[working.txt](https://github.com/dkopecek/usbguard/files/1154897/working.txt)
[not-working.txt](https://github.com/dkopecek/usbguard/files/1154898/not-working.txt)
[rules.conf.txt](https://github.com/dkopecek/usbguard/files/1154899/rules.conf.txt)

dkopecek, 2017-07-18:
Hi, I can't access the files you've attached:
```
Error><Code>SignatureDoesNotMatch</Code><Message>The request signature we calculated does not match the signature you provided. Check your key and signing method.</Message>
```
Might be some github issue...

AHeinlein, 2017-07-18:
Google says this error message is from Amazon AWS...
Anyway, try this:
https://pastebin.com/yvnEfm63 (working example)
https://pastebin.com/j0DAE2LR (non-working hub)
https://pastebin.com/4inCwFg1 (rules.conf)

dkopecek, 2017-07-18:
Could you please also attach debugging output from usbguard-daemon (for the non-working case)? debugging messages can be enabled by passing the -d option to usbguard-daemon.

Also, since you are using an older version, would it be possible for you to compile the latest version and try whether you can reproduce the issue?

AHeinlein, 2017-07-19:
Problem occurs with 0.7.0 as well. Debug Log is here: https://pastebin.com/cUxsSFgJ