USBGuard/usbguard #197: Usbguard stops/sleeps my mouse

kadinho, 2017-07-21:
Hi,
Fedora 26, when i leave the mouse to type on the keyboard the mouse stops or a kindof energy cut ocurrs then i need to move or click on the mouse then after 2 or 3 seconds the mouse start to respond again. 
The tools is great but this issue is annoying , and i think i will remove the tool :(

any help

Thx
CV

genodeftest, 2017-08-19:
> Fedora 26, when i leave the mouse to type on the keyboard the mouse stops or a kindof energy cut ocurrs then i need to move or click on the mouse then after 2 or 3 seconds the mouse start to respond again.
The tools is great but this issue is annoying , and i think i will remove the tool :(

This sounds unrelated to usbguard. Did you try stopping the usbguard daemon and see if it still happens?
Your problem sounds more like broke ACPI settings or missing ACPI compatibility. I have seen this sometimes when playing with `powertop`.

arthurlogilab, 2018-12-04:
I'm having something similar to this. A keyboard and a mouse that work, then don't, then come back. 

Can't find which log to tail to see what's wrong. Would it be possible that there are non-permanent rules at play, or a different set of rules between my user and root ?

genodeftest, 2018-12-04:
@arthurlogilab I would recommend using powertop to disable (!) all the power saving features temporarily to see whether the problem will disappear. I am quite sure USBGuard is not related to your problem at all (except that it also affects the USB stack). Instead, I think your issue is caused by overly aggressive power saving features or incompatible (bad quality) hardware/firmware.