USBGuard/usbguard #198: [RFE] Provide a policy to allow only encrypted USB devices

dpal, 2017-08-07:
Flash drives are widely used to share information between computers. In high security environments the use of the USB flash drives is not allowed because the data can be stolen. Network bound disk encryption allows data to be bound to network. So if a device can be encrypted and used only on the network then the data can't be stolen since it is encrypted. There is currently no integration with network bound disk encryption. The requested feature is to enhance USB guard to have a policy to allow only NBDE encrypted drives to enforce that only encrypted USB devices can be used in the high security environments and data is not leaked or stolen.

tammar96, 2017-09-07:
related: #42

tammar96, 2017-09-07:
related: #42

closed 2019-03-11 by dkopecek