USBGuard/usbguard #204: enumeration timeout with linux 4.13

BlackIkeEagle, 2017-09-06:
ERROR: UEventDeviceManager: present devices: enumeration timeout

Could this have something to do with the kernel config?

https://github.com/herecura/linux-bede/blob/herecura/config-desktop.x86_64

dkopecek, 2017-09-07:
Hi and thanks for the bug report. I'm aware of this and working on a fix (different device enumeration procedure) It should be available in the 0.7.1 release.

BlackIkeEagle, 2017-09-07:
superb, thanks.

dkopecek, 2017-09-12:
@BlackIkeEagle I'm working on a fix in the private-kernel-4.13 branch. Would you have time to checkout the branch and check whether it fixes the problem for you? What distro are you using? I can provide testing builds for Fedora/CentOS/RHEL distros.

BlackIkeEagle, 2017-09-12:
thanks @dkopecek , I'll test later, I'm on Arch Linux

BlackIkeEagle, 2017-09-12:
@dkopecek first test: applying the patch on top of 0.7.0 fixes `usbguard generate-policy` while using linux 4.13.1

BlackIkeEagle, 2017-09-12:
@dkopecek and for completeness I have also tested the branch and it also works there for me

dkopecek, 2017-09-12:
@BlackIkeEagle Thanks a lot for testing!

BlackIkeEagle, 2017-09-12:
@dkopecek as a bonus, just tested with linux 4.9.49 and the behaviour is the same

closed 2017-09-13 by dkopecek

genodeftest, 2017-10-02:
Thanks for the fix!

Any chance this bug fix can land in a bugfix release? I just ran into the same issue on kernel 4.13.4, which has landed in Fedora's updates-testing repo.

On Fedora, I had to change permissions on /etc/usbguard/usbguard-daemon.conf and /etc/usbguard/rules.conf using `chmod og-r` before usbguard could run again. I'm using the builds from https://copr.fedorainfracloud.org/coprs/mildew/usbguard-testing/.

albocc, 2017-10-05:
@dkopecek As @genodeftest already said, this definitely needs a bugfix release, so people can easily get the update via the Fedora repos. I just encountered this bug today when 4.13.4 rolled out to Fedora's stable repo (F26).

commit 50cb9cd3 (2017-09-12): Make UEventDeviceManager work on Linux Kernel >= 4.13

It seems that we can't rely on the uevent data to contain the
DEVTYPE value. Change the UEventDeviceManager to not rely on the
presence of the DEVTYPE attribute value and instead check the
uevent file in the device's sysfs directory.

Resolves: #204
commit c305bcec (2017-09-13): Make UEventDeviceManager work on Linux Kernel >= 4.13

It seems that we can't rely on the uevent data to contain the
DEVTYPE value. Change the UEventDeviceManager to not rely on the
presence of the DEVTYPE attribute value and instead check the
uevent file in the device's sysfs directory.

Resolves: #204
commit a3cdd13c (2017-09-13): Make UEventDeviceManager work on Linux Kernel >= 4.13

It seems that we can't rely on the uevent data to contain the
DEVTYPE value. Change the UEventDeviceManager to not rely on the
presence of the DEVTYPE attribute value and instead check the
uevent file in the device's sysfs directory.

Resolves: #204