USBGuard/usbguard #205: append-rule syntax RuleParserError

dx486, 2017-09-08:
I want to allow some USB devices but I cannot use append-rule option.

What is the correct way of using "append-rule"?

I receive the following error (replaced some chars with "9"):

```
$ sudo usbguard append-rule 'id 9999:9999 serial "" name "USB3.0 Hub             " hash "999999999999999999999999999999999999999ug3Q="'
IPC ERROR: request id=1: IPC method: usbguard.IPC.appendRule: RuleParserError

```

dkopecek, 2017-09-08:
The rule is missing the `target` part: allow, block or reject. It should be specified as the first thing in a rule and is actually the only part that is not optional.

So given your example:

$ sudo usbguard append-rule 'id 9999:9999 serial "" name "USB3.0 Hub             " hash "999999999999999999999999999999999999999ug3Q="'

Change that to:

$ sudo usbguard append-rule '**allow** id 9999:9999 serial "" name "USB3.0 Hub             " hash "999999999999999999999999999999999999999ug3Q="'

(assuming you want to allow the device)

dx486, 2017-09-08:
Oh... How I have missed that... Thank you very much for your quick reply & solution.

closed 2017-09-08 by dx486