USBGuard/usbguard #207: Rule matching: multiple interfaces with same class

AHeinlein, 2017-10-25:
I have an external USB hard drive which supports USB attached SCSI as well as "normal" USB bulk transfers. So it has two interfaces, 08:06:50 and 08:06:62.

I thought that the rule given in the examples would match:
`allow with-interface equals { 08:*:* }`
but it doesn't.
This, however, works:
`allow with-interface one-of { 08:*:* }`
I don't know if this is intended behaviour, but it is at least confusing.

alasley, 2017-11-30:
I have experienced this issue as well, on my first day of experimenting with usbguard.

This does appear to be the intended behaviour of that example rule, however it is probably not in the spirit of what it is supposed to do. I imagine devices with dual UASP and legacy bulk storage capability are fairly common these days with USB 3.0/3.1 devices. So I figure this is really just a documentation bug.

I don't think the `one-of` operator is sufficient to forbid the "suspicious combination" devices. I added the rule `allow with-interface equals { 08:06:50 08:06:62 }` to allow my device, but this might be too specific.

thedrow, 2018-12-17:
The documentation specified that `allow with-interface equals { 08:*:* }` should allow all USB devices.
If that doesn't work, I think this is a bug.

radosroka, 2019-08-28:
The behavior is expected. We will need to implement a new operator like "match-all" that will try to match each device interface against provided mask.

```
allow with-interface match-all { 08:*:* }
# will match device with all interfaces started with "08" -> { 08:06:50 08:06:62 }

allow with-interface match-all { 03:*:* }
# will match all HID
# it should not match { 03:00:01 01:00:01 }
# but
allow with-interface match-all { 03:*:* 01:*:* }
# will match.
```
Such keyword would enhance class based rules to the new level. 

@dkopecek what do you think?