USBGuard/usbguard #214: Add device class specification in rules

MatejKovacic, 2017-12-13:
In rules you can specify if you allow or block devices regarding to their class. USBGuard is using "with-interface" command.

However, classes are defined by codes, which is not really human readable. Is it possible to add "human description" for classes?
For example:
- `allow with-interface equals { "mass-storage" }`
- `allow with-interface one-of { "video" "printer"}`

List of classes are here: https://en.wikipedia.org/wiki/USB#Device_classes

BTW, maybe another proposal. There should be distinction between:
`allow with-interface equals { "mass-storage" }` and `allow with-interface equals only { "mass-storage" }`.

dkopecek, 2017-12-14:
Hi, the class number is only one of three numbers used in the USB interface type specification. Then there's also subclass and protocol. I was thinking of a more generic approach to this problem using tags, see: https://github.com/USBGuard/usbguard/issues/20.

dkopecek, 2018-01-22:
Closing as duplicate of #20. Also related #12.

closed 2018-01-22 by dkopecek