USBGuard/usbguard #217: Luks encrypted USB device hash changes when  moved from one parent port to another

jean-christophe-manciot, 2017-12-27:
Ubuntu 17.10
usbguard 0.7.1-2
libgcrypt20 1.8.1-4

Usbguard has been configured from sources with:
```
./configure  --build=x86_64-pc-linux-gnu \
             --enable-systemd \
             --with-crypto-library=gcrypt \
             --prefix=/usr --sysconfdir=/etc --localstatedir=/var
```
 
When I move a specific USB WD HDD from one USB 3.0 port <--> USB 2.0 port, its hash changes which makes its blocked at the reboot following the port change.
If I don't move it, the hash remains constant.
When I perform the same port change for other USB devices, namely SAMSUNG SSDs, there is no hash change.
The **differences between the HDD & the SSDs** are:
- the brand: WD vs SAMSUNG
- the presence of a NTFS partition only on the WD HDD
- the presence of multiple luks encrypted partitions only on WD HDD.

I'm guessing the luks partitions & the USB port change confuse usbguard.

dkopecek, 2018-01-09:
Hi,
 could you please check what the value of "bcdUSB" field is for both devices and both ports? You can check the value using the `lsusb -v` command. The bcdUSB value is included when computing the device hash. If the device announces a different value depending on the port which it is connected to, then the hash will change unfortunately.

jean-christophe-manciot, 2018-01-09:
1. **First setup**

SAMSUNG SSD on USB 3.0:
```
Bus 004 Device 002: ID 174c:55aa
...
  bcdUSB               3.00
```
WD on USB 2.0:
```
Bus 003 Device 002: ID 1058:0827. 
...
  bcdUSB               2.10
```

2. **Second setup**

SAMSUNG SSD on USB 2.0:
```
Bus 003 Device 002: ID 174c:55aa
...
  bcdUSB               2.10
```
WD on USB 3.0:
```
Bus 004 Device 002: ID 1058:0827. 
...
  bcdUSB               3.00
```

As you can see, I switched SAMSUNG SSD with WD.
For both devices, the **bcdUSB** parameter has changed symmetrically.
- SAMSUNG SSD from USB 3.0 to 2.0: bcdUSB 3.00 --> 2.10
- WD from USB 2.0 to 3.0: bcdUSB 2.10 --> 3.00

But the hash does not change symmetrically:
- SAMSUNG SSD from 1z...MM= --> 1z...MM=
- WD from bX...CY= --> OU...5A=

As you can, the hash does not change for the SAMSUNG despite the bcdUSB change, when the hash does change for the WD.

Anyhow, it's not a big deal because putting both WD hashes in /etc/usbguard/rules.conf allows to recognize the WD device when switching ports between reboots.

dkopecek, 2018-01-22:
Could you check whether any other attributes change (in the lsusb output) when switching the same device between USB 2.0 and 3.0 ports?

jean-christophe-manciot, 2018-01-24:
The following attributes change or appear or disappear when switching from USB 2 to 3:
**SAMSUNG**:
```
  bcdUSB
  bMaxPacketSize0     
    wTotalLength          
        wMaxPacketSize
        bMaxBurst
        MaxStreams
Device Status:    
  U1 
  U2 
```
**Western Digital**:
```
  bcdUSB
  bMaxPacketSize0
    wTotalLength
    MaxPower
        wMaxPacketSize
        bMaxBurst
Device Status
  U1
  U2
```