USBGuard/usbguard #220: No service on Debian

realListigerLurch, 2018-01-24:
When running 
`# systemctl start usbguard` after the installation process, I get the error
`Failed to start usbguard.service: Unit usbguard.service not found.`

The content of /usr/local/share/dbus-1/system-services/org.usbguard.service is:
`[D-BUS Service]
Name=org.usbguard
Exec=/usr/local/sbin/usbguard-dbus --system
SystemdService=dbus-org.usbguard.service`

The output of '# make install' was:
`root@test-xx:/home/xx/Downloads/usbguard# make install
Makefile:3419: warning: overriding recipe for target 'check'
Makefile:3051: warning: ignoring old recipe for target 'check'
make  install-recursive
make[1]: Entering directory '/home/xx/Downloads/usbguard'
Makefile:3419: warning: overriding recipe for target 'check'
Makefile:3051: warning: ignoring old recipe for target 'check'
Making install in src/Tests/
make[2]: Entering directory '/home/xx/Downloads/usbguard/src/Tests'
make[3]: Entering directory '/home/xx/Downloads/usbguard/src/Tests'
make[3]: Nothing to be done for 'install-exec-am'.
make[3]: Nothing to be done for 'install-data-am'.
make[3]: Leaving directory '/home/xx/Downloads/usbguard/src/Tests'
make[2]: Leaving directory '/home/xx/Downloads/usbguard/src/Tests'
make[2]: Entering directory '/home/xx/Downloads/usbguard'
Makefile:3419: warning: overriding recipe for target 'check'
Makefile:3051: warning: ignoring old recipe for target 'check'
make[3]: Entering directory '/home/xx/Downloads/usbguard'
Makefile:3419: warning: overriding recipe for target 'check'
Makefile:3051: warning: ignoring old recipe for target 'check'
 /bin/mkdir -p '/usr/local/lib'
 /bin/bash ./libtool   --mode=install /usr/bin/install -c   libusbguard.la '/usr/local/lib'
libtool: install: /usr/bin/install -c .libs/libusbguard.so.0.0.0 /usr/local/lib/libusbguard.so.0.0.0
libtool: install: (cd /usr/local/lib && { ln -s -f libusbguard.so.0.0.0 libusbguard.so.0 || { rm -f libusbguard.so.0 && ln -s libusbguard.so.0.0.0 libusbguard.so.0; }; })
libtool: install: (cd /usr/local/lib && { ln -s -f libusbguard.so.0.0.0 libusbguard.so || { rm -f libusbguard.so && ln -s libusbguard.so.0.0.0 libusbguard.so; }; })
libtool: install: /usr/bin/install -c .libs/libusbguard.lai /usr/local/lib/libusbguard.la
libtool: install: /usr/bin/install -c .libs/libusbguard.a /usr/local/lib/libusbguard.a
libtool: install: chmod 644 /usr/local/lib/libusbguard.a
libtool: install: ranlib /usr/local/lib/libusbguard.a
libtool: finish: PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/sbin" ldconfig -n /usr/local/lib
----------------------------------------------------------------------
Libraries have been installed in:
   /usr/local/lib

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the '-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the 'LD_LIBRARY_PATH' environment variable
     during execution
   - add LIBDIR to the 'LD_RUN_PATH' environment variable
     during linking
   - use the '-Wl,-rpath -Wl,LIBDIR' linker flag
   - have your system administrator add LIBDIR to '/etc/ld.so.conf'

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
----------------------------------------------------------------------
 /bin/mkdir -p '/usr/local/bin'
  /bin/bash ./libtool   --mode=install /usr/bin/install -c usbguard usbguard-rule-parser '/usr/local/bin'
libtool: install: /usr/bin/install -c .libs/usbguard /usr/local/bin/usbguard
libtool: install: /usr/bin/install -c .libs/usbguard-rule-parser /usr/local/bin/usbguard-rule-parser
 /bin/mkdir -p '/usr/local/sbin'
  /bin/bash ./libtool   --mode=install /usr/bin/install -c usbguard-daemon usbguard-dbus '/usr/local/sbin'
libtool: install: /usr/bin/install -c .libs/usbguard-daemon /usr/local/sbin/usbguard-daemon
libtool: install: /usr/bin/install -c .libs/usbguard-dbus /usr/local/sbin/usbguard-dbus
 /bin/mkdir -p '/usr/local/share/bash-completion/completions'
 /usr/bin/install -c -m 644 ./scripts/bash_completion/usbguard '/usr/local/share/bash-completion/completions'
 /bin/mkdir -p '/usr/local/share/man/man1'
 /usr/bin/install -c -m 644 ./doc/man/usbguard.1 '/usr/local/share/man/man1'
 /bin/mkdir -p '/usr/local/share/man/man5'
 /usr/bin/install -c -m 644 ./doc/man/usbguard-daemon.conf.5 ./doc/man/usbguard-rules.conf.5 '/usr/local/share/man/man5'
 /bin/mkdir -p '/usr/local/share/man/man8'
 /usr/bin/install -c -m 644 ./doc/man/usbguard-daemon.8 ./doc/man/usbguard-dbus.8 '/usr/local/share/man/man8'
 /bin/mkdir -p '/usr/local/lib/pkgconfig'
 /usr/bin/install -c -m 644 libusbguard.pc '/usr/local/lib/pkgconfig'
 /bin/mkdir -p '/usr/local/include/usbguard'
 /usr/bin/install -c -m 644 src/Library/public/usbguard/Audit.hpp src/Library/public/usbguard/ConfigFile.hpp src/Library/public/usbguard/Device.hpp src/Library/public/usbguard/DeviceManager.hpp src/Library/public/usbguard/DeviceManagerHooks.hpp src/Library/public/usbguard/Exception.hpp src/Library/public/usbguard/IPCClient.hpp src/Library/public/usbguard/IPCServer.hpp src/Library/public/usbguard/Interface.hpp src/Library/public/usbguard/Logger.hpp src/Library/public/usbguard/Policy.hpp src/Library/public/usbguard/Predicates.hpp src/Library/public/usbguard/Rule.hpp src/Library/public/usbguard/RuleCondition.hpp src/Library/public/usbguard/RuleSet.hpp src/Library/public/usbguard/Typedefs.hpp src/Library/public/usbguard/USB.hpp src/Library/public/usbguard/USBGuard.hpp '/usr/local/include/usbguard'
make  install-data-hook
make[4]: Entering directory '/home/xx/Downloads/usbguard'
Makefile:3419: warning: overriding recipe for target 'check'
Makefile:3051: warning: ignoring old recipe for target 'check'
/bin/mkdir -p //usr/local/etc/usbguard
/bin/mkdir -p //usr/local/etc/usbguard/IPCAccessControl.d
/bin/mkdir -p //usr/local/var/log/usbguard
/usr/bin/install -c -m 600 ./usbguard-daemon.conf \
  //usr/local/etc/usbguard/usbguard-daemon.conf
if ! test -f "//usr/local/etc/usbguard/rules.conf"; then \
  touch //usr/local/etc/usbguard/rules.conf && \
  chmod 600 //usr/local/etc/usbguard/rules.conf; \
fi
/bin/mkdir -p /usr/local/share/polkit-1/actions/ && \
  /usr/bin/install -c -m 644 ./src/DBus/org.usbguard.policy /usr/local/share/polkit-1/actions/
/bin/mkdir -p /usr/local/share/dbus-1/system-services && \
  /usr/bin/install -c -m 644 ./src/DBus/org.usbguard.service /usr/local/share/dbus-1/system-services
/bin/mkdir -p /usr/local/share/dbus-1/system.d && \
  /usr/bin/install -c -m 644 ./src/DBus/org.usbguard.conf /usr/local/share/dbus-1/system.d
make[4]: Leaving directory '/home/xx/Downloads/usbguard'
make[3]: Leaving directory '/home/xx/Downloads/usbguard'
make[2]: Leaving directory '/home/xx/Downloads/usbguard'
make[1]: Leaving directory '/home/xx/Downloads/usbguard'
`
 after that i ran
`# ldconfig`

`# usbguard genrate-policy > rules.conf`

`# install -m 0600 -o root -g root rules.conf /etc/usbguard/rules.conf`

System is Debian Stretch. Any ideas what's going wrong?

AHeinlein, 2018-02-04:
A dbus service is something different than a systemd service. What you have posted is the dbus service file.
Judging from the log you posted, no systemd unit was installed. It should go into /usr/lib/systemd/system on Debian.
I have never installed usbguard manually, but from looking at the source it seems there is no systemd unit file shipped.
You could try extracting it from the usbguard debian packages. You will need to adapt the paths, as the debian script will try to run /usr/sbin/... instead of /usr/local/sbin/...

hartwork, 2022-02-19:
Properly answered and no reply every since. My vote for closing as "answered".

closed 2022-02-23 by radosroka