USBGuard/usbguard #225: GUI (applet) crashes when attaching or detaching USB device while the screen is locked

genodeftest, 2018-03-10:
How reproducible:
100%

Steps to reproduce:
1. suspend computer (locks screen too)
2. attach or detach any USB device
3. wake up computer
4. unlock screen

Alternative steps to reproduce:
1. lock screen
2. attach or detach any USB device
3. unlock screen

What happens:
While the screen is locked, the Qt applet crashes with this stacktrace:
```
                Stack trace of thread 2401:
                #0  0x00007fe916558ad0 _ZNK7QWindow6handleEv (libQt5Gui.so.5)
                #1  0x00007fe908152869 _ZN10QXcbWindow36systemTrayWindowGlobalGeometryStaticEPK7QWindow (
                #2  0x00007fe916fd6d93 _ZNK18QSystemTrayIconSys14globalGeometryEv (libQt5Widgets.so.5)
                #3  0x00007fe916fd7312 _ZN22QSystemTrayIconPrivate15showMessage_sysERK7QStringS2_RK5QIcon
                #4  0x00007fe916fb941e _ZN15QSystemTrayIcon11showMessageERK7QStringS2_NS_11MessageIconEi 
                #5  0x0000563d20cc495f _ZN10MainWindow6notifyERK7QStringN15QSystemTrayIcon11MessageIconER
                #6  0x0000563d20cc4ed1 _ZN10MainWindow27notifyDevicePresenceChangedEN8usbguard13DeviceMan
                #7  0x0000563d20cca457 _ZN10MainWindow26handleDevicePresenceChangeEjN8usbguard13DeviceMan
                #8  0x0000563d20cd556b _ZN10MainWindow18qt_static_metacallEP7QObjectN11QMetaObject4CallEi
                #9  0x00007fe91602a26a _ZN7QObject5eventEP6QEvent (libQt5Core.so.5)
                #10 0x00007fe916cac953 _ZN7QWidget5eventEP6QEvent (libQt5Widgets.so.5)
                #11 0x00007fe916dc059b _ZN11QMainWindow5eventEP6QEvent (libQt5Widgets.so.5)
                #12 0x00007fe916c6d92c _ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent (libQt5W
                #13 0x00007fe916c75174 _ZN12QApplication6notifyEP7QObjectP6QEvent (libQt5Widgets.so.5)
                #14 0x00007fe916000be7 _ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent (libQt5Co
                #15 0x00007fe91600336b _ZN23QCoreApplicationPrivate16sendPostedEventsEP7QObjectiP11QThrea
                #16 0x00007fe916051e43 _ZL23postEventSourceDispatchP8_GSourcePFiPvES1_ (libQt5Core.so.5)
                #17 0x00007fe9114d9b77 g_main_dispatch (libglib-2.0.so.0)
                #18 0x00007fe9114d9f20 g_main_context_iterate (libglib-2.0.so.0)
                #19 0x00007fe9114d9fac g_main_context_iteration (libglib-2.0.so.0)
                #20 0x00007fe916051c2f _ZN20QEventDispatcherGlib13processEventsE6QFlagsIN10QEventLoop17Pr
                #21 0x00007fe915fff96a _ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE (libQt5Core.
                #22 0x00007fe916008094 _ZN16QCoreApplication4execEv (libQt5Core.so.5)
                #23 0x0000563d20cc03c6 main (usbguard-applet-qt)
                #24 0x00007fe915131f2a __libc_start_main (libc.so.6)
                #25 0x0000563d20cc078a _start (usbguard-applet-qt)
```
This crash happens immediately (or, in case of a suspended PC, immediately at wakeup), i.e. before the session has been unlocked.

What should happen:
No crash.

Additional info:
It does not matter whether the device is allowed or denied by default. It does not matter what type the device is (tested: USB thumb drive, USB sound card, Android smartphone).

The crash only happens when notifications are enabled. After disabling them, the crash goes away.

I am using the USBGuard Qt applet under gnome-shell 3.26. Since this version dropped tray icon support, I am using the `topicons` gnome-shell extension which display a system tray. Due to the design of gnome-shell, any extension must be disabled while the computer is locked (which includes being suspended), so the tray icon cannot be displayed immediately after waking up the PC from suspend. I am not 100% sure this fact is related, but I guess it is.

If this is caused by the combination of gnome-shell+extension, I guess it is a bug somewhere else, i.e. in Qt or the design of gnome-shell.

genodeftest, 2018-03-10:
Full backtrace from `coredumpctl gdb` with `bt` and `t a a bt full`:
[usbguard-applet-qt-SIGSEGV.txt](https://github.com/USBGuard/usbguard/files/1799548/usbguard-applet-qt-SIGSEGV.txt)

radosroka, 2019-08-08:
I think this is not relevant anymore because The PoC applet isn't a part of the main project anymore: bf3bb09

closed 2019-08-08 by radosroka

genodeftest, 2019-08-12:
> I think this is not relevant anymore because the PoC applet isn't a part of the main project anymore: [bf3bb09](https://github.com/USBGuard/usbguard/commit/bf3bb09edf491892617d35b55edec06f92d2743f)

Where will the applet be maintained?