USBGuard/usbguard #246: All USB ports disabled after install

Saroumane, 2018-08-18:
I installed usbguard 0.7.2 with synpatics on Xubuntu 18.04
All USB ports were instantly disabled (including USB mouse and keyboard)

I had to hard reset my computer and boot in recovery mode to remove this trapped package... (bye bye unsaved data)
I can't figure that such a dangerous package is allowed on official repo, there is a serious design flaw.

techge, 2018-08-18:
Mh, I think it's not the fault of usbguard devs, if the software is packaged in this way by the Ubuntu/Debian maintainer... I suggest to file a bug report there.

schnittchen, 2018-08-18:
A very similar thing just happened to me on Arch Linux. It was an upgrade of usbguard which somehow did not recognize the old rule configuration.

I vaguely remember that my first setup had been very painful for the same reason as the OP. A default on and filter everything installation is useless. Having to try what needs to be whitelisted with a number of hard reboots is not useful either. This problem should be adressed by USBGuard, not by package maintainers.

For now I am leaving USBGuard uninstalled, because I do not  have the time to go through the setup again (unless this has become much easier in the recent past, please let me know so.) I'm merely subscribing this issue.

jona71, 2018-12-21:
Recovery? I cannot use keyboard even in recovery mode!
Is there a solution?
The mistake i made was to install usbguard remote, perhaps.

muelli, 2018-12-27:
https://github.com/USBGuard/usbguard/pull/267 might help this situation

schnittchen, 2019-01-03:
I just went through this again. There is no way of installing USBGuard on Ubuntu through the official repositories without immediately losing all input devices. There is no warning, no delay. Recovery mode does has the same problem. The only way to recover is to boot into another linux, mount the root partition, bind-mount /dev, /proc and /sys and uninstall USBGuard.

@techge you implicitly mentioned there should be a way for this to be packaged in a sane way. Could you give a hint?

schnittchen, 2019-01-03:
I just did `apt install usbguard && apt remove -y usbguard` hoping it would restore me to a working state. It did not (I needed to reboot). Also, the install issued the warning that the systemd service was in a disabled state and would not be started.

I suspect the ubuntu package runs usbguard somehow manually.

techge, 2019-01-03:
> @techge you implicitly mentioned there should be a way for this to be packaged in a sane way. Could you give a hint?

Well, I don't know, what the debian/ubuntu package is actually doing, but the problem seems to be that it starts the usbguard daemon automatically. This is imho a bad habit in general, but can improve the user experience which is the reason it is generally done in Ubuntu (e.g. if I install X I probably want to use X). On Arch Linux no package is activated automatically. 

This is what should be done in Ubuntu for Usbguard as well, as it obviously breaks the system otherwise. You may file a bug report in Ubuntu/Debian and asks for not enabling the service automatically.

RyuzakiKK, 2019-01-04:
I also think that the PR #267 should help, because in this way the already plugged devices will get authorized by default. Ubuntu/Debian can also change this config value at installation time.

Btw if someone else gets stuck in the same @catfarts1 situation, you could recover your system with a chroot (or simply boot another linux distro from a live USB) and you can either add to the `/etc/usbguard/rules.conf` file this line `allow id *:*` or you can open the file file `/etc/usbguard/usbguard-daemon.conf` and edit the variable value of `PresentDevicePolicy`  from `apply-policy` to `allow`.
With the first method you just authorize every USB devices (in fact is like if you disable USBGuard), with the second method you authorize only the USB devices that are already present at boot time.

schnittchen, 2019-01-04:
Adding do @RyuzakiKK's recovery advice, if you prefer packaged installation because it saves you from library dependency mess, you can compile and install from source (I recommend using `checkinstall` so you can easily uninstall), then use `usbguard generate-policy` as described here: https://usbguard.github.io/documentation/rule-language.html, then uninstall and install the packaged version. It should pick up your working generated policy file in `/etc/usbguard/rules.conf` and not lock you out.

pjobson, 2019-04-18:
This is still happening, I think there are additional problems with the package. I installed on Linux Mint 19.1 based on Ubuntu 18.04.2 LTS. Luckily I'm on a laptop as all of the external USB ports got disabled, the internal keyboard and mouse still functioned.  Upon attempting to uninstall the installer would hang at 25%, I wound up stopping then disabling the `usbguard.service` and `usbguard-dbus.service` services, then uninstalling and purging worked fine.

MLutt, 2020-05-27:
If someone comes across this issue: No need to remove USBGuard (since most likely you had a reason to install it in the first place?)

-> boot up a live stick
-> Open up `/etc/usbguard/usbguard-daemon.conf` within your "actual" partition as root and with whichever editor you like
-> Change `ImplicitPolicyTarget=block` to `ImplicitPolicyTarget=allow`
-> save and reboot

Everything should be working now.

To start actually using it and having it working, execute the following in the shell of your choice (sh/bash flavored):
`sudo usbguard generate-policy > /etc/usbguard/rules.conf`

This should have added all the currently connected usb devices to the rules.conf file specifying them as allowed.
After that, restart the services: `sudo service usbguard restart && sudo service usbguard-dbus restart` and verify the applied policies with `sudo usbguard list-rules`

After you verified it, it _should_ be safe to set `ImplicitPolicyTarget=block` in your `usbguard-daemon.conf` afterwards.

Doesn't fix the install package being pretty much broken though. However, this is NOT an issue caused by USBGuard but simply by a huge and unbelievably egg-headed initial configuration (literally "block all devices without an applying policy" whilst there are no policies nor any way to make one...). Aka caused by whomever packaged it for the distro.

However, has been a year, I just ran into it anyway. I doubt it's going to be fixed. Sorry to make a zombie of the issue but there has yet to be mentioned an actual way to resolve this, so... Here you go :)

Saroumane, 2020-05-27:
That's an interesting case.
If I read correctly the thread, there are (at least) 3 major Linux distros (Ubuntu, Mint, Archlinux) who have "badly" packaged USBGuard.
And if read correctly  https://github.com/USBGuard/usbguard/issues/246#issuecomment-634967723 the situation could be fix upstream in USBGuard itself, by changing some default values.

So yes, it may be doable to go after every active linux distro (274 according to distrowatch) and ask them to check/fix their packaging. (the inefficient pro-active way)
Or let "would-be" users crash their PC a few times and (try to) handle the problem with their distro (the passive and harmful way)
Or to fix the problem upstream by tweaking a bit some default values ?

Anyway it's just my 2 cents : I don't intend to use this package anymore.
And thanks for the "zombie" : it reminded me to unsubscribe from this thread.

closed 2020-05-27 by Saroumane

Digital-Rnd-711, 2020-06-25:
When I try to build usbguard from here, the list of dependencies are too many that I can never get it to compile fully. It says libqb and libsystemd all mess up.
Any other ways to fix usbguard on ubuntu 18.04

techge, 2020-06-25:
> Any other ways to fix usbguard on ubuntu 18.04

As said before, the package maintainer can most likely fix it. At least they are the ones to address, not this repo. They should decide what is a sane default imho.

> If I read correctly the thread, there are (at least) 3 major Linux distros (Ubuntu, Mint, Archlinux) who have "badly" packaged USBGuard.

No. In Arch Linux it is not badly packaged as it is only enabled and started if you decide to do so. You may read the manual before you do so... Most Arch users will do, I guess.

hugepanda, 2020-06-26:
It is annoying that Ubuntu will auto-start daemon when the installation is done. I am trying to see if I can install this manually without locking any USB port. 

I can install it successfully, however, usbguard daemon could not start, which I will describe later.
Can you please help? @techge 

1. Here are the steps that I followed to manually install usbguard on Ubuntu 18.04:
```
apt -s

apt install git

apt install libgcrypt11-dev protobuf-compiler libprotobuf-dev libdbus-1-dev libdbus-glib-1-dev libpolkit-gobject-1-dev libqt4-dev libseccomp-dev xsltproc libtool libqb-dev libxml2-utils dh-autoreconf

git clone https://github.com/dkopecek/usbguard

cd usbguard

./autogen.sh

./configure --prefix=/usr --sysconfdir=/etc --with-bundled-catch --with-bundled-pegtl --with-crypto-library=gcrypt --with-gui-qt=qt4 –enable-systemd

make

make install
```

2. Then I generated the policy file:
`usbguard generate-policy > /etc/usbguard/rules.conf`

3. Finally, I tried to start the daemon, but it failed:
`systemctl start usbguard.service`

4. Here is the error message:

```
root@mengchao-Ubuntu:~# systemctl status usbguard.service
usbguard.service - USBGuard daemon
   Loaded: loaded (/lib/systemd/system/usbguard.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2020-06-26 01:20:35 CDT; 5min ago
     Docs: man:usbguard-daemon(8)
  Process: 2398 ExecStart=/usr/sbin/usbguard-daemon -f -s -c /etc/usbguard/usbguard-daemon.conf (code=exited, status=1/FAILURE)

Jun 26 01:20:35 mengchao-Ubuntu systemd[1]: usbguard.service: Service hold-off time over, scheduling restart.
Jun 26 01:20:35 mengchao-Ubuntu systemd[1]: usbguard.service: Scheduled restart job, restart counter is at 5.
Jun 26 01:20:35 mengchao-Ubuntu systemd[1]: Stopped USBGuard daemon.
Jun 26 01:20:35 mengchao-Ubuntu systemd[1]: usbguard.service: Start request repeated too quickly.
Jun 26 01:20:35 mengchao-Ubuntu systemd[1]: usbguard.service: Failed with result 'exit-code'.
Jun 26 01:20:35 mengchao-Ubuntu systemd[1]: Failed to start USBGuard daemon.
```

```
root@mengchao-Ubuntu:~# journalctl -xe
-- Subject: Unit usbguard.service has finished shutting down
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Unit usbguard.service has finished shutting down.
Jun 26 01:20:35 mengchao-Ubuntu systemd[1]: usbguard.service: Start request repeated too quickly.
Jun 26 01:20:35 mengchao-Ubuntu systemd[1]: usbguard.service: Failed with result 'exit-code'.
Jun 26 01:20:35 mengchao-Ubuntu systemd[1]: Failed to start USBGuard daemon.
-- Subject: Unit usbguard.service has failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Unit usbguard.service has failed.
-- 
-- The result is RESULT.
```

techge, 2020-06-26:
May I ask to try a more easy solution: already provide a sane `usbguard-daemon.conf` in `/etc/usbguard` and you should be good to go. At least to my knowledge apt should not override existing configuration files. Thus, there wouldn't be any harm if apt starts the daemon immediately. You might take the default of the debian package as a template and have a look at [the documentation](https://usbguard.github.io/documentation/configuration.html) what makes sense for you.

Please note that I can not guarantee that this will work. But I guess this approach is far more easy than debugging your attempts to build the software yourself.

Good luck!

muelli, 2020-06-26:
> As said before, the package maintainer can most likely fix it. At least they are the ones to address, not this repo. They should decide what is a sane default imho

Requiring all downstreams to patch your defaults is silly, IMHO.

You could also argue the other way around: Have a default that does not break users machines and let downstreams be responsible for breakage. https://github.com/USBGuard/usbguard/pull/267#issuecomment-458680201

> It is annoying that Ubuntu will auto-start daemon when the installation is done.

The argument, AFAIU, is that you install the software mainly to run it. From that standpoint, it makes a lot of sense. And it seems to work quite well for all the other daemons that you can install.

> I am trying to see if I can install this manually without locking any USB port.

Actually, if you install through Debian (and Ubuntu), you should have a policy generated at install time: https://salsa.debian.org/bisco-guest/usbguard/-/blob/master/debian/usbguard.postinst
I thought Ubuntu even patches the defaults, but I can't find that. Maybe I was simply mistaken.

hugepanda, 2020-06-26:
> May I ask to try a more easy solution: already provide a sane `usbguard-daemon.conf` in `/etc/usbguard` and you should be good to go. At least to my knowledge apt should not override existing [...]

Thanks for the suggestion. I tried installing usdguard through apt install usbguard without overwriting the rule files. But it still did not work.

curbengh, 2021-02-21:
I've been following this [installation guide](https://roussos.cc/2019/08/19/usbguard/) on Ubuntu, Manjaro/Arch & NixOS and never had issue. The trick is to **mask** the unit file *before* installation, so the service doesn't start immediately after install.

```
sudo ln -s /dev/null /etc/systemd/system/usbguard.service # mask it
sudo apt install usbguard
sudo -s
usbguard generate-policy > /etc/usbguard/rules.conf
exit
systemctl unmask usbguard.service # safe to unmask
systemctl start usbguard.service
systemctl enable usbguard.service
```