USBGuard/usbguard #248: More trusted devices and just trusted devices

morfikov, 2018-08-28:
In the file `/etc/usbguard/usbguard-daemon.conf` there are two settings:
```
ImplicitPolicyTarget=block
InsertedDevicePolicy=apply-policy
```

When they are set in the way you see above, you can whitelist some devices and add them to the `/etc/usbguard/rules.conf` file. [In the blog post ](https://usbguard.github.io/blog/2017/Screen-Locking), there was mentioned that the second option can be set also to `block` (via `usbguard set-parameter InsertedDevicePolicy block`), and in this way all disconnected devices would be blocked after plugging them in again, and no matter what the device rules say.

What I want to achieve is to have "more" trusted devices and "just" trusted devices. In this way only set of more trusted devices would be allowed to be replugged after disconnecting them from a USB port, and the rest would have to be verified by the user. Is that doable? :smile:

muelli, 2020-01-09:
I have InsertedDevicePolicy set to allow. I plug a device in, and then I change the InsertedDevicePolicy to block, because, say, I turn the screensaver on.

I want to be able to re-plug the device within a certain timeframe. Say, 5 seconds or so.

Is that your scenario?  If so, I suggest updating the title to reflect that. Maybe "Allow for a device to be replugged".

This probably also helps if the machine is suspended or hibernated while the screensaver is on and thus the InsertedDevicePolicy is set to blocked. We wouldn't want to lock the user out of their machine by not allowing the devices to work again.

morfikov, 2020-01-10:
Try this scenario: we turn the screensaver on and then someone disconnects the keyboard and mouse. How you would connect them to the system without hard reboot or sysrq? It even doesn't have to be someone -- sometimes weak usb ports can disconnect a device by some cable/plug movement. So I just wanted to whitelist at least my keyboard (if I remember correctly, since this is 2y old issue) to avoid this kind of lockup.