USBGuard/usbguard #258: Issues building for arm in a debian-stretch docker container

hansamann, 2018-10-12:
I am trying to build usbguard for arm devices (raspberry pi) and therefore installed docker and use a debian-stretch based image to keep it all nicely contained. Unfortunately I got stuck when running the configure script. 

Here are my steps taken in the container (first line is how I started it, fairly simple for you to retry)

# usb guard arm

docker container run -it --name usbguard debian:stretch /bin/bash

apt-get install libqb-dev protobuf-compiler libsodium-dev openssl ca-certificates git libseccomp-dev dh-autoreconf

git clone https://github.com/USBGuard/usbguard.git

autoreconf -i
./configure --with-crypto-library=sodium 

root@6b9107b2879c:/usbguard# ./configure --with-crypto-library=sodium 
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking whether make supports nested variables... (cached) yes
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking whether C compiler accepts -Wno-implicit-fallthrough... yes
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking for gcc option to accept ISO C99... none needed
checking whether make sets $(MAKE)... (cached) yes
checking build system type... armv7l-unknown-linux-gnueabihf
checking host system type... armv7l-unknown-linux-gnueabihf
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert armv7l-unknown-linux-gnueabihf file names to armv7l-unknown-linux-gnueabihf format... func_convert_file_noop
checking how to convert armv7l-unknown-linux-gnueabihf file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/bin/ld) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking how to run the C++ preprocessor... g++ -E
checking for ld used by g++... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking whether the g++ linker (/usr/bin/ld) supports shared libraries... yes
checking for g++ option to produce PIC... -fPIC -DPIC
checking if g++ PIC flag -fPIC -DPIC works... yes
checking if g++ static flag -static works... yes
checking if g++ supports -c -o file.o... yes
checking if g++ supports -c -o file.o... (cached) yes
checking whether the g++ linker (/usr/bin/ld) supports shared libraries... yes
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking for __atomic_add_fetch_8 in -latomic... yes
checking for NETLINK_KOBJECT_UEVENT definition... yes
./configure: line 16872: syntax error near unexpected token `qb,'
./configure: line 16872: `PKG_CHECK_MODULES(qb, libqb >= 0.16.0,'

Any help greatly appreciated. It would also be great to get arm binaries / installation packages for debian-stretch RPI. 

Thx!
Sven

NothinRandom, 2020-06-08:
Sven, 
Were you able to get usbguard to work on RPI?   It looks like the default version (0.6.2+ds1-2) from repo doesn't work and throws "ERROR: IPC connect: service=usbguard: Connection refused" when you run usbguard commands.

Its status
```
$ sudo service usbguard status
● usbguard.service - USBGuard daemon
   Loaded: loaded (/lib/systemd/system/usbguard.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2020-05-31 14:09:34 PDT; 1 weeks 0 days ago
     Docs: man:usbguard-daemon(8)
  Process: 1545 ExecStart=/usr/sbin/usbguard-daemon -k -c /etc/usbguard/usbguard-daemon.conf (code=exited, status=1/FAILURE)
 Main PID: 1545 (code=exited, status=1/FAILURE)
```

hartwork, 2022-02-02:
Does this still happen with recent Git `master` today?