USBGuard/usbguard #260: CentOS6/RHEL6 support?

ercan1441, 2018-10-16:
Is there a way to compile usbguard for centos6/rhel6? I can't find a pre-compiled rpm and would like to compile it myself. But unfortunately I get the following error:

# ./configure --with-crypto-library=sodium
....
<output omitted>
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for qb... yes
checking for sodium... yes
checking for libgcrypt-config... /usr/bin/libgcrypt-config
checking for LIBGCRYPT - version >= 1.5.0... no
checking for audit... no
configure: libaudit development files not found! Linux Audit support wille be turned OFF
checking for seccomp... no
configure: libseccomp development files not found! Seccomp support will be turned OFF
checking for libcapng... no
configure: libseccomp development files not found! Seccomp support will be turned OFF
checking for umockdev... no
configure: umockdev development files not found! umockdev device manager won't be available
checking for protobuf... no
configure: error: Required protobuf development files not found!

I've the following protobuf rpms installed:
# rpm -qa | grep protobuf
protobuf-c-0.15-2.el6.x86_64
protobuf-python-2.3.0-9.el6.x86_64
protobuf-devel-2.3.0-9.el6.x86_64
protobuf-lite-static-2.3.0-9.el6.x86_64
protobuf-lite-2.3.0-9.el6.x86_64
protobuf-static-2.3.0-9.el6.x86_64
protobuf-c-debuginfo-0.15-2.el6.x86_64
protobuf-compiler-2.3.0-9.el6.x86_64
protobuf-c-devel-0.15-2.el6.x86_64
protobuf-2.3.0-9.el6.x86_64
protobuf-vim-2.3.0-9.el6.x86_64
protobuf-lite-devel-2.3.0-9.el6.x86_64
protobuf-debuginfo-2.3.0-9.el6.x86_64

userWayneCampbell, 2018-10-24:
I would try compiling protobuf manually: https://github.com/protocolbuffers/protobuf

I can compile usbguard on centos7 perfectly with the following packages:
```
protobuf-compiler-2.5.0-8.el7.x86_64
protobuf-2.5.0-8.el7.x86_64
protobuf-devel-2.5.0-8.el7.x86_64

```

ercan1441, 2018-10-25:
I installed the following packages:

```
protobuf-2.5.0-16.1.x86_64
protobuf-compiler-2.5.0-16.1.x86_64
protobuf-devel-2.5.0-16.1.x86_64
```

But now I end up with:
```
# ./configure --with-crypto-library=sodium
<output omitted>
..
checking for sodium... yes
checking for libgcrypt-config... /usr/bin/libgcrypt-config
checking for LIBGCRYPT - version >= 1.5.0... no
checking for audit... no
configure: libaudit development files not found! Linux Audit support wille be turned OFF
checking for seccomp... no
configure: libseccomp development files not found! Seccomp support will be turned OFF
checking for libcapng... no
configure: libseccomp development files not found! Seccomp support will be turned OFF
checking for umockdev... no
configure: umockdev development files not found! umockdev device manager won't be available
checking for protobuf... yes
checking for protoc... protoc
checking catch.hpp usability... no
checking catch.hpp presence... no
checking for catch.hpp... no
configure: error: in `/root/usbguard-0.7.4':
configure: error: catch.hpp not found or not usable. Re-run with --with-bundled-catch to use the bundled library.
See `config.log' for more details
```

A re-run with "--with-bundled-catch" results in:
```
./configure --with-crypto-library=sodium --with-bundled-catch
<output omitted>
...
checking for protobuf... yes
checking for protoc... protoc
configure: Using bundled Catch library
checking tao/pegtl.hpp usability... no
checking tao/pegtl.hpp presence... no
checking for tao/pegtl.hpp... no
configure: error: in `/root/usbguard-0.7.4':
configure: error: PEGTL header file not found or not usable. Re-run with --with-bundled-pegtl to use the bundled library.
See `config.log' for more details
```

A re-run with "--with-bundled-pegtl" results in:
```
./configure --with-crypto-library=sodium --with-bundled-pegtl
<output omitted>
...
checking for protobuf... yes
checking for protoc... protoc
checking catch.hpp usability... no
checking catch.hpp presence... no
checking for catch.hpp... no
configure: error: in `/root/usbguard-0.7.4':
configure: error: catch.hpp not found or not usable. Re-run with --with-bundled-catch to use the bundled library.
See `config.log' for more details
```

A re-run with both flags "--with-bundled-catch --with-bundled-pegtl" results in:
```
./configure --with-crypto-library=sodium --with-bundled-catch --with-bundled-pegtl
<output omitted>
...
configure: Using bundled Catch library
configure: Using bundled PEGTL library
checking tao/pegtl.hpp usability... no
checking tao/pegtl.hpp presence... no
checking for tao/pegtl.hpp... no
configure: error: in `/root/usbguard-0.7.4':
configure: error: PEGTL header file not found or not usable. Re-run with --with-bundled-pegtl to use the bundled library.
See `config.log' for more details
```

Somehow it can't find tao/pegtl.hpp. Although its there in the src tree:
```
find . -name pegtl.hpp
./src/ThirdParty/PEGTL/include/tao/pegtl.hpp
```

userWayneCampbell, 2018-11-12:
Just an idea, but did you ever run the `ldconfig` command?

ercan1441, 2018-11-16:
Yes I ran ldconfig, but this didn't solve the problem..

zerpex, 2018-12-07:
exactly same issue here on CentOS 7.3

Trying to compile with:
./configure --with-crypto-library=gcrypt --with-bundled-catch --with-bundled-pegtl

and got the same error: 

configure: Using bundled Catch library
configure: Using bundled PEGTL library
checking tao/pegtl.hpp usability... no
checking tao/pegtl.hpp presence... no
checking for tao/pegtl.hpp... no
configure: error: in `/home/xfit/PTM/usbguard-0.7.4':
configure: error: PEGTL header file not found or not usable. Re-run with --with-bundled-pegtl to use the bundled library.
See `config.log' for more details

Tried a lot of thing (including compile PEGTL), but don't have any luck on passing this state.

pipari666, 2020-07-09:
Same error with CentOS 8.2. 

Trying to compile with:
./configure --with-crypto-library=sodium --with-bundled-catch --with-bundled-pegtl

and get the similar error:
....
configure: Using bundled Catch library
checking tao/pegtl.hpp usability... no
checking tao/pegtl.hpp presence... no
checking for tao/pegtl.hpp... no
configure: error: in `/home/username/rpmbuild/BUILD/usbguard-0.7.8':
configure: error: PEGTL header file not found or not usable. Re-run with --with-bundled-pegtl to use the bundled library.

Also tried compiling PEGTL without any success. Official USBGuard release in CentOS8 is not stable. Really need to this. 

Please help or point me in any direction! 

Best regards

radosroka, 2020-07-18:
> I installed the following packages:
>
> ```
> protobuf-2.5.0-16.1.x86_64
> protobuf-compiler-2.5.0-16.1.x86_64
> protobuf-devel-2.5.0-16.1.x86_64
> ```
>
> But now I end up with:
>
> ```
> # ./configure [...]

How did you clone the repository? Have you fetched sub-modules?

radosroka, 2020-07-18:
> Same error with CentOS 8.2.
>
> Trying to compile with:
> ./configure --with-crypto-library=sodium --with-bundled-catch --with-bundled-pegtl
>
> and get the similar error:
> ....
> configure: Using bundled [...]

Have you fetched sub-modules?

hartwork, 2022-02-19:
My vote for closing as obsolete, [Centos 6 is end-if life](https://endoflife.date/centos).

closed 2022-02-23 by radosroka