USBGuard/usbguard #261: EINTR causes: sigwaitinfo failed: errno=4; Shutting down.

Allen-Webb, 2018-10-24:
Under some circumstances sigwaitinfo (and other syscalls) can get an EINTR which generally means the program should repeat the previous syscall because of an interrupt, so it might make sense to call sigwaitinfo again.

However, this might be intended behavior or better left as is if usbguard were to lose track of uevents or other system state changes having it close on EINTR would give it the opportunity for the daemon to be respawned and pickup the current state instead of resuming with the old state. I am not convinced the state would get out of sync in practice,

You can trigger this behavior by the following:
sudo kill -SIGSTOP ${USBGUARD_PID}; sudo kill -SIGCONT ${USBGUARD_PID}
It isn't necessary to send a SIGSTOP though, an EINTR can happen for a variety of reasons, this is just an easy way to trigger the behavior. Ideally, if you want to handle EINTR, you would need to do it for a good number of the syscalls besides just sigwaitinfo.

infokiller, 2019-10-02:
I can confirm what @rathann is seeing.

infokiller, 2019-10-02:
If I understand the code correctly, the big issue is that when this error happens, it doesn't affect the exit code of the process, which will always return success:
https://github.com/USBGuard/usbguard/blob/4957d2d0bc4c2ed4529e8b69f9813c735d51a69a/src/Daemon/main.cpp#L196-L197

And it seems that the systemd service is configured to only restart on failures:
https://github.com/USBGuard/usbguard/blob/4957d2d0bc4c2ed4529e8b69f9813c735d51a69a/usbguard.service.in#L25

So it means that once this error happens, the systemd service will finish with success and not be restarted. I guess that as a workaround, I can `systemd edit` the service file to always restart.
As for a long term fix, is there any objection for a PR that makes the process exit with an error in this case?

zuzzurro, 2019-11-13:
On my fedora 31 usbguard is basically useless as it keeps exiting. Is someone actually looking into this?

genodeftest, 2019-12-21:
> On my fedora 31 usbguard is basically useless as it keeps exiting. Is someone actually looking into this?

There is an update, also [for Fedora 31](https://bodhi.fedoraproject.org/updates/FEDORA-2019-d4c4bf697a) but I now instead see issue #279 which breaks usbguard completely in my case.

radosroka, 2019-12-30:
From upstream point of view this seems to be fixed.

closed 2019-12-30 by radosroka