USBGuard/usbguard #262: Example in documentation seems irrational

rlcarro, 2018-10-25:
According to the documentation, when a USB device is connected, it scans the existing rules until it finds a match.
In the example of blocking a flash disk which implements a keyboard or network interface, it seems the first rule will match and it will be allowed.  If it continues to scan rules after finding a match, I find no mention of that behavior.  To me it seems the "allow" line should be at the end of the rules.

                  allow with-interface equals { 08:*:* }
                  reject with-interface all-of { 08:*:* 03:00:* }
                  reject with-interface all-of { 08:*:* 03:01:* }
                  reject with-interface all-of { 08:*:* e0:*:* }
                  reject with-interface all-of { 08:*:* 02:*:* }

AHeinlein, 2018-10-25:
Actually, this is correct. If a device has multiple interfaces, an "equals" rule must contain all of them to match.
Even a rule 
`allow with-interface equals { 08:*:* 08:*:* }`
makes sense. This is for USB hard drives which support UASP (USB attached SCSI protocol), those too have two USB interfaces, one for conventional USB mass storage and one for UASP.

rlcarro, 2018-10-25:
Thanks!  It is useful to know that it will continue scanning rules after the first match.  I was hoping to allow only Ironkey USB storage devices, reject any other USB storage, and leave the rest alone with this construct:

allow id equals { 1953:* } with-interface equals { 08:*:* }
reject with-interface equals { 08:*:* }
allow with-interface equals { *:*:* }

Does that look right?  Too simple?

rlcarro, 2018-10-25:
Oops, something happened to my asterisks with the paste.

allow id equals { 1953:* } with-interface equals { 08:*:* }
reject with-interface equals { 08:*:* }
allow with-interface equals { *:*:* }

rlcarro, 2018-10-25:
How embarrassing!  The asterisks are there before I post the comment, but they are being parsed out.  
Now I see why they have the Preview tab.   Sorry for being a nube.  I found the "Insert code" tool.

`    allow id equals { 1953:* } with-interface equals { 08:*:* }`
`    reject with-interface equals { 08:*:* }`
`    allow with-interface equals { *:*:* }`

closed 2018-10-29 by rlcarro