USBGuard/usbguard #277: Insecure Temporary Files

topimiettinen, 2019-03-01:
Usbguard-daemon creates files in world-writable directories (`/dev/shm`, `/tmp`) with rather predictable file names (e.g. `/dev/shm/qb-usbguard-request-7096-835-12-data`). Also `O_EXCL` flag is not used when opening the files. This could be exploited by a local attacker to overwrite privileged system files (if not restricted by sandboxing, MAC or symlinking policies).

Maybe usbguard-daemon could initialize IPC server with 
`qb_ipcs_create("usbguard-XXXXXX",,,,)` to make libqb to use `mkstemp()` for the files, but the problem is also on libqb's side (for example not using `O_EXCL`). I would not use files for IPC.

topimiettinen, 2019-03-01:
I've opened issue ClusterLabs/libqb#338 for libqb.

jnpkrn, 2019-03-04:
> Maybe usbguard-daemon could initialize IPC server with
> `qb_ipcs_create("usbguard-XXXXXX",,,,)` to make libqb to use
> `mkstemp()` for the files

Sadly that won't work, since `XXXXXX` needs to be the final suffix,
which won't be the case here (rather `-header` etc. will be such
a suffix).

> but the problem is also on libqb's side (for example not using
> `O_EXCL`). 

Agree.  I actually briefly thought about this surface in the past,
related to some other issues, but then it got backgrounded for
other prio stuff.

arjenlentz, 2020-04-03:
Addition  of the "PrivateTmp=true" systemd profile setting as described in #231 would mitigate this issue.
Of course it'd be best if it's secured both inside the daemon as well as in the profile.