USBGuard/usbguard #279: daemon Cannot resolve realpath

hlekin, 2019-03-10:
usbguard 0.7.4 on Arch Linux, fresh install.

Started service
$ systemctl status usbguard.service
...
usbguard-daemon[814]: Cannot resolve realpath for /sys/bus/usb/devices/../../../devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0
usbguard-daemon[814]: Cannot resolve realpath for /sys/bus/usb/devices/../../../devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0
usbguard-daemon[814]: Cannot resolve realpath for /sys/bus/usb/devices/../../../devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0

attached
$ lsusb | grep JetFlash
Bus 002 Device 003: ID 8564:1000 Transcend Information, Inc. JetFlash

which is blocked, of course
$  usbguard list-devices
...
12: block id 8564:1000 serial "xox" name "Mass Storage Device" hash "xox" parent-hash "xox" via-port "2-2" with-interface 08:06:50

and can unfortunately not be allowed
$ usbguard allow-device 8564:1000
IPC ERROR: request id=1: Device lookup: device id: id doesn't exist

Device ID does exist (somewhere)
$ cat /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/idVendor
8564
$ cat /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/idProduct
1000

Some more perhaps useful information
$ lsusb -d 0x8564:0x1000
Bus 002 Device 003: ID 8564:1000 Transcend Information, Inc. JetFlash
$  grep . /sys/devices/pci0000\:00/0000\:00\:14.0/usb2/2-3/*
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/authorized:0
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/avoid_reset_quirk:0
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/bcdDevice:0204
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/bDeviceClass:00
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/bDeviceProtocol:00
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/bDeviceSubClass:00
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/bMaxPacketSize0:9
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/bNumConfigurations:1
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/busnum:2
Binary file /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/descriptors matches
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/dev:189:129
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/devnum:2
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/devpath:3
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/devspec:          (null)
grep: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/driver: Is a directory
grep: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/ep_00: Is a directory
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/idProduct:0316
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/idVendor:0bda
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/ltm_capable:yes
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/manufacturer:Generic
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/maxchild:0
grep: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/port: Is a directory
grep: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/power: Is a directory
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/product:USB3.0-CRW
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/quirks:0x0
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/removable:fixed
grep: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/remove: Permission denied
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/rx_lanes:1
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/serial:xox
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/speed:5000
grep: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/subsystem: Is a directory
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/tx_lanes:1
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/uevent:MAJOR=189
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/uevent:MINOR=129
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/uevent:DEVNAME=bus/usb/002/002
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/uevent:DEVTYPE=usb_device
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/uevent:DRIVER=usb
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/uevent:PRODUCT=bda/316/204
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/uevent:TYPE=0/0/0
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/uevent:BUSNUM=002
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/uevent:DEVNUM=002
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/urbnum:662658
/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/version: 3.00

No idea how to proceed...

dkopecek, 2019-03-11:
Hmm, interesting, I'll have to add some more debugging information to the realpath error output.

hlekin, 2019-03-11:
Thanks for the quick reply.

Don't hesitate to request any information you want.

dkopecek, 2019-03-13:
@hlekin could you try to start the daemon manually via strace to see what the underlying error in realpath is?

hlekin, 2019-03-13:
On 13.03.19 10:33, Daniel Kopeček wrote:
> @hlekin <https://github.com/hlekin> could you try to start the daemon
> manually via strace to see what the underlying error in realpath is?
> 

1) I misunderstood, that 'usbguard block-device id' does NOT mean
   $ usbguard allow-device 8564:1000  :)
   IPC ERROR: request id=1: Device lookup: device id: id doesn't exist
   and I assumed, that this is related to the 'realpath' notice.

2) Meanwhile I generated an initial policy and aforementioned 'realpath'
   notice does NOT occur at daemon start anymore.

3) The 'realpath' notice can be reproduced by using an empty
   /etc/usbguard/rules.conf, which is part of the Arch package.

4) $ strace usbguard-daemon does NOT yield any 'realpath' output.
   Closest related output might be:
   (if too painful with line breaks, let me know)

fstat(12, {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
getdents64(12, /* 17 entries */, 32768) = 440
lstat("/sys/bus/usb/devices/2-3", {st_mode=S_IFLNK|0777, st_size=0,
...}) = 0
readlink("/sys/bus/usb/devices/2-3",
"../../../devices/pci0000:00/0000"..., 4096) = 49
lstat("/sys/bus/usb/devices/1-9", {st_mode=S_IFLNK|0777, st_size=0,
...}) = 0
readlink("/sys/bus/usb/devices/1-9",
"../../../devices/pci0000:00/0000"..., 4096) = 49
lstat("/sys/bus/usb/devices/usb3", {st_mode=S_IFLNK|0777, st_size=0,
...}) = 0
readlink("/sys/bus/usb/devices/usb3",
"../../../devices/pci0000:00/0000"..., 4096) = 84
lstat("/sys/bus/usb/devices/1-7", {st_mode=S_IFLNK|0777, st_size=0,
...}) = 0
readlink("/sys/bus/usb/devices/1-7",
"../../../devices/pci0000:00/0000"..., 4096) = 49
lstat("/sys/bus/usb/devices/usb1", {st_mode=S_IFLNK|0777, st_size=0,
...}) = 0
readlink("/sys/bus/usb/devices/usb1",
"../../../devices/pci0000:00/0000"..., 4096) = 45
lstat("/sys/bus/usb/devices/3-0:1.0", {st_mode=S_IFLNK|0777, st_size=0,
...}) = 0
readlink("/sys/bus/usb/devices/3-0:1.0",
"../../../devices/pci0000:00/0000"..., 4096) = 92
lstat("/sys/bus/usb/devices/1-10", {st_mode=S_IFLNK|0777, st_size=0,
...}) = 0
readlink("/sys/bus/usb/devices/1-10",
"../../../devices/pci0000:00/0000"..., 4096) = 50
lstat("/sys/bus/usb/devices/1-3", {st_mode=S_IFLNK|0777, st_size=0,
...}) = 0
readlink("/sys/bus/usb/devices/1-3",
"../../../devices/pci0000:00/0000"..., 4096) = 49
lstat("/sys/bus/usb/devices/4-0:1.0", {st_mode=S_IFLNK|0777, st_size=0,
...}) = 0
readlink("/sys/bus/usb/devices/4-0:1.0",
"../../../devices/pci0000:00/0000"..., 4096) = 92
lstat("/sys/bus/usb/devices/usb4", {st_mode=S_IFLNK|0777, st_size=0,
...}) = 0
readlink("/sys/bus/usb/devices/usb4",
"../../../devices/pci0000:00/0000"..., 4096) = 84
lstat("/sys/bus/usb/devices/1-8", {st_mode=S_IFLNK|0777, st_size=0,
...}) = 0
readlink("/sys/bus/usb/devices/1-8",
"../../../devices/pci0000:00/0000"..., 4096) = 49
lstat("/sys/bus/usb/devices/usb2", {st_mode=S_IFLNK|0777, st_size=0,
...}) = 0
readlink("/sys/bus/usb/devices/usb2",
"../../../devices/pci0000:00/0000"..., 4096) = 45
lstat("/sys/bus/usb/devices/1-0:1.0", {st_mode=S_IFLNK|0777, st_size=0,
...}) = 0
readlink("/sys/bus/usb/devices/1-0:1.0",
"../../../devices/pci0000:00/0000"..., 4096) = 53
lstat("/sys/bus/usb/devices/1-6", {st_mode=S_IFLNK|0777, st_size=0,
...}) = 0
readlink("/sys/bus/usb/devices/1-6",
"../../../devices/pci0000:00/0000"..., 4096) = 49
lstat("/sys/bus/usb/devices/2-0:1.0", {st_mode=S_IFLNK|0777, st_size=0,
...}) = 0
readlink("/sys/bus/usb/devices/2-0:1.0",
"../../../devices/pci0000:00/0000"..., 4096) = 53
getdents64(12, /* 0 entries */, 32768)  = 0
lstat("/sys", {st_mode=S_IFDIR|0555, st_size=0, ...}) = 0
lstat("/sys/bus", {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
lstat("/sys/bus/usb", {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
lstat("/sys/bus/usb/devices", {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
lstat("/sys/devices", {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
lstat("/sys/devices/pci0000:00", {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
lstat("/sys/devices/pci0000:00/0000:00:14.0", {st_mode=S_IFDIR|0755,
st_size=0, ...}) = 0
lstat("/sys/devices/pci0000:00/0000:00:14.0/usb1",
{st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
openat(AT_FDCWD, "/sys/devices/pci0000:00/0000:00:14.0/usb1",
O_RDONLY|O_PATH|O_DIRECTORY) = 13
openat(13, "uevent", O_RDONLY)          = 14
read(14, "MAJOR=189\nMINOR=0\nDEVNAME=bus/us"..., 4096) = 124
close(14)                               = 0
openat(13, "uevent", O_WRONLY)          = 14
write(14, "add", 3)                     = 3
close(14)                               = 0

hlekin, 2019-03-13:
Arch Linux just updated usbguard-0.7.4-4 to usbguard-0.7.4-5
as protobuf-3.7.0-1 rebuild.

Restarting the daemon does now NOT show disturbing messages like
usbguard-daemon[10577]: Ignoring unknown UEvent action:
sysfs_devpath=/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1 action=bind
anymore.

By the way, the Arch PKGBUILD can be found here:
https://git.archlinux.org/svntogit/packages.git/tree/trunk/PKGBUILD?h=packages/usbguard

qrwteyrutiyoup, 2019-04-11:
> @hlekin could you try to start the daemon manually via strace to see what the underlying error in realpath is?

Here' s an `strace` output with the "Cannot resolve realpath" message: https://gist.github.com/qrwteyrutiyoup/773550ea7cb8be0e4f0be50b746f296b

I can reproduce it if my `rules.conf` is empty. I have `ImplicitPolicyTarget=block` and `PresentDevicePolicy=apply-policy`, so it's basically going to block everything.

I was watching the contents of `/sys/devices/pci0000:00/0000:00:07.1/0000:20:00.3/usb3/3-2` and once I started `usbguard-daemon`, these three directories disappeared from the listing:
`3-2:1.0`, `3-2.3` and `3-2.4`, which caused `realpath` to fail with `ENOENT (No such file or directory)`. Once I allowed the devices again, those directories reappeared. It sounds like it's working as expected, no?

genodeftest, 2019-12-20:
Same issue here, which prevents me from using usbguard. Is there anything I can do? Is there a workaround?

genodeftest, 2020-02-05:
> Same issue here, which prevents me from using usbguard. Is there anything I can do? Is there a workaround?

I'm actually no longer sure this is true. I still see the warning but I don't *think* it breaks usbguard for me.

ZoltanFridrich, 2021-03-04:
@hlekin you are right that in `usbguard allow-device` you have to use device 'rule' id which is displayed on the left side of the list-devices command and not the actual device id. The warning "Cannot resolve realpath" is generated when there is a device being blocked while scanning. This can happen for example when you start usbguard and not all scanned devices are allowed within your policy. While scanning the device, it will be tested against your ruleset and if the device gets blocked as a result, then scanning the children paths of that device will fail with "Cannot resolve realpath" message (because it has been blocked). You can simply ignore these messages.

@qrwteyrutiyoup yes, it is working as expected.

@genodeftest I explained this, you can ignore such warnings.