USBGuard/usbguard #286: What is the order and priority of rule processing?

emanruse, 2019-04-12:
My goal is to
- all all devices I trust (existing)
- allow unknown **storage-only** USB devices (as it seems this is safe)

I have done:
```
usbguard generate-policy > /etc/usbguard/rules.conf
```
At the end of the `rules.conf` file generated this way I have added the example lines from `usbguard-rules.conf.5.adoc`  ("Reject devices with suspicious combination of interfaces"):
```
allow with-interface equals { 08:*:* }
reject with-interface all-of { 08:*:* 03:00:* }
reject with-interface all-of { 08:*:* 03:01:* }
reject with-interface all-of { 08:*:* e0:*:* }
reject with-interface all-of { 08:*:* 02:*:* }
```
Also `/etc/usbguard/usbguard-daemon.conf` is:
```
RuleFile=/etc/usbguard/rules.conf
ImplicitPolicyTarget=block
PresentDevicePolicy=apply-policy
PresentControllerPolicy=allow
IPCAllowedUsers=root
```
Until today everything seemed to work fine. Suddenly now something weird happens: the whitelisted devices which have `with-interface` starting with `03:00` or `03:01` (i.e. match the blacklist) get blocked when the usbguard daemon starts. This makes me think that until now the order of processing of these rules was the order in which they are listed in `rules.conf`. IOW: if device X is allowed, processing of further rules stops. But now it seems the `reject` rules override the `allow` ones.

My questions are:

1. What is the order in which rules are processed? (I can't find documentation for it)
2. Is what I am doing correct (considering the goal)? If not - what would be correct?

*Using usbguard-0.6.2-lp150.1.1.x86_64 (on openSUSE)

tweksteen, 2019-04-15:
I believe that the first rule that matches is used: https://github.com/USBGuard/usbguard/blob/9ed71790c622aaabaa19e4619d7d56a0fa55a149/src/Daemon/Daemon.cpp#L843

> Until today everything seemed to work fine.

Did you change the configuration? Update your distribution? Used a new USB device?

Could you share the output of `usbguard list-rules`?
Also, your version of USBGuard is slightly outdated. Any chance you could reproduce using the latest?
Thanks

emanruse, 2019-04-15:
> Did you change the configuration?
No.

> Update your distribution?
Not the whole distribution. Just the regular `zypper up` but that didn't update usbguard packages.

> Used a new USB device?
No.

> Could you share the output of `usbguard list-rules`?

As of the time of reporting the issue my rules.conf was (identifiers hidden):

```
# usbguard list-rules
allow id ----:---- serial "----" name "----" hash "----" parent-hash "----" with-interface 09:00:00
allow id ----:---- serial "----" name "----" hash "----" parent-hash "----" via-port "1-3" with-interface 03:00:00
allow id ----:---- serial "----" name "----" hash "----" parent-hash "----" via-port "1-4" with-interface ff:ff:ff
allow id ----:---- serial "----" name "----" hash "----" parent-hash "----" with-interface 09:00:00
allow id ----:---- serial "----" name "----" hash "----" parent-hash "----" with-interface 09:00:00
allow id ----:---- serial "----" name "----" hash "----" parent-hash "----" via-port "5-1" with-interface 09:00:00
allow id ----:---- serial "----" name "----" hash "----" parent-hash "----" via-port "5-1.5" with-interface 03:00:00
allow id ----:---- serial "----" name "----" hash "----" parent-hash "----" via-port "5-1.6" with-interface 09:00:00
allow id ----:---- serial "----" name "----" hash "----" parent-hash "----" via-port "5-1.6.1" with-interface 03:01:02
allow id ----:---- serial "----" name "----" hash "----" parent-hash "----" via-port "5-1.6.3" with-interface 03:00:00
allow id ----:---- serial "----" name "----" hash "----" parent-hash "----" with-interface 09:00:00
allow id ----:---- serial "----" name "----" hash "----" parent-hash "----" via-port "3-1" with-interface 03:01:02
allow id ----:---- serial "----" name "----" hash "----" parent-hash "----" with-interface 09:00:00
allow id ----:---- serial "----" name "----" hash "----" parent-hash "----" with-interface 09:00:00
allow id ----:---- serial "----" name "----" hash "----" parent-hash "----" via-port "6-1" with-interface 09:00:00
allow with-interface equals { 08:*:* }
reject with-interface all-of { 08:*:* 03:00:* }
reject with-interface all-of { 08:*:* 03:01:* }
reject with-interface all-of { 08:*:* e0:*:* }
reject with-interface all-of { 08:*:* 02:*:* }
```

And `usbguard list-rules` outputs the rules in the same order.

#### Question
As of today I have removed the last 5 rules and simply whitelisted 3 known USB storage devices and everything works as expected. I notice that all 3 of them have `with-interface 08:06:50`. Is it safe to assume that for the goal I am aiming at I just need to add a last line in `rules.conf`:
```
allow with-interface equals { 08:06:50 }
```
?
I am asking because the URL [list of assigned classes, subclasses and protocols](http://www.usb.org/developers/defined_class) linked in the [usbguard-rules.conf man page](https://github.com/USBGuard/usbguard/blob/master/doc/man/usbguard-rules.conf.5.adoc) leads to 404 page (perhaps something which needs fixing).

> Also, your version of USBGuard is slightly outdated. Any chance you could reproduce using the latest?
Unfortunately that is the latest version on the official openSUSE Leap 15.0 repos (and for security reasons I am not willing to use unofficial ones).

tweksteen, 2019-04-15:
It is worth noticing that you are using the port as part of the allow rules, which means that if the device is plugged in a different port, it will not be matching (you can use the `-P` flag of `generate-policy` to avoid that). You may also want to check the logs (`journalctl -u usbguard`) to see what happens when started.

To your question: your implicit policy is already defined as blocking (ImplicitPolicyTarget). Since the `equals` operator is strict on the interfaces declared, the last 4 rejects are already redundant with your ImplicitPolicyTarget (well, almost as block is not quite reject). So, yes you can remove them and it is up to you if you want to restrict further the subclass and protocol.

An overview of the Mass Storage subclass and protocol is available at https://www.usb.org/sites/default/files/Mass_Storage_Specification_Overview_v1.4_2-19-2010.pdf. The class codes can be found at https://www.usb.org/defined-class-codes.
Please send a patch with the updated documentation links :)

emanruse, 2019-04-15:
> It is worth noticing that you are using the port as part of the allow rules

It is not a problem because those particular devices are plugged in those ports all the time.

My question was about 08:06:50 in particular and the links you give don't quite answer it. Here is another one which seems to be clearer:

http://www.linux-usb.org/usb.ids

As far as I can read 08:06:50 means Mass storage:SCSI:Bulk-Only but I have no idea what SCSI has to do with a USB flash stick or what "Bulk-Only" is supposed to mean.

> Please send a patch with the updated documentation links :)

I don't know how to do that.

tweksteen, 2019-06-12:
It depends on the USB Mass Storage devices you are using. But yes, as a rule of thumb, `08:06:50` works for USB Mass Storage [in general](https://wiki.osdev.org/USB_Mass_Storage_Class_Devices#BBB).
I created PR #298 to update the documentation link.