USBGuard/usbguard #287: ERROR: IPC connect: service=usbguard: Permission denied since kernel 5.0.9

morfikov, 2019-04-20:
I just upgraded my kernel to the latest stable (5.0.7->5.0.9) and I notticed that `usbguard-applet-qt` has a gray icon. In it's log I can see the following messages:

```
[Sat Apr 20 14:55:21 2019] Connection failed: IPC connect: service=usbguard: Permission denied
[Sat Apr 20 14:55:22 2019] Connection failed: IPC connect: service=usbguard: Permission denied
[Sat Apr 20 14:55:23 2019] Connection failed: IPC connect: service=usbguard: Permission denied
```

Also I'm using AppArmor profiles for all the usbguard execs, and I had to add to `usr.sbin.usbguard-daemon` and `usr.sbin.usbguard-dbus` profiles the following rule:

 `/dev/shm/qb-[0-9]*-[0-9]*-[0-9]*-*/qb-{request,response,event}-usbguard-{header,data} rw,`

There was similar rule:

` /dev/shm/qb-usbguard-{request,response,event}-[0-9]*-[0-9]*-[0-9]*-{header,data} rw,`

and probably that's the cause.

morfikov, 2019-04-24:
It looks like the systemd service causes the problem. 

When I commented out the following line:
`CapabilityBoundingSet=CAP_CHOWN CAP_FOWNER`
It works well now. It's weird...

morfikov, 2019-04-24:
I had to add two extra CAPs:

`CapabilityBoundingSet=CAP_CHOWN CAP_FOWNER CAP_DAC_READ_SEARCH CAP_DAC_OVERRIDE`.

Now it works well.

tweksteen, 2019-09-16:
Re-opening this issue as the necessary changes may not have been committed. This was pulled in the Debian package (https://salsa.debian.org/bisco-guest/usbguard/commit/492e3a38ae0bce192fca926e1eedc0e71dbdb5a2), but I'm not sure these are necessary (and potentially dangerous).

Any chance you could share the version of systemd that you were using? Thanks.

morfikov, 2019-09-16:
See  #289 . 

```
#  systemd --version
systemd 242 (242)
+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP 
+GCRYPT +GNUTLS +ACL +XZ +LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD +IDN2 -IDN 
+PCRE2 default-hierarchy=hybrid
```

I'm using Debian Sid.

drGrove, 2019-11-03:
Seeing the same thing in Linux Kernel `5.3.8

cyrinux, 2019-11-04:
I think there is some more capabilities missing to be able to write on the /etc/usbguard/rules.conf

alcros, 2020-01-05:
@cyrinux yes, you must allow writing to your rules.conf file.

A quick fix is to place this file at /etc/systemd/system/usbguard.service.d/override.conf:
```
[Service]
CapabilityBoundingSet=CAP_DAC_OVERRIDE
ReadWritePaths=-/etc/usbguard/rules.conf
```
But ideally, this should be fixed upstream.

genodeftest, 2020-02-05:
@alcros wrote:
> ```
> [Service]
> CapabilityBoundingSet=CAP_DAC_OVERRIDE
> ReadWritePaths=-/etc/usbguard/rules.conf
> ```

The latter is already in the usbguard.service file: https://github.com/USBGuard/usbguard/blob/master/usbguard.service.in#L24 (fffd3d3fc16ff3426e95350b061a44f0ce02d3c7)

diabonas, 2020-02-09:
Regarding the necessary `CAP_DAC_OVERRIDE` capability, see #289 for more information and an alternative to adding this potentially dangerous capability.