USBGuard/usbguard #288: Debian 9 amd64 terminate called without an active exception

rgmalagon, 2019-04-23:


tweksteen, 2019-06-12:
Could you please provide more details on this issue? How did you trigger this message? Which version of USBGuard are you using? Could you share your `usbguard-daemon.conf`?

rgmalagon, 2019-06-12:
When install from debian 9 official repository usbguard 0.6.2 amd64, can not execute program, always show terminate called without an active exception.
usbguard-daemon.conf
//------------------------
RuleFile=/usr/local/etc/usbguard/rules.conf
ImplicitPolicyTarget=block
PresentDevicePolicy=keep
PresentControllerPolicy=keep
IPCAllowedUsers=root
IPCAllowedGroups=wheel
DeviceRulesWithPort=false
//--------------------------

rgmalagon, 2019-06-12:
i fix the problem compiling from source code to 32 bit architecture and add  i386 architecture to amd64 system. 
I think it's a compatibility problem with 64 bit libraries, because the 32 bit version is working

closed 2019-06-28 by dkopecek