USBGuard/usbguard #289: usbguard-0.7.2-7.fc30.x86_64 or one of its dependencies now requires CAP_DAC_OVERRIDE

doverride, 2019-05-03:
Since today on Fedora Rawhide i noticed that usbguard needs CAP_DAC_OVERRIDE
It might be one of its dependencies that actually triggers this since i havent actually seen a usbguard update in fedora for some time.

Anyhow. I noticed that usbguard started to maintain its IPC /dev/shm objects in a directory:

```
[root@brutus ~]# ls -alZ /dev/shm
total 0
drwxrwxrwt.  3 root     root     sys.id:sys.role:fs.tmpfs.fs:s0      60 May  3 10:38 .
drwxr-xr-x. 20 root     root     sys.id:sys.role:fs.devtmpfs.fs:s0 4160 May  3 10:32 ..
drwxrwx---.  2 kcinimod kcinimod sys.id:sys.role:fs.tmpfs.fs:s0     160 May  3 10:34 qb-2728-1703-24-h4NOFB
```

However if you look at the permissions and ownership of this directory then you notice that `root` does not have access to this location.

This triggers a CAP_DAC_OVERRIDE:

```
May 03 10:34:06 brutus audit[1226]: AVC avc:  denied  { dac_override } for  pid=1226 comm="usbguard-daemon" capability=1  scontext=sys.id:sys.role:usbguard.daemon.subj:s0 tcontext=sys.id:sys.role:usbguard.daemon.subj:s0 tclass=capability permissive=1
May 03 10:34:06 brutus audit[1226]: AVC avc:  denied  { dac_read_search } for  pid=1226 comm="usbguard-daemon" capability=2  scontext=sys.id:sys.role:usbguard.daemon.subj:s0 tcontext=sys.id:sys.role:usbguard.daemon.subj:s0 tclass=capability permissive=1
May 03 10:34:06 brutus audit[1226]: AVC avc:  denied  { rmdir } for  pid=1226 comm="usbguard-daemon" name="`qb-1208-1703-24-pUgZpU" dev="tmpfs" ino=39514 scontext=sys.id:sys.role:usbguard.daemon.subj:s0 tcontext=sys.id:sys.role:fs.tmpfs.fs:s0 tclass=dir permissive=1
May 03 10:34:06 brutus systemd[1]: Stopping USBGuard daemon...
```

If you fix the ownership and permissions of the "/dev/shm/qb-1208-1703-24-pUgZpU" directory, then usbguard-daemon does not need access to CAP_DAC_OVERRIDE. Needless to say that this would be a big security improvement.

diabonas, 2020-02-09:
Fedora has addressed this in usbguard-0.7.6-4.fc31, see [rhbz#1776357](https://bugzilla.redhat.com/show_bug.cgi?id=1776357), using the following patch: [`usbguard-0.7.6-libqb.patch`](https://src.fedoraproject.org/rpms/usbguard/blob/62377fbd34991902a17f1c04495395a2e04d9cc9/f/usbguard-0.7.6-libqb.patch). Unfortunately the patch never seems to have been submitted here :( @alakatos, could you create a pull request to get this fixed in upstream USBGuard as well, please?

radosroka, 2020-02-10:
> Fedora has addressed this in usbguard-0.7.6-4.fc31, see [rhbz#1776357](https://bugzilla.redhat.com/show_bug.cgi?id=1776357), using the following patch: [...]

Hi @diabonas, 
this patch is fedora specific hack to avoid using this capability and I would never merge it in upstream. The issiue has to be fixed in libqb.

diabonas, 2020-02-10:
@alakatos I see, [`qb_ipcs_connection_auth_set`](https://github.com/USBGuard/usbguard/blob/50c99211d512b963a8bcf271e316b65b967ab585/src/Library/IPCServerPrivate.cpp#L261) from libqb does not work as expected. Thank you for your insight, I have opened ClusterLabs/libqb#381 to address the issue there.

diabonas, 2020-02-12:
This has been fixed upstream, so waiting for the next libqb release or backporting the fix in ClusterLabs/libqb#382 will resolve the issue.

radosroka, 2020-02-12:
> This has been fixed upstream, so waiting for the next libqb release or backporting the fix in [ClusterLabs/libqb#382](https://github.com/ClusterLabs/libqb/pull/382) will resolve the issue.

Thank you for your effort, these are great news! I will create fedora bugzilla for that so they will have to put it into some future update.

diabonas, 2020-05-20:
> This has been fixed upstream, so waiting for the next libqb release or backporting the fix in [ClusterLabs/libqb#382](https://github.com/ClusterLabs/libqb/pull/382) will resolve the issue.

The patch has been included in libqb [1.0.6](https://github.com/ClusterLabs/libqb/releases/tag/v1.0.6) as well as the new major release [2.0.0](https://github.com/ClusterLabs/libqb/releases/tag/v2.0.0), upgrading to either of these will fix the issue.

closed 2020-05-20 by radosroka