USBGuard/usbguard #295: Numerous tests fail

jean-christophe-manciot, 2019-05-30:
Ubuntu 19.10
usbguard usbguard-0.7.4-70-gf011198
g++ .3.0-1ubuntu3

Usbguard has been configured from sources with:
```
./configure                             --build=x86_64-pc-linux-gnu \
                                        --enable-asan \
                                        --enable-debug-build \
                                        --enable-full-test-suite \
                                        --enable-systemd \
                                        --with-bundled-catch \
                                        --with-bundled-pegtl \
                                        --with-crypto-library=gcrypt \
                                        --with-dbus \
                                        --with-gui-qt=qt5 \
                                        --with-ldap \
                                        --with-polkit \
                                        --prefix=/usr --sysconfdir=/etc --localstatedir=/var
```
```make``` leads to:
```
...
============================================================================
Testsuite summary for usbguard 0.7.4
============================================================================
# TOTAL: 16
# PASS:  8
# SKIP:  0
# XFAIL: 0
# FAIL:  8
# XPASS: 0
# ERROR: 0
============================================================================
See src/Tests/test-suite.log
Please report to usbguard@usbguard.org
============================================================================
```
[test-suite.log](https://github.com/USBGuard/usbguard/files/3236163/test-suite.log)

jean-christophe-manciot, 2019-07-03:
Same issue with ```usbguard-0.7.5```:
```
PASS: test-unit
PASS: test-regression
PASS: USB/test-descriptor-parser.sh
PASS: Rules/test-rules.sh
FAIL: Source/check-driver.sh
PASS: UseCase/000_executable.sh
SKIP: UseCase/001_cli_policy.sh
FAIL: UseCase/002_cli_devices.sh
SKIP: UseCase/003_cli_devices_umockdev.sh
PASS: UseCase/004_daemonize.sh
PASS: LDAP/Sanity/ldap-nsswitch.sh
FAIL: LDAP/UseCase/ldap-test-1.sh
FAIL: LDAP/UseCase/ldap-test-2.sh
FAIL: LDAP/UseCase/ldap-test-3.sh
PASS: LDAP/UseCase/ldap-test-4.sh
FAIL: LDAP/UseCase/ldap-test-5.sh
============================================================================
Testsuite summary for usbguard 0.7.5
============================================================================
# TOTAL: 16
# PASS:  8
# SKIP:  2
# XFAIL: 0
# FAIL:  6
# XPASS: 0
# ERROR: 0
============================================================================
See src/Tests/test-suite.log
Please report to usbguard@usbguard.org
============================================================================
```
[test-suite.log](https://github.com/USBGuard/usbguard/files/3354327/test-suite.log)

BoneTone, 2019-11-30:
@jean-christophe-manciot just curious if this repros with 0.76.  It only came out a few weeks ago, so understandable if you haven't had a chance to test yet.  I'd try it myself but I'm in dealing with a hardware issue on my main machine, hopefully I will get it fixed soon as I need it to use USBGuard which I really want to try since I was just gifted 3 USB drives of less-than-known origin.

jean-christophe-manciot, 2019-12-02:
I had issues when testing ```0.7.6```, but only with LDAP tests because it requires a special LDAP server setup. Cf. [All LDAP tests fail](https://github.com/USBGuard/usbguard/issues/345#issue-525741078).

closed 2019-12-02 by jean-christophe-manciot