USBGuard/usbguard #299: Yubikey with Usbguard

RaNdOm0313, 2019-06-12:
I'm using the yubikey 5 with usbguard set up on my laptop. It seems that usbguard first blocks the device before checking the rules and then permits after checking. It kinda screws up the yubikey for some reason where the system can't read it anymore.

Using the IPC watch function...

tweksteen, 2019-06-19:
This is by design. As USBGuard is interfacing with sysfs, we need to have a sane default which gets adjusted once the device is authorized. The other way around (default allow -> block) would leave an obvious gap.

For your specific problem, anything from `dmesg` that you could share? Does the key get recognised at all? Or only part of the functions are available?

RaNdOm0313, 2019-06-25:
---START---
[   39.322675] usb 1-3: Product: YubiKey OTP+FIDO+CCID
[   39.322679] usb 1-3: Manufacturer: Yubico
[   39.323018] usb 1-3: Device is not authorized for usage
[   39.327754] usb 1-3: authorized to connect
[   39.347391] hidraw: raw HID events driver (C) Jiri Kosina
[   39.354905] usbcore: registered new interface driver usbhid
[   39.354906] usbhid: USB HID core driver
[   39.358224] input: Yubico YubiKey OTP+FIDO+CCID as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:1050:0407.0001/input/input21
[   39.414062] hid-generic 0003:1050:0407.0001: input,hidraw0: USB HID v1.10 Keyboard [Yubico YubiKey OTP+FIDO+CCID] on usb-0000:00:14.0-3/input0
[   39.414795] hid-generic 0003:1050:0407.0002: hiddev0,hidraw1: USB HID v1.10 Device [Yubico YubiKey OTP+FIDO+CCID] on usb-0000:00:14.0-3/input1
---END---

Device gets recognised but because it gets blocked first for some reason the CCID function ceases to function...

dkopecek, 2019-06-28:
@RaNdOm0313 please try restarting the pcscd daemon after you authorize the yubikey and let us know if that helps

RaNdOm0313, 2019-06-28:
> @dkopecek Sadly that didn't work. Still doesn't get recognised

k3an3, 2020-09-13:
Any updates on this issue? I may be having the same problem.

RaNdOm0313, 2020-11-02:
May have found a solution. Usbguard asbolutely refused to apply the "allow" rule so while testing recently(about a year after the issue started...oops) i figured out that the main problem lies with usbguard being unable to write in the /etc/usbguard/rules.conf file.

Even if you change the rights to for example 660 usbguard fails to start. Will try to figure something out ASAP...

closed 2020-11-02 by RaNdOm0313

Cropi, 2020-11-03:
Have you tried to set the permission of /etc/usbguard/rules.conf to 600?
What about file ownership?