USBGuard/usbguard #308: USBGuard not working with new kernel in Arch

ibahnasy, 2019-06-26:
Arch Linux with kernel: 5.1.15-arch1-1-ARCH.

After starting the service unit:
# usbguard list-devices 
ERROR: IPC send/recv: connection: Timed out while waiting for IPC reply

# systemctl status usbguard
● usbguard.service - USBGuard daemon
   Loaded: loaded (/usr/lib/systemd/system/usbguard.service; disabled; vendor preset: disabled)
   Active: deactivating (stop-sigkill) (Result: timeout) since Wed 2019-06-26 11:55:51 EET; 1min 30s ago
     Docs: man:usbguard-daemon(8)
  Process: 1684 ExecStart=/usr/bin/usbguard-daemon -f -s -c /etc/usbguard/usbguard-daemon.conf (code=exited, status=0/SUCCESS)
 Main PID: 1686 (usbguard-daemon)
    Tasks: 2 (limit: 4915)
   Memory: 6.2M
   CGroup: /system.slice/usbguard.service
           └─1686 [usbguard-daemon]

Jun 26 11:55:17 myarch usbguard-daemon[1686]: Cannot resolve realpath for /sys/bus/usb/devices/../../../devices/pci0000:00/0000:00:14.0/usb1/1->
Jun 26 11:55:17 myarch usbguard-daemon[1686]: Cannot resolve realpath for /sys/bus/usb/devices/../../../devices/pci0000:00/0000:00:14.0/usb1/1->
Jun 26 11:55:17 myarch usbguard-daemon[1686]: Cannot resolve realpath for /sys/bus/usb/devices/../../../devices/pci0000:00/0000:00:14.0/usb1/1->
Jun 26 11:55:17 myarch usbguard-daemon[1686]: Cannot resolve realpath for /sys/bus/usb/devices/../../../devices/pci0000:00/0000:00:14.0/usb1/1->
Jun 26 11:55:17 myarch usbguard-daemon[1686]: Cannot resolve realpath for /sys/bus/usb/devices/../../../devices/pci0000:00/0000:00:14.0/usb1/1->
Jun 26 11:55:17 myarch usbguard-daemon[1686]: Cannot resolve realpath for /sys/bus/usb/devices/../../../devices/pci0000:00/0000:00:14.0/usb1/1->
Jun 26 11:55:17 myarch usbguard-daemon[1686]: device enumeration exception: /sys/bus/usb/devices/1-8:1.0: SysFSDevice: uevent: No such file or >
Jun 26 11:55:51 myarch systemd[1]: Stopping USBGuard daemon...
Jun 26 11:57:21 myarch systemd[1]: usbguard.service: State 'stop-sigterm' timed out. Killing.
Jun 26 11:57:21 myarch systemd[1]: usbguard.service: Killing process 1686 (usbguard-daemon) with signal SIGKILL.

qinohe, 2019-06-26:
Arch user here ;-)
uname -r 5.1.15-arch1-1-ARCH
usbguard is running just fine here, I guess you didn't create a policy first!
If you don't create a policy, than start usbguard, and than try to list-devices with usbguard the IPC error is thrown at you standard.
If you have created a policy and still this happens, you need to strace and consult the maker, though my guts say it's the former, see 'man usbguard'.

ibahnasy, 2019-06-26:
You are right, it was because of the policy, thanks!

closed 2019-06-26 by ibahnasy