USBGuard/usbguard #314: PresentDevicePolicy: allow combination of keep and apply-policy

bschacht, 2019-07-16:
Hi,

I think it would be great if `PresentDevicePolicy` would allow a combination of `keep` and `apply-policy`. When setting the `PresentDevicePolicy` to `keep` and then adding a rule to block a device to the rulefile, a reboot with the device attached would allow the device. A combination of `keep` and `apply-policy` would then only fallback to `keep` if there is no policy defined for the device.

muelli, 2019-07-17:
Intersting. Is `ImplicitDevicePolicy` evaluated at all when `PresentDevicePolicy` is used?

bschacht, 2019-07-17:
> Interesting. Is `ImplicitDevicePolicy` evaluated at all when `PresentDevicePolicy` is used?

There is no `ImplicityDevicePolicy`, i guess you mean `ImplicitPolicyTarget`? Its not evaluated, as far as I can say from my tests. (I have `ImplicitPolicyTarget=block` and `PresentDevicePolicy=keep` and one device blocked in the rule file. After a reboot `usbguard list-rules` lists the rules correct, but `usbguard list-devices` lists them all as allowed)

I guess I would expect a behavior like in some firewall solutions, with the most specific rule being the one applied.

tweksteen, 2019-07-25:
This is the context around this question: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=928032.

Generally, to achieve this behaviour, you would set `PresentDevicePolicy=apply-policy` and add your rules (rules.conf) which may fallback on `ImplicitPolicyTarget`.

I'm actually wondering if USBGuard should drop the `keep` option. By definition, it decides on which devices are authorised or blocked. Leaving a "floating" state like that just bring confusion on what to expect and how to handle it. (This would mean move the `PresentControllerPolicy` to `allow`).

tweksteen, 2019-09-16:
@dkopecek and @radosroka What do you think our approach should be here? Thanks

radosroka, 2019-09-16:
I can see a benefit of ```keep``` option in respecting device in ```blocked``` state.

> This is the context around this question: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=928032.
>
> Generally, to achieve this behaviour, you would set `PresentDevicePolicy=apply-policy` and add [...]

But it does make sense in the way how you described it.