USBGuard/usbguard #315: Race condition at boot time: unknown sysfs path

tweksteen, 2019-07-17:
I believe there is some kind of race condition (happened at boot time), where the order of processing of uevents does not necessarily matches the dependencies between devices. Here is the log I have (extracted with what I believe is useful, note the time at the beginning of the line):

```
00:48:19.699401 kernel: [    8.800316] usb 2-4.3: New USB device found, idVendor=xxx
00:48:20.608843 usbguard-daemon: uid=0 pid=924 device.rule='allow id 1d6b:0002 serial "0000:00:14.0" name "xHCI Host Controller" hash "" parent-hash "" via-port "usb2" with-interface 09:00:00' type='Device.Present' result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:14.0/usb2'
00:48:20.610275 usbguard-daemon: Device insert exception: UEventDeviceManager: /devices/pci0000:00/0000:00:14.0/usb2/2-4: unknown sysfs path
00:48:20.610383 usbguard-daemon: UEventDeviceManager: /devices/pci0000:00/0000:00:14.0/usb2/2-4: unknown sysfs path
00:48:20.610490 usbguard-daemon: Rejecting device at syspath=/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4.3
00:48:20.610993 usbguard-daemon: USB Device Exception: SysFSDevice: (_sysfs_dirfd = open((G_sysfs_root + _sysfs_path).c_str(), O_PATH\|O_DIRECTORY)) < 0: No such file or directory
00:48:20.611101 usbguard-daemon: SysFSDevice: (_sysfs_dirfd = open((G_sysfs_root + _sysfs_path).c_str(), O_PATH\|O_DIRECTORY)) < 0: No such file or directory
00:48:20.671264 usbguard-daemon: uid=0 pid=924 device.rule='allow id 05e3:0610 serial "" name "USB2.0 Hub" hash "" parent-hash "" via-port "2-4" with-interface { 09:00:01 09:00:02 }' type='Device.Present' result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:14.0/usb2/2-4'
```

It seems that device 2-4.3 is trying to get authorised but the parent sysfs path is not present. This triggers an exception to reject the device (`unknown sysfs path`). The rejection triggers another exception, because the filepath is still missing (`No such file or directory`). Finally, the intermediate missing hub (2-4) is authorised.

Why is 2-4.3 not finding their parent sysfs path (2-4)? 
1. Assuming that even if 2-4 is not authorised, its sysfs path should exist. Then, are we sure that the sysfs path for any device will exist as soon as the "add" uevent  is received? Could there be some delay here?
2. If the sysfs path does not exist because not authorised yet, does USBGuard do (or should do) any kind of ordering of the uevents received to ensure that roots are processed before leaves?

The policy for this case is an allow all. I have another similar behaviour that I'll report in a separate bug. Any input is appreciated.

dkopecek, 2019-07-17:
This looks like a bug in the enumeration procedure. sysfs paths should be triggered in order from parent to children, so that the uevents arrive in correct order. If the parent gets rejected/blocked then the subsequent paths in queue should be ignored. 

The relevant locations are these I think:

Compare function:
https://github.com/USBGuard/usbguard/blob/master/src/Library/UEventDeviceManager.cpp#L678

Enumeration:
https://github.com/USBGuard/usbguard/blob/master/src/Library/UEventDeviceManager.cpp#L707

Looking at the code now, I think that there needs to be a check for sysfs_path existence at https://github.com/USBGuard/usbguard/blob/master/src/Library/UEventDeviceManager.cpp#L600

tweksteen, 2019-07-17:
By reading the code, this is my understanding of what is happening:
1. sysfs is enumerated (via `scan()`), making sure that roots are parsed first (I couldn't find any issue there). For each device found, `ueventEnumerateTriggerDevice` is called: it adds the device to the list of known syspathes. It then triggers an artificial `add` event by [writing](https://github.com/USBGuard/usbguard/blob/8bace7af0417787d525eab1cb74f44e7e53a366d/src/Library/UEventDeviceManager.cpp#L793) to the `uevent` file of the device in sysfs (acting as if the device just got inserted).
2. This `add` event gets picked up by a separate thread that process the device.
3. There is some kind of synchronisation between the scan()-thread and the uevent thread: the condition_variable `enumeration_complete` is used to wait for the uevent to be processed before continuing (or timeout after 2 seconds).

This works fine under the assumption that no uevent will be generated during the scan() that are not originating from USBGuard. I believe this assumption is wrong and is what triggers this issue.

I've managed to reproduce artificially this issue by constantly writing to an arbitrary usb device `uevent` file while USBGuard is starting. To reproduce,

1. Disable the origin check in [ueventProcessRead](https://github.com/USBGuard/usbguard/blob/8bace7af0417787d525eab1cb74f44e7e53a366d/src/Library/UEventDeviceManager.cpp#L556). 
2. Pick a USB device (ideally, far from the root, last usb hub, or hub behind a hub). In its sysfs directory:
`# while :; do echo "add" > uevent; done;`
3. Start USBGuard with debug mode enabled: `# usbguard-daemon -d -l log`
4. Observe the unknown sys path error.

I therefore think this is a design issue. Is there a particular reason for generating an artificial `add` uevent instead of calling the internal function directly? We might be able to solve that by only processing uevent originating from our own pid while scan() is running (we would need to add some extra variable to let the uevent thread know when scan() finishes).

tweksteen, 2019-07-17:
I can see that the `_enumeration` Restorer object is already in use to let the uevent processing thread we are still in the `scan()` method. 

Generally, what should we do when we see a new device added while we are still doing the initial scanning? We can't ignore it nor try to process it until the necessary parents are set up. Should we somehow store it for later parsing once the enumeration is done?

dkopecek, 2019-07-18:
> I therefore think this is a design issue. Is there a particular reason for generating an artificial `add` uevent instead of calling the internal function directly? We might be able to solve that by [...]

The intention was to use the same execution path for both "present" and "inserted" devices. It seems that the "present" and "inserted" device state distinction is what makes this design ugly, but I thought it would make sense (from a security policy and auditing point of view) to have this distinction available.

> Generally, what should we do when we see a new device added while we are still doing the initial scanning? We can't ignore it nor try to process it until the necessary parents are set up. Should we somehow store it for later parsing once the enumeration is done?

Yeah, I think you are pointing to the right direction here, we have to queue these events and process them after the enumeration is done.

tweksteen, 2019-08-07:
Fixed in #318

closed 2019-08-07 by tweksteen