USBGuard/usbguard #325: How to get internal devices ignored?

bitstreamout, 2019-08-08:
I'd like not to get messages after e.g. Lid open of my notebook on USB devices like Bluetooth or finger print scanner ... something like an option to tell the applet not to notify for this kind of devices (also useful keyboard, mouse, ...)

userWayneCampbell, 2019-08-08:
The PoC applet isn't a part of the main project anymore: https://github.com/USBGuard/usbguard/commit/bf3bb09edf491892617d35b55edec06f92d2743f

bitstreamout, 2019-08-08:
Hmmm ... without an applet or similar it becomes rather useless

radosroka, 2019-08-08:
AFAIK there was some ongoing work on gnome upstream that should integrate The USBGuard.

https://gitlab.gnome.org/GNOME/gnome-settings-daemon/merge_requests/75

bitstreamout, 2019-08-09:
I do not use GNOME (user unfriendly) nor KDE (overloaded) but lxpanel with fvwm.  What is the aim of only supporting GNOME?

radosroka, 2019-08-28:
@bitstreamout btw you can try to use the usbguard-notifier. It can provide you notifications from the USBGuard.

https://github.com/Cropi/usbguard-notifier

closed 2019-08-28 by radosroka