USBGuard/usbguard #328: list-rules: Accessing an empty attribute

RyuzakiKK, 2019-08-30:
Steps to reproduce:
```
$ sudo usbguard append-rule 'allow label "my_label"'
17
$ sudo usbguard list-rules
17: allow label "my_label"
$ sudo usbguard list-rules -l "my_label"            
IPC ERROR: request id=1: IPC method: usbguard.IPC.listRules: BUG: Accessing an empty attribute
```

I'm using USBGuard 0.7.5 with `/etc/usbguard/` appended to `ReadWritePaths` because of https://github.com/USBGuard/usbguard/issues/284

radosroka, 2019-12-06:
Is this still reproducible with https://github.com/USBGuard/usbguard/pull/348/commits/8b6b47f8531c981c88008947f944d59af7d3893b ?

RyuzakiKK, 2019-12-29:
@radosroka I can confirm that with the 8b6b47f commit this bug is now fixed.

RyuzakiKK, 2020-01-12:
The other day it surely worked but now I'm keep getting the same "Accessing an empty attribute" error, even with https://github.com/USBGuard/usbguard/commit/8b6b47f8531c981c88008947f944d59af7d3893b.

muelli, 2020-01-31:
I have tried to check whether https://github.com/USBGuard/usbguard/commit/8b6b47f8531c981c88008947f944d59af7d3893b fixes this problem, but it does not seem to be the case:

```
ubuntu@usbuntu:~$ cat /usr/lib/systemd/system/usbguard.service 
[Unit]
Description=USBGuard daemon
Wants=systemd-udevd.service local-fs.target
Documentation=man:usbguard-daemon(8)

[Service]
AmbientCapabilities=
CapabilityBoundingSet=CAP_CHOWN CAP_FOWNER CAP_DAC_READ_SEARCH CAP_DAC_OVERRIDE
DeviceAllow=/dev/null rw
DevicePolicy=strict
ExecStart=/usr/sbin/usbguard-daemon -k -c /etc/usbguard/usbguard-daemon.conf
IPAddressDeny=any
LockPersonality=yes
MemoryDenyWriteExecute=yes
NoNewPrivileges=yes
PrivateDevices=yes
PrivateTmp=yes
ProtectControlGroups=yes
ProtectHome=yes
ProtectKernelModules=yes
ProtectSystem=yes
PIDFile=/run/usbguard.pid
ReadOnlyPaths=-/
ReadWritePaths=-/dev/shm -/var/log/usbguard -/tmp -/etc/usbguard/rules.conf -/etc/usbguard/
Restart=on-failure
RestrictAddressFamilies=AF_UNIX AF_NETLINK
RestrictNamespaces=yes
RestrictRealtime=yes
SystemCallArchitectures=native
SystemCallFilter=@system-service
Type=simple
UMask=0077

[Install]
WantedBy=basic.target
ubuntu@usbuntu:~$ sudo systemctl reload
reload             reload-or-restart  
ubuntu@usbuntu:~$ sudo systemctl daemon-reload 
ubuntu@usbuntu:~$ sudo systemctl restart usbguard
ubuntu@usbuntu:~$ sudo usbguard append-rule 'allow label "my_label"'
10
ubuntu@usbuntu:~$ sudo usbguard list-rules
1: allow id 1d6b:0002 serial "0000:00:1d.7" name "EHCI Host Controller" hash "9JszuPWESBEur6P34acG7ZtBUqRa6fRgdsma6hSNpOk=" parent-hash "S0B+OWTcDg6PfHFlF0chk7joLm2raiSYdhNb1N1/BOk=" with-interface 09:00:00 with-connect-type ""
2: allow id 1d6b:0001 serial "0000:00:1d.0" name "UHCI Host Controller" hash "bjtVSUusIDxj13VXcZojsuMRNykZ7U/PLmXWG7xPhDo=" parent-hash "jW2YTPWRLeQOE7Q8I2f0pdN13zFYXVQGQoNmr1gDZgg=" with-interface 09:00:00 with-connect-type ""
3: allow id 1d6b:0001 serial "0000:00:1d.1" name "UHCI Host Controller" hash "Z0pthcq0nFBS1408iQbne9dbBzJ5DDg5j3DOTFBO+C0=" parent-hash "1f6hp0G45b+z7zSlP4CX4HXDdlrldzpbVhwZ1h7zsjI=" with-interface 09:00:00 with-connect-type ""
4: allow id 1d6b:0001 serial "0000:00:1d.2" name "UHCI Host Controller" hash "BKvEUUpIAivwthZQbvlpoMynYY8ZgHaR3v2nUupxuIE=" parent-hash "nKOQytSaVkGSK63fd4ahzmPpgWm2LNkuHaRNyk+xbkg=" with-interface 09:00:00 with-connect-type ""
5: allow label "my_label"
10: allow label "my_label"
ubuntu@usbuntu:~$ sudo usbguard list-rules -l "my_label"
IPC ERROR: request id=1: IPC method: usbguard.IPC.listRules: BUG: Accessing an empty attribute
ubuntu@usbuntu:~$ 
```

PR #409 (2020-08-28): Fix rule filtering by label
Calling usbguard list-rules with option --label fails if at least one rule does not have label attribute set.

closes #328
closed 2020-08-28 by radosroka