USBGuard/usbguard #330: [usbguard-applet-qt] Showing two dialogs for one device.

vesim987, 2019-09-15:
Hi
usbguard-applet-qt is showing two dialogs for one device connected via USB-C 3.0, but only one instance of that device in the GUI and CLI.
[image]
That might be some bug in the USB driver, because it is showing in dmesg that the device was added two times:
```
[46531.581214] xhci_hcd 0000:3b:00.0: xHCI Host Controller
[46531.581221] xhci_hcd 0000:3b:00.0: new USB bus registered, assigned bus number 3
[46531.582362] xhci_hcd 0000:3b:00.0: hcc params 0x200077c1 hci version 0x110 quirks 0x0000000000009810
[46531.582606] usb usb3: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.02
[46531.582608] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[46531.582609] usb usb3: Product: xHCI Host Controller
[46531.582610] usb usb3: Manufacturer: Linux 5.2.10-artix1-1-ARTIX xhci-hcd
[46531.582611] usb usb3: SerialNumber: 0000:3b:00.0
[46531.582742] hub 3-0:1.0: USB hub found
[46531.582753] hub 3-0:1.0: 2 ports detected
[46531.583344] probe of 3-0:1.0 returned 1 after 609 usecs
[46531.583365] probe of usb3 returned 1 after 670 usecs
[46531.583452] xhci_hcd 0000:3b:00.0: xHCI Host Controller
[46531.583457] xhci_hcd 0000:3b:00.0: new USB bus registered, assigned bus number 4
[46531.583460] xhci_hcd 0000:3b:00.0: Host supports USB 3.0 SuperSpeed
[46531.583513] usb usb4: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.02
[46531.583515] usb usb4: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[46531.583517] usb usb4: Product: xHCI Host Controller
[46531.583519] usb usb4: Manufacturer: Linux 5.2.10-artix1-1-ARTIX xhci-hcd
[46531.583521] usb usb4: SerialNumber: 0000:3b:00.0
[46531.583673] hub 4-0:1.0: USB hub found
[46531.583685] hub 4-0:1.0: 2 ports detected
```
This can also be fixed on the usbguard side, but i don't think that this is an good idea.

tweksteen, 2019-09-16:
The output of dmesg is matching what is displayed within the applet. That is, 2 hubs have just been added (you can tell that it is unlikely to be a duplicate because they have a different idProduct, reported by the device). You can see that this has happened over PCIe (`0000:3b:00.0`), so I'm going to guess that you've plugged in a Thunderbolt hub or docking station. In this case, it is normal to see multiple USB hubs over Thunderbolt being reported.

> [...] only one instance of that device is in the GUI and CLI 

Could you please share more details here? Could you also share the output of `usbguard list-devices`? Thanks

closed 2019-09-20 by tweksteen

AreYouLoco, 2019-09-21:
I am reopening. Bit more info here.

I have systemd service installed and running and when it's only this service open I do get one notification/popup correctly once for the particular device.

But when I have systemd service running and usbguard-applet-qt running connected to IPC at the same time I do get two notifications for the same device. It's the same bus for sure. Same everything.

Solution would be to send notification by only one of the services at the time. Decide which one, choose which one or even better detect which is better at the time.

tweksteen, 2019-09-24:
The systemd service itself would not create any popup. You may already have the applet running in the background. If you run another instance, it is somewhat expected to have 2 notifications for the same event.

Try to reproduce your first case and list the current processes running.

AreYouLoco, 2019-09-24:
@tweksteen Killed all qt-applets. Runned one. Cannot reproduce anymore. You were right!