USBGuard/usbguard #334: Usbguard applet

aelth, 2019-10-09:
Please bring back the Qt applet - it was much easier and more user-friendly to quickly allow/block the device than using the CLI.
Or at least make it a separate project that can be used as a GUI for usbguard.

Thanks in advance!

muelli, 2019-10-09:
FTR: There is https://github.com/6E006B/usbguard-gnome

And GNOME will most likely have a built-in frontend in the next release (3.36).

aelth, 2019-10-10:
Thanks muelli, I'm aware of that applet, but unfortunately I'm not using GNOME as my WM:(

tammar96, 2019-11-05:
Hi guys, check this out! https://github.com/Cropi/usbguard-notifier

Lem, 2019-11-06:
As @dkopecek mentioned in his commit, it should be in a separate repo. Maybe he can give us some information?

calestyo, 2020-01-22:
It would have been much better to leave the Qt applet as is, until other solutions are available in all distributions. Right now and for it's probable main purpose (desktop systems), usbguard seems pretty limited by this.

pinotree, 2020-02-18:
> As @dkopecek mentioned in his commit, it should be in a separate repo. Maybe he can give us some information?

Indeed -- Daniel, can you please upload the repository with the Qt applet? In case it is needed, I can help to create it from the usbguard.git repository.

kisst, 2020-06-16:
My quick and dirty script as an easy to use "gui" https://gist.github.com/kisst/3932f2cc30281915f8ec5be7deed98b3

calestyo, 2020-07-03:
Anything new on this?

munix9, 2020-07-23:
Based on the work of @pinotree (https://github.com/pinotree/usbguard-applet-qt - thanks!) and my own changes, I have created packages for openSUSE Tumbleweed (https://build.opensuse.org/package/show/home:munix9/usbguard-applet-qt).
Runs so far, but is not yet fully tested.
I can't make a final statement about security, maybe an additional password protection would be useful (paranoia setting).

pinotree, 2020-07-23:
I actually meant to mention my repo, just that I forgot to do that.
As @munix9 noticed, I created https://github.com/pinotree/usbguard-applet-qt, which provides the applet combining all the git history available for it:
- Qt 5 only (Qt 4 support is dropped)
- builds with cmake
- still uses the libusbguard library, although I already started to work to get rid of that dependency (it communicates with usbguard-dbus)
  - at least in https://bugs.debian.org/961374 it was mentioned that this library is not stable; also this bug means you cannot built the applet on Debian distros currently

I need to write a proper README for it...

radosroka, 2020-07-23:
> I actually meant to mention my repo, just that I forgot to do that.
> As @munix9 noticed, I created https://github.com/pinotree/usbguard-applet-qt, which provides the applet combining all the git [...]

Thanks for your initiative. I believe we are close to 1.0.0 release and stable API.

pinotree, 2020-07-23:
> Thanks for your initiative. I believe we are close to 1.0.0 release and stable API.

Oh, and of course it is fine for me if you want to take the applet back within the usbguard project. However, judging from the past happenings, I'd think that this is not wanted. To be clear, I have no problem either way.

genodeftest, 2020-07-23:
@munix9 wrote
>  I can't make a final statement about security, maybe an additional password protection would be useful (paranoia setting).

I wouldn't do that as it provides no additional security. If your GUI application can access the DBus service, so can any other application written to use DBus. Everyone who can run processes with your permission can do that.

csoler, 2020-10-07:
I just tried the CLI and it doesn't work:

> usbguard allow-device 059f:1027
IPC ERROR: request id=1: Device lookup: device id: id doesn't exist

I just dont want to waste time with this. It's a shame the applet has been removed. So I'm uninstalling usbguard.

muelli, 2020-10-07:
okay. There are a few alternatives that you might consider exploring, e.g. [usbauth](https://github.com/kochstefan/usbauth-all). For the future, I'd be helpful to stick on-topic and avoid inflammatory language to keep everybody focussed on the issue at hand. Thanks.

radosroka, 2020-10-07:
> I just tried the CLI and it doesn't work:
> 
> > usbguard allow-device 059f:1027
> > IPC ERROR: request id=1: Device lookup: device id: id doesn't exist
> 
> I just dont want to waste time with this. It's a shame the applet has been removed. So I'm uninstalling usbguard.

The applet is now in different location:

https://github.com/pinotree/usbguard-applet-qt

it is not part of the usbguard anymore...

When you want to allow a device you need an id. You can get it by listing devices.

```
$ usbguard list-devices # optionally --blocked
...
25: block id 17ef:6044 serial "" name "ThinkPad USB Laser Mouse" hash "3jp26uF2tSyJrW9tvgnCnniDb2Q4nxKAe/jKFcOGPuA=" parent-hash "1bfHz4/5nO4aYIwQG5Ci/F/9HBCKCPOdq/1eoUswB0M=" via-port "3-9.2" with-interface 03:01:02 with-connect-type "unknown"
...
$ usbguard allow-device 25
```

kisst, 2020-10-07:
@muelli don't give up, you still have easy options like https://github.com/USBGuard/usbguard/issues/334#issuecomment-644663738 or the external qt libs.

csoler, 2020-10-07:
Arh. Sorry about the ID. It wasn't very clear in the documentation.
When you see "15: block id 059f:1027 serial "S170628000000114", it's not clear that the ID is 15 rather than 059...
Thx anyway for the quick response. I'll stick to the CLI for now.

MatejKovacic, 2022-11-01:
Hi guys, anything new on this? I am using command line commands, this is not a problem for me. But for many users it is. And i believe those users also need protection and security, i. e. this application is intended to them. It would be great if there was some good GUI for them.