USBGuard/usbguard #337: Whitelisting allows classes of USB drives, not specific devices

jasonwryan, 2019-10-23:
We purchased 200 branded USB drives to use for marketing.

A random sample of twelve devices from the 200 sees them all registered withy exactly the same ID:

```
usbguard list-devices -b                                                                                                  
31: block id abcd:1234 serial "\xd0\x89" name "UDisk           " hash "WwLIxRVjks6yVS9HHhP1Vn3rryx6WklV8RXeIlNoXag=" parent-ha
sh "kv3v2+rnq9QvYI3/HbJ1EV9vdujZ0aVCQ/CGBYIkEB0=" via-port "2-1.1" with-interface 08:06:50                                    

usbguard list-devices -b
32: block id abcd:1234 serial "\xd0\x89" name "UDisk           " hash "WwLIxRVjks6yVS9HHhP1Vn3rryx6WklV8RXeIlNoXag=" parent-ha
sh "kv3v2+rnq9QvYI3/HbJ1EV9vdujZ0aVCQ/CGBYIkEB0=" via-port "2-1.1" with-interface 08:06:50
 
usbguard list-devices -b
33: block id abcd:1234 serial "\xd0\x89" name "UDisk           " hash "WwLIxRVjks6yVS9HHhP1Vn3rryx6WklV8RXeIlNoXag=" parent-ha
sh "kv3v2+rnq9QvYI3/HbJ1EV9vdujZ0aVCQ/CGBYIkEB0=" via-port "2-1.1" with-interface 08:06:50
```

Consequently, whitelisting any one of these devices allows any number of the others to also be mounted.

This is concerning because it means that, effectively, whitelisting a specific device potentially means inadvertently allowing a much larger number of devices to qualify.

userWayneCampbell, 2019-10-23:
If the devices all have the same vid, pid, and serial, then usbguard has nothing else to uniquely identify those devices. 

Really not usbguard's fault if all of the usb descriptors are the same.

jasonwryan, 2019-10-23:
Yes, I can see that. The fact that manufacturers don't use unique IDs, though, does significantly reduce the utility of a tool like this. 

And it makes me wonder how many people naively assumed, as I did, that the opposite was true.

closed 2019-11-05 by radosroka

anarcat, 2019-11-05:
shouldn't this have some sort of resolution? i made the same mistake here, and i think there should be clearer documentation on this problem, if not lobbying of the manufacturer themselves to solve this problem, maybe by having the USB spec to force manufacturers to use unique identifiers for each device...