USBGuard/usbguard #341: An idea for USBGuard if not already implemented (?)

D-signr, 2019-11-05:
Using this software sounded a bit complicated to me and i tried to figure out how could it have a simple and streamlined basic functions offering a good security level once enabled. I came up with this (sorry if you already have this implemented, but i did not find a notion about it from your web page after some searching):

1. An authorized user can prompt the USBGuard to read serial numbers from the currently physically connected usb devices and then it asks user to allow them (y/n) and if the timeout rule is used or not (y/n) for each of them.
2. An authorized user can define a connection timeout for allowed usb devices. Ports open automatically when correct sudo password is entered to the terminal, or alternatively after that when prompted (an option) and then close when the timeout value is reached.

Any thoughts? If it is already there and similarly simplified for an average user, then never mind this post. Or then it might be good to tell it clearly in the beginning of the instructions page.

tammar96, 2019-11-05:
Hi,

1.  https://github.com/Cropi/usbguard-notifier

2. timeout is good idea, we will consider it.

muelli, 2020-01-10:
can you please change the title to something more meaningful? Like "timeout for blocking of devices" or so.

equwal, 2021-01-12:
Seems pretty reasonable to do something like this:
1) First connection of a device is "instant."
2) After a device is connected, all device connections are ignored for a timeout. This can even be incredibly short to have a great effect, like 200ms, and prevent any kind of brute force attack. Another option is to do something like what pam does, and put a big timeout after three tries (I don't like this option as much, it is too jarring).

This should be a default feature, in my opinion, and you should be able to raise the timeout (which you might want to do especially if you think you are vulnerable to a brute force attack for some reason). 99.9% of users wouldn't notice any change in their system with this edit.