USBGuard/usbguard #345: All LDAP tests fail

jean-christophe-manciot, 2019-11-20:
Debian bullseye 11.x
usbguard usbguard-0.7.6
g++ .4:9.2.1-3.1

Usbguard has been configured from sources with:
```
./configure                             --build=x86_64-pc-linux-gnu \
                                        --enable-asan \
                                        --enable-debug-build \
                                        --enable-systemd \
                                        --with-bundled-catch \
                                        --with-bundled-pegtl \
                                        --with-crypto-library=gcrypt \
                                        --with-dbus \
                                        --with-gui-qt=qt5 \
                                        --with-ldap \
                                        --with-polkit \
                                        --prefix=/usr --sysconfdir=/etc --localstatedir=/var
```
```make``` leads to:
```
...
PASS: test-unit
PASS: test-regression
PASS: USB/test-descriptor-parser.sh
PASS: Rules/test-rules.sh
PASS: LDAP/Sanity/ldap-nsswitch.sh
FAIL: LDAP/UseCase/ldap-test-1.sh
FAIL: LDAP/UseCase/ldap-test-2.sh
FAIL: LDAP/UseCase/ldap-test-3.sh
FAIL: LDAP/UseCase/ldap-test-4.sh
FAIL: LDAP/UseCase/ldap-test-5.sh
============================================================================
Testsuite summary for usbguard 0.7.6
============================================================================
# TOTAL: 10
# PASS:  5
# SKIP:  0
# XFAIL: 0
# FAIL:  5
# XPASS: 0
# ERROR: 0
```
[test-suite.log](https://github.com/USBGuard/usbguard/files/3868912/test-suite.log)

radosroka, 2019-11-20:
This looks like a leak in the usbguard ldap client. You could work around this problem by omitting ```--enable-asan``` from configure command line.

radosroka, 2019-11-20:
But the real problem here is that ldap client is not able to connect to ldap server from some reason.

It looks like even the ldap utilities are not able to connect to the server to initialize it.
Have you looked into the .travis.yml? Do you have the ansible installed?

https://github.com/USBGuard/usbguard/blob/master/.travis.yml

jean-christophe-manciot, 2019-11-20:
Ansible is not installed because I use it from sources (with hacking/env-setup).
You may want to add this requirement to the [Compilation & Installation](https://github.com/USBGuard/usbguard#compilation--installation) guidelines.
Anyhow, installing ansible does not change the outcome: 
[test-suite.log](https://github.com/USBGuard/usbguard/files/3869084/test-suite.log)

Checking slapd status:
```
# systemctl status slapd
● slapd.service - LSB: OpenLDAP standalone server (Lightweight Directory Access Protocol)
   Loaded: loaded (/etc/init.d/slapd; generated)
  Drop-In: /lib/systemd/system/slapd.service.d
           └─slapd-remain-after-exit.conf
   Active: active (running) since Wed 2019-11-20 13:06:49 CET; 5min ago
     Docs: man:systemd-sysv-generator(8)
    Tasks: 3 (limit: 19063)
   Memory: 4.6M
   CGroup: /system.slice/slapd.service
           └─1159156 /usr/sbin/slapd -h ldap:/// ldapi:/// -g openldap -u openldap -F /etc/ldap/slapd.d

Nov 20 13:06:49 samsung5-debian systemd[1]: Starting LSB: OpenLDAP standalone server (Lightweight Directory Access Protocol)...
Nov 20 13:06:49 samsung5-debian slapd[1159155]: @(#) $OpenLDAP: slapd  (Oct 27 2019 09:38:19) $
                                                        Debian OpenLDAP Maintainers <pkg-openldap-devel@lists.alioth.debian.org>
Nov 20 13:06:49 samsung5-debian slapd[1159156]: slapd starting
Nov 20 13:06:49 samsung5-debian slapd[1159149]: Starting OpenLDAP: slapd.
Nov 20 13:06:49 samsung5-debian systemd[1]: Started LSB: OpenLDAP standalone server (Lightweight Directory Access Protocol).
```
Removing ```--enable-asan``` and ```-with-gui-qt=qt5``` from configure and rebuilding (with ansible installed) is a little better: 1/5 LDAP tests pass: 
[test-suite.log](https://github.com/USBGuard/usbguard/files/3869142/test-suite.log)

jean-christophe-manciot, 2019-11-20:
More details about the build:
```
    Build Configuration Summary    
===================================

 Run full test suite: no

## Libraries

      libqb: system-wide;  -lqb -ldl
     crypto: system-wide;  -lgcrypt
   libaudit: system-wide; -laudit
    libldap: system-wide;  -lldap
 libseccomp: system-wide; -lseccomp
  libcap-ng: system-wide; -lcap-ng
   protobuf: system-wide; -pthread -lprotobuf -pthread
      Catch: bundled; -I$(top_srcdir)/src/ThirdParty/Catch/include 
      PEGTL: bundled; -I$(top_srcdir)/src/ThirdParty/PEGTL/include ; version <= 1.3.1: 
 GLib D-Bus: system-wide; -pthread -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -ldbus-glib-1 -ldbus-1 -lgio-2.0 -lgobject-2.0 -lglib-2.0
   umockdev: system-wide; -lumockdev -lgobject-2.0 -lglib-2.0

## Directories

 D-Bus System Services: ${datadir}/dbus-1/system-services
       D-Bus Busconfig: ${datadir}/dbus-1/system.d
    PolicyKit Policies: ${prefix}/share/polkit-1/actions/
      systemd unit dir: /lib/systemd/system
   Bash completion dir: ${datadir}/bash-completion/completions

## Compilation Flags

Debug Build: yes
       DEFS: -DHAVE_CONFIG_H
   CXXFLAGS: -std=c++11  -g -O0 -fvisibility=hidden  -pedantic -Wno-unknown-pragmas -Wall -Wextra -Wformat=2 -Wredundant-decls -Wcast-align -Wmissing-declarations -Wmissing-include-dirs -Wmissing-format-attribute -Wswitch-enum -Wswitch-default -Winvalid-pch -Wformat-nonliteral -Wno-deprecated-register -Wno-missing-field-initializers -Wno-implicit-fallthrough 
     CFLAGS: -std=c99   -fvisibility=hidden  -pedantic -Wno-unknown-pragmas -Wall -Wextra -Wformat=2 -Wredundant-decls -Wcast-align -Wmissing-declarations -Wmissing-include-dirs -Wmissing-format-attribute -Wswitch-enum -Wswitch-default -Winvalid-pch -Wformat-nonliteral -Wno-deprecated-register -Wno-missing-field-initializers -Wno-implicit-fallthrough 
   CPPFLAGS: -DHAVE_BUILD_CONFIG_H 
    LDFLAGS: 
 ASAN_FLAGS: 
 TSAN_FLAGS: 
```
Also, a few warnings:
```
Makefile:3415: warning: overriding recipe for target 'check'
Makefile:2927: warning: ignoring old recipe for target 'check'
...
src/Library/UMockdevDeviceManager.cpp: In member function ‘bool usbguard::UMockdevDevice::isLinuxRootHubDeviceDescriptor(const usbguard::USBDescriptor*)’:
src/Library/UMockdevDeviceManager.cpp:242:68: warning: type qualifiers ignored on cast result type [-Wignored-qualifiers]
  242 |       reinterpret_cast<const USBDeviceDescriptor* const>(descriptor);
      |                                                                    ^
src/Library/UMockdevDeviceManager.cpp: In member function ‘void usbguard::UMockdevDevice::updateHashLinuxRootHubDeviceDescriptor(const usbguard::USBDescriptor*)’:
src/Library/UMockdevDeviceManager.cpp:262:109: warning: type qualifiers ignored on cast result type [-Wignored-qualifiers]
  262 |     USBDeviceDescriptor descriptor_modified = *reinterpret_cast<const USBDeviceDescriptor* const>(descriptor);
      |                                                                                                             ^
src/Library/UMockdevDeviceManager.cpp: In member function ‘virtual std::shared_ptr<usbguard::Device> usbguard::UMockdevDeviceManager::applyDevicePolicy(uint32_t, usbguard::Rule::Target)’:
src/Library/UMockdevDeviceManager.cpp:617:21: warning: redundant move in return statement [-Wredundant-move]
  617 |     return std::move(device);
      |            ~~~~~~~~~^~~~~~~~
src/Library/UMockdevDeviceManager.cpp:617:21: note: remove ‘std::move’ call
src/Library/UMockdevDeviceManager.cpp: In member function ‘void usbguard::UMockdevDeviceManager::ueventProcessRead()’:
src/Library/UMockdevDeviceManager.cpp:799:75: warning: type qualifiers ignored on cast result type [-Wignored-qualifiers]
  799 |         reinterpret_cast<const struct ucred* const>(CMSG_DATA(cmsg_header));
      |                                                                           ^
At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-deprecated-register’
...
/usr/bin/ld: warning: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libqb.so contains output sections; did you forget -T?
copying selected object files to avoid basename conflicts...
ar: `u' modifier ignored since `D' is the default (see `U')
```

jean-christophe-manciot, 2019-11-20:
I realize now that I misunderstood you regarding ansible.
It seems that it is only used in connection with .travis.
However, your .travis cannot be used as is: it is way too old.
Dismissing LDAP support is the only viable workaround for now.

radosroka, 2019-11-20:
> I realize now that I misunderstood you regarding ansible.
> It seems that it is only used in connection with .travis.
> However, your .travis cannot be used as is: it is way too old.
> Dismissing LDAP support is the only viable workaround for now.

The ansible is there only for setting openldap server properly when running ldap tests in testing environment.

radosroka, 2019-11-20:
If you are not interested in using usbguard-ldap configuration or testing it is OK to omit ldap support during compilation completely.

radosroka, 2019-11-20:
Would you open a new issue for this leak with ASAN ? We can close this issue completely then.

jean-christophe-manciot, 2019-11-21:
I don't know what makes you think there is a "ASAN leak".

closed 2019-11-21 by jean-christophe-manciot