USBGuard/usbguard #346: Rules for one and only one mouse (and keyboard) block mouse (and kb) at boot

auzias, 2019-11-21:
An issue arose when I tried to allow one and only one mouse and one and only one keyboard (let's focus on the mouse, the issue is the same for the keyboard). The rule used to do so, is the rule used in the [documentation](https://usbguard.github.io/documentation/rule-language) :

    allow with-interface one-of { 03:01:02 } if !allowed-matches(with-interface one-of { 03:01:02 })

After a (re)boot, the mouse is blocked by usbguard and it needs to be unplugged and then re-plugged so it is allowed. Logs show these two lines :

    nov. 21 10:36:29 hostname usbguard-daemon[1037]: uid=0 pid=663 device.rule='allow id 17ef:6019 serial "" name "Lenovo USB Optical Mouse" hash "WXaMPh5VWHf9avzB+Jpua45j3EZK6KeLRdPcoEwlWp4=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "1-9" with-interface 03:01:02' type='Device.Present' result='SUCCESS' device.system_name='/sys/devices/pci0000:00/0000:00:14.0/usb1/1-9'
    nov. 21 10:36:29 hostname usbguard-daemon[1037]: uid=0 pid=663 result='SUCCESS' device.system_name='/sys/devices/pci0000:00/0000:00:14.0/usb1/1-9' target.new='block' type='Policy.Device.Update' device.rule='allow id 17ef:6019 serial "" name "Lenovo USB Optical Mouse" hash "WXaMPh5VWHf9avzB+Jpua45j3EZK6KeLRdPcoEwlWp4=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "1-9" with-interface 03:01:02' target.old='allow'

It seems that, at an update occurs and then block the mouse that was just allowed.

In the second line, the value of `type` (`Policy.Device.Update`) caught my attention : there is no option in the configuration file for `UpdateDevicePolicy` whereas there is one for the type `Device.Present` (`PresentDevicePolicy`). Nevertheless, I'm really unsure of its relevance with the issue.

Since the daemon does not log that much, the `usbguard.service` has been updated to include the `-d` flag to get more messages. After a reboot, the first time the string `Mouse` occurs is on the line :

    nov. 21 11:40:25 hostname usbguard-daemon[741]: DevicePrivate.cpp@120/getDeviceRule: return: device_rule=block id 17ef:6019 serial "" name "Lenovo USB Optical Mouse" hash "WXaMPh5VWHf9avzB+Jpua45j3EZK6KeLRdPcoEwlWp4=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "1-9" with-interface 03:01:02

Where the mouse is blocked, which seems unlogic with the log with no debug.

Any suggestion ?

---
```
Kernel : 4.15.0-55-generic
Unbuntu 18.04.3 LTS
usbguard0 package version 0.7.2+ds-1
libusbguard0 package version 0.7.2+ds-1
```

tammar96, 2019-11-21:
Have you tried it on version 0.7.6?

auzias, 2019-11-21:
No, but I might. Has there been any changes on this ?

ZoltanFridrich, 2021-03-04:
@auzias See #456 (which is a duplicate of this issue) for an explanation.