USBGuard/usbguard #347: Can't allow usbguard device permanently

ibahnasy, 2019-11-22:
Unable to whitelist device permanently due to filesystem restriction in the systemd unit file.
If "ReadOnlyPaths=-/" is disabled, it works fine

Package version(s): 0.7.5-1

Steps to reproduce:
# usbguard allow-device 15 -p
IPC ERROR: request id=1: FileRuleSet saving: /etc/usbguard/rules.conf: Read-only file system

c0d3z3r0, 2019-11-23:
I had the same problem; workaround:

~~~
sudo mkdir /etc/systemd/system/usbguard.service.d
cat <<"EOF" | sudo tee /etc/systemd/system/usbguard.service.d/override.conf
[Service]
ReadWritePaths=-/dev/shm -/var/log/usbguard -/tmp -/etc/usbguard
EOF
~~~

radosroka, 2019-12-06:
Fixed in : https://github.com/USBGuard/usbguard/commit/fffd3d3fc16ff3426e95350b061a44f0ce02d3c7

closed 2019-12-06 by radosroka