USBGuard/usbguard #350: usbguard daemon dies with sigwaitinfo failed

robo2bobo, 2019-12-15:
While doing a regular security audit, I tested several laptops with foreign USB sticks and to my astonishment they all worked, even though the laptops are protected by usbguard.

Upon further investigation I found out that usbguard dies for some reason, systemctl status usbguard output:

```
● usbguard.service - USBGuard daemon
   Loaded: loaded (/usr/lib/systemd/system/usbguard.service; enabled; vendor preset: disabled)
   Active: inactive (dead) since Wed 2019-12-11 16:38:06 UTC; 3 days ago
     Docs: man:usbguard-daemon(8)
 Main PID: 2472 (code=exited, status=0/SUCCESS)

Dec 11 16:38:06 laptop usbguard-daemon[2472]: sigwaitinfo failed: errno=4; Shutting down.
```

I've never seen this on desktops, this appears to be related to laptops only and in my limited testing I believe its related to the suspend ability. These laptops run Fedora 30 or Fedora 31 and they are all DELL XPS 13.

Thank you.

radosroka, 2019-12-16:
Can you verify that https://github.com/USBGuard/usbguard/pull/343 fixes the issue?

robo2bobo, 2019-12-16:
Unfortunately I am not able to run my own compiled binaries on these laptops. I have to wait for a new RPM package to be released by Fedora.

radosroka, 2019-12-16:
You can download and install an update from here https://bodhi.fedoraproject.org/updates/FEDORA-2019-d4c4bf697a . There is a problem in selinux subpackage ant that's the reason why it wasn't released yet. You can still remove usbguard-selinux*.rpm after instalation and daemon will work.
The fix is present in that update.

genodeftest, 2020-02-05:
Interestingly, *without* updating usbguard lately, the issue has not been happening for quite a while for me.

robo2bobo, 2020-02-12:
Its a serious issue for me, the process won't stay alive for long, not even on servers that are 100% online. At first I thought this only happened to laptops, apparently not.

Just checking several CentOS 8 servers and I see:

`usbguard-daemon[1764]: sigwaitinfo failed: errno=4; Shutting down.`

radosroka, 2020-02-12:
> Unfortunately I am not able to run my own compiled binaries on these laptops. I have to wait for a new RPM package to be released by Fedora.

The update should be available for Fedora 31.

genodeftest, 2020-02-12:
> Interestingly, _without_ updating usbguard lately, the issue has not been happening for quite a while for me.

After updating to `usbguard-0.7.6-6.fc31.x86_64`, I also don't see the issue (not more than before). I guess this is related to some changes outside of usbguard, e.g. in the kernel, which influences timing.

closed 2020-02-18 by radosroka