USBGuard/usbguard #359: Unable to edit rules.conf from usbguard cli

genodeftest, 2020-02-05:
Steps to reproduce:
1. I've added `root` so that my `/etc/usbguard/IPCAccessControl.d/root` looks like this: ```
Devices=list,modify,listen
Policy=list,modify
Exceptions=listen
```
2. I've restarted usbguard using `systemctl restart usbguard`
3. I'm trying to permanently enable an existing device using `usbguard allow-device -p 9`

What happens:
> IPC ERROR: request id=1: FileRuleSet saving: /etc/usbguard/rules.conf: Permission denied

What should happen:
Modify policy without error.

Additional info:
`ls -al /etc/usbguard` shows that all files and folders in `/etc/usbguard` are owned by `root:root`. The files are `rw-------` and the folder `IPCAccessControl.d` is `rwxr-xr-x`.

genodeftest, 2020-02-05:
I also tried everything in #287 but I wasn't successful.

genodeftest, 2020-02-05:
Workaround: Check the output of `usbguard generate-policy` and if it is ok, write it to the config file: `# usbguard generate-policy > /etc/usbguard/rules.conf`.

radosroka, 2020-02-10:
What about SELinux? Are there any AVCs?

genodeftest, 2020-02-10:
> What about SELinux? Are there any AVCs?

No, systemd-journald has nothing on rules.conf and SELinux Troubleshooter has nothing on usbguard.

tammar96, 2020-02-10:
Hi, I was examining the problem but everything works when I tried it. 
Can you please provide context of a folder /etc/usbguard `ls -l -Z /etc/usbguard` ? Is the usbguard running within correct context? `ps aux -Z | grep usbguard` ?

genodeftest, 2020-02-11:
First command:
```
$ LC_ALL=C ls -l -Z /etc/usbguard
total 24
drwxr-xr-x. 1 root root system_u:object_r:usbguard_conf_t:s0        26 Feb  5 22:50 IPCAccessControl.d
-rw-------. 1 root root unconfined_u:object_r:usbguard_conf_t:s0  2039 Feb  5 23:40 rules.conf
-rw-------. 1 root root unconfined_u:object_r:usbguard_rules_t:s0 4745 Feb  5 22:31 rules.conf.bak
-rw-------. 1 root root unconfined_u:object_r:usbguard_conf_t:s0  1867 Feb  5 22:52 rules.conf.new
-rw-------. 1 root root system_u:object_r:usbguard_conf_t:s0      5088 Jan 24  2018 usbguard-daemon.conf
```

Second command:
```
$ LC_ALL=C ps aux -Z | grep usbguard
system_u:system_r:usbguard_t:s0 root         765  0.0  0.0  94828  8428 ?        Ssl  Feb10   0:00 /usr/sbin/usbguard-daemon -k -c /etc/usbguard/usbguard-daemon.conf
```

genodeftest, 2020-02-11:
Restorecon tells me that something is wrong with these files:
```
$ restorecon -nvr /etc/usbguard
Would relabel /etc/usbguard/rules.conf.bak from unconfined_u:object_r:usbguard_rules_t:s0 to unconfined_u:object_r:usbguard_conf_t:s0
Would relabel /etc/usbguard/rules.conf from unconfined_u:object_r:usbguard_conf_t:s0 to unconfined_u:object_r:usbguard_rules_t:s0
```
After running `restorecon -r /etc/usbguard`, it works fine again. In other words: It looks like a wrong SELinux label on the file caused this issue. I probably got the label wrong when I ran the workaround (described above).

Sorry for the noise and thank you very much for the help!

closed 2020-02-11 by genodeftest