USBGuard/usbguard #362: First block then allow

epichub, 2020-02-12:
I am trying to allow a Lacie external drive to connect to my system. I have added it to the rules, but I now get a strange behaviour of the usbguard first blocking, then allowing the device. The end result is no block device available in my system ..

`[104055.106475] usb 2-5: USB disconnect, device number 8
[104056.973773] usb 2-5: new SuperSpeed Gen 1 USB device number 9 using xhci_hcd
[104056.986508] usb 2-5: New USB device found, idVendor=059f, idProduct=108c, bcdDevice= 1.00
[104056.986514] usb 2-5: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[104056.986518] usb 2-5: Product: Rugged USB-C
[104056.986521] usb 2-5: Manufacturer: LaCie
[104056.986523] usb 2-5: SerialNumber: 0000NL696EZK
[104056.986787] usb 2-5: Device is not authorized for usage
[104056.990195] usb 2-5: authorized to connect
`

genodeftest, 2020-02-12:
> The end result is no block device available in my system .

What do you mean by this sentence?

Can you see the device with `lsusb` or `lsblk`?

ZoltanFridrich, 2021-02-11:
I believe this immediate block and allow process is the result of the device immediately getting blocked due to the default-policy target, then getting allowed because there is a rule found within a policy that allows it. This is mostly my guess.
May I suggest you try setting default-policy-target to allow and placing a block rule to the end of your policy, so the devices will by default get allowed and then blocked after going through the policy. However, there might be some security issues to leave this short allow window before blocking.