USBGuard/usbguard #363: /sys is read-only with current systemd service file

AnanasPfirsichSaft, 2020-02-19:
In March 2019 the systemd unit file was updated and more hardening options have been enabled. After upgrading from 0.7.4 to 0.7.6 I got the following error multiple times. Effectively making **changes** to device authorizations impossible:
```
(E) Device insert exception: SysFSDevice: authorized_default: Read-only file system
(W) SysFSDevice: authorized_default: Read-only file system
(W) Rejecting device at syspath=
(W) USB Device Exception: SysFSDevice: remove: Bad file descriptor
```
This is because the service file sets "ReadOnlyPaths=-/". If I comment this out, usbguard works fine again. According to systemd manpage you could set "ProtectSystem=strict" to make everything except /dev, /proc and /sys read-only. But I do not know since when systemd supports this option and if in example falls back to "ProtectSystem=true".

One can mitigate this issue by editing the service file with "systemctl edit usbguard.service" and put this into the editor:
```
[Service]
ProtectSystem=strict
ReadOnlyPaths=
```
Or this:
```
[Service]
ReadWritePaths=-/dev/shm -/var/log/usbguard -/tmp -/sys
```
----------------------------------
systemd 237

NAME="KDE neon"
VERSION="5.18"
ID=neon
ID_LIKE="ubuntu debian"
PRETTY_NAME="KDE neon User Edition 5.18"
VARIANT="User Edition"
VERSION_ID="18.04"
HOME_URL="http://neon.kde.org/"
SUPPORT_URL="http://neon.kde.org/"
BUG_REPORT_URL="http://bugs.kde.org/"
LOGO=start-here-kde-neon
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

genodeftest, 2020-03-02:
Interesting. In my case, `/sys` is readonly, but I see no such warnings or errors.

This is my usbguard.service file: https://gist.github.com/genodeftest/09f0e93300828a7f2c69f1cdc2f02c2c
I have no override.conf or similar stuff. This is with `systemd-243.7-1.fc31.x86_64` and `systemd-243.7-1.fc31.x86_64`.

In the systemd "changelog" I could only find stuff which looks remotely related:
* https://github.com/systemd/systemd/blob/master/NEWS#L1804 Maybe try `restorecon` on your file system?
* https://github.com/systemd/systemd/blob/master/NEWS#L2413 but I doubt that usbguard is using bpf

radosroka, 2020-09-28:
I believe this is fixed already... Can you verify that?

AnanasPfirsichSaft, 2020-10-02:
I have updated my Neon to Ubuntu focal base last month. It is now systemd 245 and I installed the package from the repository (0.7.6+ds-1build1).

It runs without errors with the stock service file – identical to [genodeftest](https://gist.github.com/genodeftest/09f0e93300828a7f2c69f1cdc2f02c2c) and no overrides anymore.

radosroka, 2020-10-04:
> I have updated my Neon to Ubuntu focal base last month. It is now systemd 245 and I installed the package from the repository (0.7.6+ds-1build1).
>
> It runs without errors with the stock service file [...]

Thanks.

closed 2020-10-04 by radosroka