USBGuard/usbguard #364: No USBGuard GUI Front-end Available for 0.7.5+

LifeAsALemon, 2020-03-02:
So, this is my first "issue" that I would have ever submitted to any 'Git' project of any kind, so apologies in advance if this is the incorrect way to approach the problem.

To the problem:---

As things currently stand, USBGuard 0.7.5 (and newer) does not have ANY compatible GUI Front-End for the end user to easily operate.

I can see that mid-way through 2019 (last year) a issue was raised in regards to this and alternatives were suggested, the issue now is that the alternatives are no longer viable, and because of the decision to stop support the older Qt-based interface we have been left with no option for a GUI front-end for USBGuard.

What efforts are currently being made to create a new, user friendly, GUI front-end for USBGuard?

Is their a GUI currently in development and if their is not, what can we do as a community in order to create one that can be eventually up-streamed into the main project repository.

I would love to create a GUI front-end if someone was willing to point me in the right direction of where to start, I have NO prior software development experience, I do however have a keen eye for GUI design/styling and have plenty of free time that I would love to offer to help progress this effort forwards for the benefit of the project, and for the benefit of the wider community as I feel that a lack of any GUI really hampers USBGuard's appeal to the end-user.

Thanks for creating such an awesome project, and I hope that my offer to assist is welcomed with open arms :)

All the best,
LAAL.

muelli, 2020-03-02:
AFAICS, you have a few options. One is the frontend [built into GNOME](https://gitlab.gnome.org/GNOME/gnome-settings-daemon/merge_requests/75), [USBGuard GNOME](https://github.com/k00n/usbguard-gnome), or [usbguard-notifier](https://github.com/Cropi/usbguard-notifier). There are probably more.

genodeftest, 2020-03-02:
> AFAICS, you have a few options. One is the frontend [built into GNOME](https://gitlab.gnome.org/GNOME/gnome-settings-daemon/merge_requests/75), [USBGuard GNOME](https://github.com/k00n/usbguard-gnome), or [usbguard-notifier](https://github.com/Cropi/usbguard-notifier). There are probably more.

The first one is not released yet, you'll have to wait for the release of GNOME 3.36 around 2020-03-11 and until it gets to your favorite linux distro.

USBGuard GNOME is not maintained and [incompatible](6E006B/usbguard-gnome) with the latest USBGuard releases.

usbguard-notifier only provides notifications but no way to interact with usbguard, e.g. allowing devices.

Personally, I have switched to using the terminal. USBGuard's CLI is very nice and stable for me. Just open a terminal and use `usbguard list-devices` and `usbguard allow-device [number]`.

LifeAsALemon, 2020-03-02:
> > AFAICS, you have a few options. One is the frontend [built into GNOME](https://gitlab.gnome.org/GNOME/gnome-settings-daemon/merge_requests/75), [USBGuard GNOME](https://github.com/k00n/usbguard-gnome), or [usbguard-notifier](https://github.com/Cropi/usbguard-notifier). There are probably more.
> 
> The first one is not released yet, you'll have to wait for the release of GNOME 3.36 around 2020-03-11 and until it gets to your favorite linux distro. Furthermore, it is not providing much of USBGuard's functionality but only a very limited subset of it.
> 
> USBGuard GNOME is not maintained and [incompatible](6E006B/usbguard-gnome) with the latest USBGuard releases.
> 
> usbguard-notifier only provides notifications but no way to interact with usbguard, e.g. allowing devices.
> 
> Personally, I have switched to using the terminal. USBGuard's CLI is very nice and stable for me. Just open a terminal and use `usbguard list-devices` and `usbguard allow-device [number]`.

I was unaware that the "USBGuard Gnome" package was actually different to the package that is built into Gnome itself (Will be included in Gnome 3.36). So that is an interesting development to me.

The actual package named "USBGuard Gnome" is in fact not maintained and you are correct in saying that it is incompatible as that has also been my experience with it so far.

"USBGuard Notifier" is not suitable as all it does is provide notifications and not an actual ability to interact with any form of GUI to control the various functions of USBGuard such as allowing, disallowing or blocking of devices upon connection to the host.

This is why I desperately would love to help create something to replace the hole that the now defunct USBGuard Qt-Applet has left.

For me as an end-user of USBGuard, I can see the huge potential that it offers, but that potential will not be fully realised for end-users unless we as end-users ourselves contribute and offer a decent GUI front-end that is not only modern is design, but functional in it's use, and reliable in that it is kept up-to-date alongside the main USBGuard releases.

Do you know of where I might be able to look if I wanted to start such an endeavour to offer a truly amazing front-end GUI for this really cool & very important project?

I ask this as right now I do not know where to start, who to ask, or what would be needed in terms of workload in order to pull this off.
What I do know is that I have a lot of free time and a willingness to improve the whole experience, plus a good eye for design and a logical mindset for GUI functionality/usability. 

Thank you for contributing to my initial question, hopefully it will bring more attention to the issue at hand.

Many Thanks,
LAAL

LifeAsALemon, 2020-03-02:
> AFAICS, you have a few options. One is the frontend [built into GNOME](https://gitlab.gnome.org/GNOME/gnome-settings-daemon/merge_requests/75), [USBGuard GNOME](https://github.com/k00n/usbguard-gnome), or [usbguard-notifier](https://github.com/Cropi/usbguard-notifier). There are probably more.

As I have stated above alongside other users, "USBGuard Notifier" is not a interact-able GUI, it is just a simple notification system for USBGuard. And "USBGuard Gnome" is no longer maintained as well as being incompatible with the current version of USBGuard.

But I appreciate your contribution to the discussion, it helps to bring more light to this issue, and so hopefully we/I will be able to address the problem with a viable alternative GUI in the future.

I just need to understand where I need to start, what I will need to do to create a GUI front-end and then actually make it, using a Graphical Interface that is cross-compatible, so either Qt, or GTK, unless their is another option other than those two.

genodeftest, 2020-03-02:
You may start with the old USBGuard applet and copy it from this git repository and try to build it against the current version of USBGuard. Or go and fix bugs in USBGuard GNOME. Or help beta-test GNOME 3.36.

Lem, 2020-03-25:
Seems to be a duplicate of #334.

MatejKovacic, 2024-03-15:
Anything new on this?